bluekitchen/btstack #94: ESP32 Bluetooth connection through Matlab?

ljdankov, 2017-12-26:
Hello,
I have been working my way through some of your SPP Counter for the ESP32, and am trying to open a connection in Matlab. The eventual goal is to pass data from sensors to Matlab for analysis in real time. I have gotten my PC to recognize the device as advertising COM3 as a serial port service, and gotten Matlab to recognize the device. 

However Matlab is consistently failing to connect to device with following error.
         >> fopen(B)
         Error using icinterface/fopen (line 83)
         Unsuccessful open: Cannot connect to the device. Possible reasons are another application is 
         connected or the device is not available. 
Is there something specific in the code that needs to be changed to allow incoming connections from Matlab?
Thanks in advance,
Louis

ljdankov, 2017-12-26:
Never Mind. I fixed it myself. Apparently it really helps if I tell Matlab to connect to the correct channel! ::facepalm::

closed 2017-12-26 by ljdankov

mringwal, 2017-12-26:
Nice idea to get live data into Matlab. Have fun!