bluekitchen/btstack #95: Help me,about hfp_hf_demo and a2dp_sink_demo,there are some Serious problem

907038134, 2018-01-02:
I used usb dongle to run hfp_hf_demo and a2dp_sink_demo in ubuntu and asked for help，Thank you very much for your help

[1], running hfp_hf_demo, redial the call and then hang up, the program stuck, stuck in the btstack_run_loop_posix_execute () function ts-> process (ts); Here, every time there will be such a situation, I do not know What causes this kind of situation,

[2] hang up the phone, through the log view, hang up when the state is wrong log error:
---------------------------------------------------------------------------------------------
[2018-01-01 22:01:36.605] LOG hfp.c.709: Received gap disconnect in wrong hfp state
[2018-01-01 22:01:36.605] LOG hfp.c.711: Check SCO handle: incoming 0x40, hfp_connection 0x40

[2018-01-01 22:01:36.605] LOG hfp.c.714: SCO disconnected, w2 disconnect RFCOMM

[2018-01-01 22:01:36.606] LOG hci.c.826: Connection closed: handle 0x40, 80:BE:05:51:DD:8B
[2018-01-01 22:01:36.606] LOG hci.c.3759: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[2018-01-01 22:01:36.606] EVT <= 61 01 01 
[2018-01-01 22:01:36.606] LOG hci_transport_h2_libusb.c.1336: usb_set_sco_config: voice settings 0x0043, num connections 0
[2018-01-01 22:01:36.607] LOG hci_transport_h2_libusb.c.849: usb_sco_stop
Error parsing pklg at offset 585889 (8f0a1).
root@ubuntu:/opt/btstack-master/tool# 
---------------------------------------------------------------------------------------------

[3] running hfp_hf_demo connected to the phone, the call did not pick up the program on the collapse, what is the situation, pick up the call, more than a minute or two a hang up also crashed

Thanks for everyone's opinion

907038134, 2018-01-02:
Who knows what the problem is, or who's going to have the same problem, and you can help with information, thank you

mringwal, 2018-01-08:
Hi 907038134. I don't fully understand what problem(s) you are reporting. Could you describe a single problem in detail: what did you do, what happened, what did you expect that should happen? Together with a complete packet log file (/tmp/hci_dump.pklg) for comparision?

mringwal, 2018-07-31:
no response, closing. we also fixed other issues in a2dp/avrcp.

closed 2018-07-31 by mringwal