bluekitchen/btstack #98: About btstack_chipset_csr.c,What does init_script mean?

907038134, 2018-01-16:
hi, everyone ,in my test i should set scr bluetooth sco routing to PCM, but i don't know btstack_chipset_csr.c,What does init_script mean?Thank you very much for your help

    In my test ,i have to set sco routing to PCM,but when i use msbc in sco connect , I check sampling rate in PCM io only 8Khz, i should set it to 16khz, maybe i should use btstack_chipset_csr.c to set it ,but i don't know what it means?and hao can i learn it?I checked some information of CSR, but these data and corresponding?
// minimal CSR init script to configure PSKEYs and activate them
static const uint8_t init_script[] = {

    // 0x01fe: Set ANA_Freq to 26MHz
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xfe, 0x01, 0x01, 0x00, 0x00, 0x00, 0x90, 0x65,

    // 0x00f2: Set HCI_NOP_DISABLE
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xf2, 0x00, 0x01, 0x00, 0x00, 0x00, 0x01, 0x00,

#ifdef ENABLE_SCO_OVER_HCI
    // 0x01ab: Set HOSTIO_MAP_SCO_PCM to 0
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xab, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
    // 0x01b0: Set HOSTIO_MAP_SCO_CODEC to 0
    0x00, 0xFC, 0x13, 0xc2, 0x00, 0x00, 0x09, 0x00, 0x03, 0x00, 0x03, 0x70, 0x00, 0x00, 0xb0, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
    // 0x22c9: Set ENABLE_SCO_STREAMS to 0
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xc9, 0x22, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
#endif
    // 0x01bf: Enable RTS/CTS for BCSP (0806 -> 080e)
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x01, 0x00, 0x03, 0x70, 0x00, 0x00, 0xbf, 0x01, 0x01, 0x00, 0x00, 0x00, 0x0e, 0x08,

    // 0x01ea: Set UART baudrate to 115200
    0x00, 0xFC, 0x15, 0xc2, 0x02, 0x00, 0x0a, 0x00, 0x02, 0x00, 0x03, 0x70, 0x00, 0x00, 0xea, 0x01, 0x02, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0xc2,

    // 0x0001: Set Bluetooth address
    0x00, 0xFC, 0x19, 0xc2, 0x02, 0x00, 0x0A, 0x00, 0x03, 0x00, 0x03, 0x70, 0x00, 0x00, 0x01, 0x00, 0x04, 0x00, 0x00, 0x00, 0xf3, 0x00, 0xf5, 0xf4, 0xf2, 0x00, 0xf2, 0xf1,

    //  WarmReset
    0x00, 0xFC, 0x13, 0xc2, 0x02, 0x00, 0x09, 0x00, 0x03, 0x0e, 0x02, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
};

please help me, thanks!

mringwal, 2018-01-17:
hi. BTstack uses "init script" for anything related to configuration of the Bluetooth Controller before it can be used. On CSR chipsets, so-called PS-Keys can be written. Also read https://github.com/bluekitchen/btstack/blob/master/chipset/README.md

Now. On an CSR controller with UART connection, SCO data is delivered via the PCM/I2S pins. This is enough for narrow-band (CVSD) audio. For wide-band audio, it's necessary to get the SCO data to BTstack so it can be decoded by the mSBC decoder. In theory, the commands above should configure the CSR to send SCO via HCI, but it doesn't work. The best source for this kind of details has been the BlueZ / Linux kernel, but I couldn't figure out what's needed.

907038134, 2018-01-18:
@mringwal 
         hi, Thanks for your reply！
         Now，I  know what is the meaning of the data, I use a csr8811, I must be configured SCO to PCM, but now is that the MSBC data flow to the PCM pin out after only 8 KHZ of sampling rate, As you said:  MSBC can't flow to PCM,  only flow to  btstack hci..

907038134, 2018-01-18:
i have use btstack_chipset_csr.c and add csr's psk to configured csr's pcm, but it doesn't work,  maybe i Configuration error （about PSKEY_PCM_CONFIG32 and  PSKEY_PCM_USE_LOW_JITTER_MODE），it so confusing for me

mringwal, 2018-01-18:
SCO data has a fixed data rate of 64 kbps = 8 kB/s. This is either used for CVLD or for mSBC. mSBC encodes/decodes 16 kHz/16-bit data. If the SCO data goes over HCI, BTstack can decode/encode it.

907038134, 2018-01-19:
Hello!
(1) my bluetooth is connected by a serial port.
(2) you say that a bluetooth connection via a serial port can't take sco over hci, can it only be routed to PCM?
(3) you describe that the WBS(MSBC) data cannot be routed to PCM, which requires HCI to be decoded.
According to the above description, my current condition only supports narrow-band voice routing to PCM, and WBS voice cannot be routed to PCM, and it cannot be routed to HCI?
Please confirm it for me and thank you for your reply!

mringwal, 2018-01-19:
It depends on the Bluetooth Controller. BTstack knows how to configure TI CC256x and Broadcom chipsets to route SCO over HCI. I don't know how to do it for CSR8811, but it's possible.

By default SCO data is routed to I2S. In theory, this should also work for mSBC data and if the I2S output is connected to an I2S input on the MCU, I guess it can be decoded, too. I didn't try this yet.

You could try out the CC256x, there's a cheap breakout-board: http://www.ti.com/tool/BOOST-CC2564MODA

ramki31, 2018-11-03:
@mringwal 
TOOLS:BLUEZ
BOARD:BT830-SA-01 with psr8811 support
OS:LINUX

1.I have configured bt830   board  with i2s/pcm by using attached psr file.
2.I need audio data on this i2s/pcm interface if play some some audio from my mobile.
3.I have connected bt830 board to my laptop using uart.I have connected to my mobile with bt830 board using bluetooh and started playing songs but no data is coming on this i2s/pcm lines.but audio is playing on my laptop.

**NOTE:**I am probing at open interface means no slave is connected to it
4.why data is not coming on i2s lines do i need further changes in psr file

5.Do i need to change bluez stack for routing ? please atleast 
[BT830_Linuxpsr.txt](https://github.com/bluekitchen/btstack/files/2545078/BT830_Linuxpsr.txt)
guide me what has to do in stack level.
6.How to differentiate between i2s/pcm interface?how to select them as same passkey is used for pcm /i2s?
7.Can we tranasmit wide band data ovet pcm/i2s and pass this to codec

mringwal, 2018-11-05:
I've answered on the mailing list:
https://groups.google.com/forum/#!topic/btstack-dev/Q9r3GZrvz_g

closed 2018-11-05 by mringwal