bluekitchen/btstack #99: hi，i find a bug in hfp.c parse_sequence()

907038134, 2018-01-17:
when i use AT+CLCC check phone's call states , i get a wrong number from app.i check log and pint it ,i think this is a bug:now i get a right number.
      i'm poor in english ,bug i think you can understand what I mean
/* wrong
                case 4:
                    strncpy(hfp_connection->bnip_number, (char *)hfp_connection->line_buffer, sizeof(hfp_connection->bnip_number));
                    hfp_connection->bnip_number[sizeof(hfp_connection->bnip_number)-1] = 0;
                    break;
                case 5:
                    value = btstack_atoi((char *)&hfp_connection->line_buffer[0]);
                    hfp_connection->bnip_type = value;
                    break;
*/
//right
                case 4:
                    value = btstack_atoi((char *)&hfp_connection->line_buffer[0]);
                    hfp_connection->bnip_type = value;
                    break;
                case 5:
                    strncpy(hfp_connection->bnip_number, (char *)hfp_connection->line_buffer, sizeof(hfp_connection->bnip_number));
                    hfp_connection->bnip_number[sizeof(hfp_connection->bnip_number)-1] = 0;
                    break;

907038134, 2018-01-17:
also when i has 2 call (one hold and one active )i use at+clcc ,only record the first one call state

Query current call status
HFP_RX 
+CLCC: 1,1,1,0,0,"xxxxxx00260",129,"  xxxxxx  "
Enhanced Call Status: idx 1, dir 1, status 1, mpty 0, number xxxxxx, type 0
HFP_RX 
+CLCC: 2,1,0,0,0,"xxxxxx00261",129," xxxxxx  "
Enhanced Call Status: idx 1, dir 1, status 1, mpty 0, number xxxxxx, type 0

mringwal, 2018-01-17:
hi. thanks for reporting! we'll look into this and let you know here.

mringwal, 2021-07-08:
Closing this due to inactivity. :)
We also have reworked the at parser, so it might be fixed, too.

closed 2021-07-08 by mringwal