bluekitchen/btstack #100: ESP32/A2DP Sink Example issue (does not work)

andrewvoznytsa, 2018-01-18:
I'm using https://github.com/espressif/esp-idf/releases/tag/v3.0-rc1 and your master.

Basically A2DP sink example does something that looks like SBC decoding and then stops. If CPU does not resets (reset happens from time to time) then code always reports that 216 frames were decoded (never more than 216).

Attached log output is from your modified example (port/esp32/example/a2dp_sink_demo) where wav writer calls were replaced by calls to I2S renderer. Unmodified example works exactly the same.

As for me it looks that decoding hangs at

[00:00:20.320] EVT <= 6E 00

then after some time channel is closed.

[log.txt](https://github.com/bluekitchen/btstack/files/1643436/log.txt)

Could you please clarify if you have any plans and time frames for fixing this issue?

mringwal, 2018-01-18:
Hi. The full packet log prevents the A2DP examples from working. Could you comment the call to hci_dump_open() in esp-idf/components/btstack/main.c and try again?

andrewvoznytsa, 2018-01-18:
Much -much better :) Works a few minutes already.

Thanks a lot - I''ll continue evaluation.

andrewvoznytsa, 2018-01-18:
Seems like times it worked longer than 216 frames a few times.

Now, I'm always getting ASSERT_ERR(). Do you have some quick fix for that?

 -- a2dp sink demo: stream started, a2dp cid 0x01, local_seid 1
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_acl.c at line 1900abort() was called at PC 0x40082af9 on core 0

Backtrace: 0x4008a1f8:0x3ffc02c0 0x4008a2f7:0x3ffc02e0 0x40082af9:0x3ffc0300 0x40082c19:0x3ffc0330 0x400d385a:0x3ffc0350 0x400d38b5:0x3ffc0390 0x400e422a:0x3ffc03b0 0x400e423b:0x3ffc03d0 0x400f164c:0x3ffc0400 0x400eb892:0x3ffc0420 0x4001791d:0x3ffc0440 0x400ebe11:0x3ffc0460 0x400ebffa:0x3ffc04a0 0x400188f5:0x3ffc04d0 0x40055fca:0x3ffc0510 0x40084012:0x3ffc0530 0x40032ae6:0x3ffc0550 0x4003459e:0x3ffc0590 0x4003a85e:0x3ffc05b0 0x400554a1:0x3ffc05d0 0x40086b4b:0x3ffc05f0 0x400823c1:0x3ffc0610 0x4000bfed:0x00000000

Rebooting...

Full log attached.
[log2.txt](https://github.com/bluekitchen/btstack/files/1643958/log2.txt)

andrewvoznytsa, 2018-01-18:
Disabled watchdog - problem gone. Are you able to confirm that this issue is caused by watchdog task? Or I was just lucky at this moment?

mringwal, 2018-01-19:
I can confirm that I have a report where someone used an external Bluetooth module for A2DP with the ESP32 and it was running without errors for days. I have to admit that I don't fully understand the ESP32 Watchdogs, but I think I've also disabled them.

Can you post the end of the log, maybe from just before it disconnects to the end?

andrewvoznytsa, 2018-01-19:
Attached log was full log - nothing to add. I need to uncomment hci_dump_open() to get more info and we'll end up where we started.

I think it all depends on ESP IDF version - I'd suggest to try your stack with ESP IDF v3.0-rc1 and you'll reproduce that on your side.

Or we can go other way- tell me please which version of ESP IDF should I use and how sdkconfig should looks like.

I'm interested in (working) dual mode stack for our product - need GATTS and A2DP running in parallel, with low footprint.

mringwal, 2018-01-19:
hi. I'm using the latest from esp-idf master with the sdkconfig from the BTstack ESP32 port on the develop branch. (identical to master right now).

in log2.txt, there's a reboot caused by watchdog but no mentioning of a second connection. 
I'll start the a2dp_sink_demo now and let it run for a while without I2S.

Feel free to contact me at contact@bluekitchen-gmbh.com about your product. If you don't need Wifi together with Bluetooth Classic, ESP32 + BTstack should be a viable option.

mringwal, 2018-01-19:
I've streamed a full children's audio book to the ESP32 without I2S and it still reacts as expected, I can pause/resume and also reconnect. Could you try with BTstack & esp-idf master without the additional i2s code to see if it is stable? You may e.g. print the number of received SBC media packets to have some output.

andrewvoznytsa, 2018-01-19:
Thanks for update - will try later today or tomorrow - need to finish something.

I tried yesterday without I2S (your code without changes) - did not work for me, behaviour was like with I2S. I think I need to repeat everything from scratch - perhaps it is just SDK misconfiguration.

Regarding BLE+BTC+WiFi in parallel - any serious issues? If Espressif helps us - will it work?

mringwal, 2018-01-19:
Bluetooth Classic + Wifi doesn't work currently. Recently, there was some feedback by Espressif that indicates that they work on it. See: https://github.com/espressif/esp-idf/issues/created_by/mringwal

andrewvoznytsa, 2018-05-08:
I'm trying to reach you at contact@bluekitchen-gmbh.com... Perhaps landed in spam fodler?

mringwal, 2018-05-28:
(original problem seems fixed -> closing)

closed 2018-07-31 by mringwal