bluekitchen/btstack #101: Btstack on ESP32, classic BT legacy authentication crash

rochus-keller, 2018-01-20:
I tried to implement an ESP32 gadget which is not discoverable and where the client connects directly over RFCOMM to a known BT address. I set the RFCOMM security level to 0 and disabled SPP. As long as the client doesn't want to authenticate everything works like charm (see [esp32_btstack_level0_no_auth_no_crash.log](https://github.com/bluekitchen/btstack/files/1648846/esp32_btstack_level0_no_auth_no_crash.log)). If the client requests authentication (on Windows by setsockopt SO_BTH_AUTHENTICATE = true), Btstack fires HCI_EVENT_PIN_CODE_REQUEST and then causes an ESP32 core dump (see [esp32_btstack_level0_auth_crash.log](https://github.com/bluekitchen/btstack/files/1648847/esp32_btstack_level0_auth_crash.log)). The same happens if Btstack RFCOMM security level is set to 2 and the client doesn't explicitly request authentication (see [esp32_btstack_level2_crash.log](https://github.com/bluekitchen/btstack/files/1648848/esp32_btstack_level2_crash.log)). I attached the ESP32 side source code ([spp_streamer.txt](https://github.com/bluekitchen/btstack/files/1648852/spp_streamer.txt)) and the relevant snippets of the Windows client ([win32_bt_client_snippets.txt](https://github.com/bluekitchen/btstack/files/1648853/win32_bt_client_snippets.txt)). Thanks in advance for taking care of this issue.

mringwal, 2018-01-22:
Hi Rochus. I've tried your code (I've re-enabled discoverable to be able to pair from Mac and set the PIN to '0000') and had no problems to connect & pair. The log looks like this

Pin code request len 4
[00:00:05.726] EVT <= 6E 00
[00:00:05.729] ACL => 80 00 10 00 0C 00 01 00 0B 01 08 00 02 00 00 00 80 02 00 00
[00:00:05.736] EVT <= 6E 00
[00:00:05.739] CMD => 35 0C 05 01 80 00 01 00
[00:00:05.743] EVT <= 13 05 01 80 00 01 00
[00:00:05.747] ACL <= 80 20 0A 00 06 00 01 00 0A 02 02 00 03 00
[00:00:05.753] ACL <= 80 20 0A 00 06 00 01 00 0A 03 02 00 03 00
[00:00:05.758] EVT <= 6E 00
[00:00:05.761] ACL => 80 00 14 00 10 00 01 00 0B 02 0C 00 03 00 00 00 06 00 00 00 00 00 00 00
[00:00:05.769] EVT <= 6E 00
[00:00:05.772] ACL => 80 00 14 00 10 00 01 00 0B 03 0C 00 03 00 00 00 06 00 00 00 00 00 00 00
[00:00:05.780] EVT <= 6E 00
[00:00:05.783] CMD => 35 0C 05 01 80 00 02 00
[00:00:05.787] EVT <= 6E 00
[00:00:05.790] CMD => 0D 04 17 E1 1B 3B 24 0F F4 04 30 30 30 30 00 61 72 63 68 5F 6D 61 69 6E 2E 63
[00:00:05.799] EVT <= 13 05 01 80 00 01 00
[00:00:05.802] EVT <= 13 05 01 80 00 01 00
[00:00:05.806] EVT <= 0E 0A 05 0D 04 00 E1 1B 3B 24 0F F4
[00:00:05.811] EVT <= 6E 00
[00:00:05.815] EVT <= 18 17 E1 1B 3B 24 0F F4 F6 21 9C D6 BC CB AC 29 10 D0 55 89 0A 30 FF 1F 00
[00:00:05.823] LOG -- hci.c.351: gap_store_link_key_for_bd_addr: F4:0F:24:3B:1B:E1, type 0

 The ACL packets are probably related to SDP queries.

If you send me your WIndows client, I can try to run that one as all.

Otherwise, I'm afraid you either need to setup debugging for the ESP32 (I recently learned that it is possible, but didn't try yet), or trace the call to gap_pin_code_response (which will end up in hci_run() for sure) with more log_info. The next step should be that BTstack sends the PIN code in an HCI command.

rochus-keller, 2018-01-22:
Hi Matthias, thanks for your response. Good idea to first try to make it work using the OS built-in pairing functions. As you suggest I changed the provided spp_streamer.c with gap_discoverable_control(1) and s_pwd[] = "0000" (no other changes). I also used a Mac this time (Mac mini, El Capitan). The device is visible to the Mac, but when I do pairing the device does again a core dump as soon as I enter the pin on the Mac. The same also happens when I switch off (HCI) logging and assertions.
What ESP-IDF and ESP32 module/version did you use? I'm using ESP-IDF 2.1.1 with a ESP-WROOM-32 (chip ESP32D0WDQ6 revision 1) based board.

mringwal, 2018-01-22:
Well, I just didn't want to write a Windows app to test it :)

I'm using a ESP32 Thing from Sparkfun (was available from https://www.maker-shop.ch when I was looking for one).

I'm using the master branch of esp-idf. Please try on master as well and make sure that components/bt/lib is also updated.

mringwal, 2018-01-22:
(I recently bought 3x  ESP-WROOM-32 as well but I don't expect the hardware to be the problem here)

rochus-keller, 2018-01-22:
It worked. I was able to successfully authenticate and presumably even encrypt from my Windows app without any pairing user interaction, just by using BluetoothSendAuthenticationResponseEx. You can close the issue. 
How can I see in the log (
[device_level_0_no_ssp_no_sdp_windows_lient_auth_encrypt.log](https://github.com/bluekitchen/btstack/files/1653159/device_level_0_no_ssp_no_sdp_windows_lient_auth_encrypt.log)
) whether the link was encrypted or not?

mringwal, 2018-01-22:
Great! 

You can convert the log with tool/create_packet_log.py into a .pklg file which can be opened with Wireshark for analysis. See http://bluekitchen-gmbh.com/btstack/how_to/#bluetooth-hci-packet-logs

In your log, you have 
[00:03:16.733] LOG -- hci.c.3787: hci_emit_security_level 3 for handle 81

The security levels are (also) documented in src/gap.h 
Level 2 requires encryption, Level 3 additionally requires MITM protection.

closed 2018-01-22 by mringwal

rochus-keller, 2018-01-22:
Ok, got it. Thank you very much for your support and this great work.