bluekitchen/btstack #105: hi，About a2dp_sink_demo controls the volume of music played on the iphone

907038134, 2018-01-22:
hi，why Avrcp controls the volume failure of music.Have you ever try it?

mringwal, 2018-01-22:
For volume control, there are volume up/down commands as well as set absolute volume. The iPhone does only support one of them, if I remember correctly. However, I think the volume control still doesn't work with iPhone and we need to investigate this a bit more - at least my Jawbone MINIJAMBOX can control the volume most of the time

907038134, 2018-01-22:
My iphone refused to execute, and I used my android phone to accept it, but there was no actual volume change.

moyaoyi, 2019-01-04:
In my opinion, the volume control issue could be solved, if BtStack supports AVRCP CT and TG in the same time. TG role can send volume changed event to CT when TG device would like to change the volume of CT device. Do you think so? But I did not know how to support CT and TG in the same time.

moyaoyi, 2019-02-11:
To support AVRCP volume control I modified the AVRCP controller code. Now it can support TG volume changed event and CT play/pause function.

907038134, 2019-02-13:
Can you tell me how to modify the btstack to support TG (iphone) to synchronize the volume? Now I need the phone to turn on and off bluetooth to synchronize the volume every time I connect. However, I do not support how to set it. I found that the CT connected to iphone with TG can make the phone synchronize the volume every time!
Thank you very much!

mringwal, 2019-02-19:
Thanks for your input! We've missed that there is only a single AVCTP connection between two devices and that both sides can assume both roles at the same time. We're trying to come up with a minimal fix to support both roles in parallel next. FWIW, we've used HID Volume Up/Down until now...

mringwal, 2019-02-21:
Hi again. Please try a2dp_sink_controller from develop. After an iPhone connects, the volume changes are not reported until the audio starts. After that, volume can be modified on both sides.

Now, we still need to actually control the volume either by extending the btstack_audio_t api or by multiplying the samples. We'll probably go for an application level multiplier for now.

mringwal, 2020-06-26:
Closing as no further input. AVRCP management works, we'll add support for actually changing playback volume in desktop ports

closed 2020-06-26 by mringwal