bluekitchen/btstack #109: How to distinguish a pairing is justworks or numeric comparison

LuckyHowl, 2018-01-26:
Hi，
I porting btstack to BCM20702B， it work new. But there are some problems with SSP pairing.
I set the local IO capacity to SSP_IO_CAPABILITY_DISPLAY_YES_NO, and the peer device set to 
SSP_IO_CAPABILITY_NO_INPUT_NO_OUTPUT. I can't distinguish the is justworks or numeric comparison.
I expect to implement a function: when the pairing is justworks will auto accept, and when the pairing is 
numeric comparison will notify user to comfirm.
Could you tell how to do it? Thanks~

mringwal, 2018-01-26:
hi. in both cases we get an User Confirmation Request Event from the Bluetooth Controller. As the numeric comparison is the result of some random data put into a crypto operation, you might assume that it's just works if the number is zero.

closed 2018-01-26 by mringwal

LuckyHowl, 2018-01-29:
Hi mringwal:
Thank for your answer. 
As you said, assume that it's justworks if the number is zero. I can get an non-zero number when the IO capacity is NO_INPUT_NO_OUTPUT.
I have an idea, could I distinguish it by the local IO capacity and the remote IO capacity?

mringwal, 2018-01-29:
Yes, the chosen pairing method is based on the IO Capabilities and MITM yes/no. However, I don't think you should need to do that.

Could you post a packet log (txt or .pklg) for the case where the number is no zero and you've selected NO_INPUT_NO_OUTPUT?

Found this in the Bluetooth spec: "... If the Host has output capability (DisplayYesNo or KeyboardOnly), it shall display the Numeric_Value until the Simple Pairing Complete event is received...'' - so if you have NO_INPUT_NO_OUTPUT, only Just Works is possible anyway, and you should not display anything. The spec even indicates that you should just accept it: "If the Host has no input and no output it shall reply with the User Confirmation Request Reply command."

LuckyHowl, 2018-01-29:
Hi mringwal, thank for your answer.

Nowday I use the IO capability to distinguish the pairing is justworks or numeric comparison, my code as fellow:

                case HCI_EVENT_USER_CONFIRMATION_REQUEST:
                    // inform about user confirmation request
                    log_info("SSP User Confirmation Request with numeric value '%06"PRIu32"'\n", little_endian_read_32(packet, 8));
                    log_info("SSP User Confirmation Auto accept\n");
                    if(btstack_status.local_ssp_io_capability == 0x01 && 
                        btstack_status.peer_ssp_io_capability == 0x01)
                    {
                        log_info("User confirm\n");
                        memset(&message, 0, sizeof(message));
                        hci_event_user_confirmation_request_get_bd_addr(packet, message.sec_pairing.bd_addr);
                        memcpy(btstack_status.peer_bd_addr, message.sec_pairing.bd_addr, sizeof(message.sec_pairing.bd_addr));
                        message.sec_pairing.key = little_endian_read_32(packet, 8);
                        message.sec_pairing.mode = NUMERIC_COMPARISON_PAIRING_MODE;
                        Paxme_BtMessageEmit(GAP_PAIRING_EVT, &message);
                    }
                    else
                    {
                        log_info("Justworks\n");
                        hci_event_user_confirmation_request_get_bd_addr(packet, event_addr);
                        gap_ssp_confirmation_response(event_addr);
                    }

                    break;

I set my device IO capability to NO_INPUT_NO_OUTPUT, and use an Android phone connect to my device,  the packet log as fellow:

[02:13:02.510] EVT <=(04) 04 0A F6 8B FB 16 8D A0 0C 02 5A 01 
LOG -- event_handler.1882: Connection_incoming: A0:8D:16:FB:8B:F6, type 1
LOG -- create_connection_for_bd_addr_and_type.185: create_connection_for_addr A0:8D:16:FB:8B:F6, type ff
LOG -- hci_run.3146: sending hci_accept_connection_request, remote eSCO 0
[02:13:02.510] CMD =>(01) 09 04 07 F6 8B FB 16 8D A0 01 
[02:13:02.510] EVT <=(04) 6E 00 
[02:13:02.520] EVT <=(04) 0F 04 00 01 09 04 
[02:13:02.540] EVT <=(04) 03 0B 00 0E 00 F6 8B FB 16 8D A0 01 00 
LOG -- event_handler.1906: Connection_complete (status=0) A0:8D:16:FB:8B:F6
LOG -- event_handler.1919: New connection: handle 14, A0:8D:16:FB:8B:F6
LOG -- hci_emit_nr_connections_changed.3763: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[02:13:02.540] EVT <=(04) 61 01 01 
[02:13:02.540] CMD =>(01) 1B 04 02 0E 00 
[02:13:02.540] EVT <=(04) 1B 03 0E 00 05 
[02:13:02.540] ACL <=(02) 0E 20 0A 00 06 00 01 00 0A 02 02 00 02 00 
[02:13:02.540] EVT <=(04) 6E 00 
LOG -- hci_number_free_acl_slots_for_connection_type.386: ACL classic buffers: 0 used of 8
LOG -- hci_number_free_acl_slots_for_connection_type.386: ACL classic buffers: 0 used of 8
LOG -- hci_number_free_acl_slots_for_connection_type.386: ACL classic buffers: 0 used of 8
LOG -- hci_send_acl_packet_fragments.559: hci_send_acl_packet_fragments entered
LOG -- hci_send_acl_packet_fragments.565: hci_send_acl_packet_fragments loop entered
LOG -- hci_send_acl_packet_fragments.590: hci_send_acl_packet_fragments loop before send (more fragments 0)
[02:13:02.540] ACL =>(02) 0E 00 10 00 0C 00 01 00 0B 02 08 00 02 00 00 00 80 02 00 00 
LOG -- hci_send_acl_packet_fragments.608: hci_send_acl_packet_fragments loop after send (more fragments 0)
LOG -- hci_send_acl_packet_fragments.617: hci_send_acl_packet_fragments loop over
[02:13:02.550] EVT <=(04) 0F 04 00 01 1B 04 
[02:13:02.550] EVT <=(04) 0B 0B 00 0E 00 BF FE CF FE DB FF 7B 87 
LOG -- event_handler.1985: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 6, eSCO 1
[02:13:02.550] EVT <=(04) 6E 00 
[02:13:02.560] EVT <=(04) 32 09 F6 8B FB 16 8D A0 01 00 03 
[02:13:02.560] EVT <=(04) 31 06 F6 8B FB 16 8D A0 
LOG -- event_handler.2033: Remote io capability:1

LOG -- event_handler.2034: Remote OOB data:0

LOG -- event_handler.2035: Remote authentication req:3

LOG -- hci_run.3198: IO Capability Request received, stack bondable 1, io cap 3
LOG -- gap_mitm_protection_required_for_security_level.3872: gap_mitm_protection_required_for_security_level 0
[02:13:02.560] CMD =>(01) 2B 04 09 F6 8B FB 16 8D A0 03 00 04 
[... 188 lines cut ...]
[02:13:03.130] EVT <=(04) 6E 00 
[02:13:03.160] EVT <=(04) 13 05 01 0E 00 02 00 
[02:13:03.160] ACL <=(02) 0E 20 0C 00 08 00 01 00 06 09 04 00 41 00 43 00 
LOG -- l2cap_signaling_handler_channel.2328: L2CAP signaling handler code 6, state 12
LOG -- hci_number_free_acl_slots_for_connection_type.386: ACL classic buffers: 0 used of 8
LOG -- hci_number_free_acl_slots_for_connection_type.386: ACL classic buffers: 0 used of 8
LOG -- hci_number_free_acl_slots_for_connection_type.386: ACL classic buffers: 0 used of 8
LOG -- hci_send_acl_packet_fragments.559: hci_send_acl_packet_fragments entered
LOG -- hci_send_acl_packet_fragments.565: hci_send_acl_packet_fragments loop entered
LOG -- hci_send_acl_packet_fragments.590: hci_send_acl_packet_fragments loop before send (more fragments 0)
[02:13:03.160] ACL =>(02) 0E 00 0C 00 08 00 01 00 07 09 04 00 41 00 43 00 
LOG -- hci_send_acl_packet_fragments.608: hci_send_acl_packet_fragments loop after send (more fragments 0)
LOG -- hci_send_acl_packet_fragments.617: hci_send_acl_packet_fragments loop over
LOG -- l2cap_emit_channel_closed.889: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[02:13:03.160] EVT <=(04) 71 02 41 00 
LOG -- l2cap_stop_rtx.995: l2cap_stop_rtx for local cid 0x41
[02:13:03.160] EVT <=(04) 6E 00 
[02:13:03.310] EVT <=(04) 13 05 01 0E 00 01 00 
[02:13:07.070] EVT <=(04) 05 04 00 0E 00 13 
LOG -- hci_emit_discoverable_enabled.3812: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[02:13:07.070] EVT <=(04) 66 01 01 
do_disconnect_evt-369
LOG -- hci_shutdown_connection.827: Connection closed: handle 0xe, A0:8D:16:FB:8B:F6
LOG -- hci_emit_nr_connections_changed.3763: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[02:13:07.070] EVT <=(04) 61 01 00