bluekitchen/btstack #110: Possible to reconnect hid keyboard example after long inactivity?

JuanPotato, 2018-01-28:
Hi, I was wondering if there was a way to reconnect the hid keyboard example after it has disconnected for whatever reason (inactivity, connecting device rebooted)

mringwal, 2018-01-29:
hi. for reconnect, the hid keyboard needs to connect to the remote HID Control and Interrupt L2CAP Channels. This isn't implemented yet, but we will add support for reconnect eventually. What's your project?

JuanPotato, 2018-01-29:
It's really simple, a program that sends two spaces to a phone when a gpio pin is triggered. Two spaces wake up and unlock the phone. And this library was the only once that worked with iOS so it was great. But unfortunately the phone will reboot and having it auto reconnect would be great. I greatly appreciate the work you've done.

mringwal, 2018-01-30:
hi. in this case, there's a simple fix: please use the hog_keyboard instead of hid_keyboard. With HID over LE, the iPhone will automatically reconnect after reboot and such.

JuanPotato, 2018-02-01:
Hi, thanks for the information, I will try this out and tell you how it goes. I had issues with hog before but that was while I was still trying to get anything running so it was probably my fault.

mringwal, 2018-07-31:
fwiw, the hid_device on develop has an create connection function and it's used in the hid_keyboard demo.

closed 2018-07-31 by mringwal