bluekitchen/btstack #111: Is there any event to indicate LE pairing finish?

LuckyHowl, 2018-01-29:
I try to use SM_EVENT_AUTHORIZATION_RESULT event to indicate the LE pairing finish and tell the user the pairing is successful or not. But this event never emit yet.

mringwal, 2018-01-29:
SM_EVENT_AUTHORIZATION_RESULT is supposed to be for authorized access, which is authentication + asking the user once more. 

Anyway, you'd like to get an event for sm pairing complete and that one actually is missing - I've stumbled upon that just last week. We'll add a sm_event_pairing_complete or so event in the next days.

LuckyHowl, 2018-01-30:
Thank you very much!

mringwal, 2018-02-03:
btw. on the develop branch, you'll get a SM_EVENT_PAIRING_COMPLETE, see sm_pairing_peripheral/central.

closed 2018-02-03 by mringwal