bluekitchen/btstack #112: Hi,How can I get the name and device class of the remote device when I am connected to the device?

907038134, 2018-01-31:
How can I get the name and device class of the remote device when I am connected to the device?

907038134, 2018-01-31:
I haven't found the process of pair failure, can you give me directions? thanks!

mringwal, 2018-01-31:
For Bluetooth Classic, you can gap_remote_name_request when connected or not. You may also get the remote name as part of the Extended Inquiry Response (see gap_inquiry example). The Class of Device is only available as part of the inquiry.

For LE, you might get the name as part of the advertisements and/or scan response. After connected, you can use the gatt_client to find the GAP Name Characteristic.

mringwal, 2018-01-31:
This is not a chat. Please try to explain better what you've tried, what you've expected, and what happened. Also, general "how to" might be better on the BTstack  mailing list. http://groups.google.com/group/btstack-dev

closed 2018-01-31 by mringwal