bluekitchen/btstack #119: in ubuntu V14.04 VMware，*** stack smashing detected ***: ./hfp_hf_demo terminated

hgand, 2018-03-19:
1 . running  ./hfp_hf_demo
2 . search in my phone.
3. conneted  HFP HF Demo 00:1A:7D:DA:71:13
3. will appear bellow   error.   

   seems the stack is not enought, but  i had  increase stack to 10M or 100M ,but no usefull.

strange:   when i   do this      
// HAVE_MALLOC    in  Btstack_config.h
my phone will not be  conneted to bluetooth controller.

root@ubuntu:/btstack/btstack/port/libusb# ./hfp_hf_demo
Packet Log: /tmp/hci_dump.pklg
[2018-03-19 02:51:20.942] LOG -- hci_transport_h2_libusb.c.1352: registering packet handler
SCO Demo: Sending sine wave, hexdump received data.
.......
........
[2018-03-19 02:51:46.627] LOG -- hci.c.363: gap_store_link_key_for_bd_addr: 1C:77:F6:88:3D:03, type 4
[2018-03-19 02:51:47.145] ACL <= 47 20 0C 00 08 00 01 00 02 01 04 00 01 00 41 00 
[2018-03-19 02:51:47.145] LOG -- hci.c.761: acl_handler: size 16
[2018-03-19 02:51:47.146] LOG -- hci.c.3993: gap_request_security_level 0, current level 0
[2018-03-19 02:51:47.146] LOG -- hci.c.3891: hci_emit_security_level 0 for handle 47
[2018-03-19 02:51:47.146] EVT <= E0 03 47 00 00 
[2018-03-19 02:51:47.146] LOG -- l2cap.c.2128: l2cap - security level update
[2018-03-19 02:51:47.147] LOG -- l2cap.c.2138: channel state 5: actual 0 >= required 0?
[2018-03-19 02:51:47.147] ACL => 47 00 0A 00 06 00 01 00 0A 01 02 00 02 00 
[2018-03-19 02:51:47.149] EVT <= 6E 00 
[2018-03-19 02:51:47.149] LOG -- hci.c.1780: HCI:EVENT:6e
[2018-03-19 02:51:47.154] EVT <= 13 05 01 47 00 01 00 
[2018-03-19 02:51:47.154] LOG -- hci.c.1780: HCI:EVENT:13
[2018-03-19 02:51:47.161] ACL <= 47 20 10 00 0C 00 01 00 0B 01 08 00 02 00 00 00 BB 02 00 00 
[2018-03-19 02:51:47.166] LOG -- hci.c.761: acl_handler: size 20
[2018-03-19 02:51:47.166] LOG -- l2cap.c.2653: extended features mask 0x2bb
[2018-03-19 02:51:47.166] LOG -- l2cap.c.910: L2CAP_EVENT_INCOMING_CONNECTION addr 1C:77:F6:88:3D:03 handle 0x47 psm 0x1 local_cid 0x40 remote_cid 0x41
[2018-03-19 02:51:47.166] EVT <= 72 0E 03 3D 88 F6 77 1C 47 00 01 00 40 00 41 00 
[2018-03-19 02:51:47.166] LOG -- l2cap.c.2249: L2CAP_ACCEPT_CONNECTION local_cid 0x40
[2018-03-19 02:51:47.166] LOG -- l2cap.c.1416: L2CAP_STATE_WILL_SEND_CONNECTION_RESPONSE_ACCEPT
[2018-03-19 02:51:47.166] ACL => 47 00 10 00 0C 00 01 00 03 01 08 00 40 00 41 00 00 00 00 00 
[2018-03-19 02:51:47.169] EVT <= 6E 00 
[2018-03-19 02:51:47.169] LOG -- hci.c.1780: HCI:EVENT:6e
[2018-03-19 02:51:47.169] ACL => 47 00 1E 00 1A 00 01 00 04 02 16 00 41 00 00 00 04 09 00 00 00 00 00 00 00 00 00 01 02 9B 06 05 01 00 
[2018-03-19 02:51:47.170] LOG -- l2cap.c.1048: l2cap_stop_rtx for local cid 0x40
[2018-03-19 02:51:47.170] LOG -- l2cap.c.1054: l2cap_start_rtx for local cid 0x40
[2018-03-19 02:51:47.170] LOG -- l2cap.c.1058: l2cap_start_rtx for local cid finish

*** stack smashing detected ***: ./hfp_hf_demo terminated
Aborted (core dumped)

mringwal, 2018-03-19:
thanks for reporting. could you post/upload the binary? or let us know what compiler you've used - I didn't see the ** stack smashing detected *** yet.

pansila, 2018-04-02:
I see the issue as well. I'm using Lubuntu 17.10 64bit, libusb port along with a CSR 8510 dongle.

Here is the core dump:

Reading symbols from ./hfp_hf_demo...done.
[New LWP 17420]
[New LWP 17421]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/i386-linux-gnu/libthread_db.so.1".
Core was generated by `./hfp_hf_demo'.
Program terminated with signal SIGABRT, Aborted.
#0  0xb7772c31 in __kernel_vsyscall ()
[Current thread is 1 (Thread 0xb7521700 (LWP 17420))]
(gdb) bt
#0  0xb7772c31 in __kernel_vsyscall ()
#1  0xb75b2ea9 in __GI_raise (sig=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#2  0xb75b4407 in __GI_abort () at abort.c:89
#3  0xb75ee37c in __libc_message (do_abort=1, fmt=0xb76e52c7 "*** %s ***: %s terminated\n")
    at ../sysdeps/posix/libc_fatal.c:175
#4  0xb767e708 in __GI___fortify_fail (msg=<optimized out>) at fortify_fail.c:37
#5  0xb767e698 in __stack_chk_fail () at stack_chk_fail.c:28
#6  0x0805f09d in l2cap_run () at ../../src/l2cap.c:1708
#7  0x0805f96a in l2cap_hci_event_handler (packet_type=4 '\004', cid=0, packet=0xbf8521ca "n", size=2)
    at ../../src/l2cap.c:2025
#8  0x08058b60 in hci_emit_event (event=0xbf8521ca "n", size=2, dump=0) at ../../src/hci.c:3701
#9  0x080563fb in event_handler (packet=0xbf8521ca "n", size=2) at ../../src/hci.c:2365
#10 0x0805654c in packet_handler (packet_type=4 '\004', packet=0xbf8521ca "n", size=2) at ../../src/hci.c:2410
#11 0x0804c74f in handle_completed_transfer (transfer=0x9344b80) at ../../platform/libusb/hci_transport_h2_libusb.c:497
#12 0x0804c818 in usb_process_ds (ds=0x0, callback_type=DATA_SOURCE_CALLBACK_READ)
    at ../../platform/libusb/hci_transport_h2_libusb.c:529
#13 0x0804c887 in usb_process_ts (timer=0x809eb64 <usb_timer>) at ../../platform/libusb/hci_transport_h2_libusb.c:555
#14 0x0804ecc1 in btstack_run_loop_posix_execute () at ../../platform/posix/btstack_run_loop_posix.c:229
#15 0x0804ae8f in btstack_run_loop_execute () at ../../src/btstack_run_loop.c:181
#16 0x08049ce4 in main (argc=1, argv=0xbf852534) at main.c:182

pansila, 2018-04-07:
It seems to be a crash in a BLE code block in the l2cap_run, but can't disable BLE in the btstack_config.h to pass the build. Is there a way to disable BLE entirely? Since CSR 8510 is only a classic bluetooth device.

mringwal, 2018-04-07:
@pansila CSR8510 is a regular Classic + BLE controller (https://www.qualcomm.com/products/csr8510). If you change btstack_config.h, you need to make a make clean first.

mringwal, 2018-04-07:
@pansila I'd be happy to fix this, but I first need to either learn what exact code causes the crash or become able to reproduce it. I've just tried hfp_hf_demo from master on a Debian 9 VM.

- What version is your gcc?
- Can  you upload the packet log from /tmp/hci_dump.pkgl please?
- Could you upload the compiled hfp_hf_demo from your system? (maybe it runs in my VM, too). 
- Did you figure out what line in l2cap_run causes the problem? From the backtrace, that's a 'can send now' event which triggers sending of a packet

mringwal, 2018-04-07:
Nvermind... adding -fstack-protector -fstack-protector-all -fstack-protector-strong also gives me a core dump on connect. I'll have a look!

mringwal, 2018-04-07:
hi! I found the buffer overflow. Sorry, it was already reported in a pull request before, but I've somehow missed it. It's fixed on develop and master. have fun!

closed 2018-04-07 by mringwal