bluekitchen/btstack #120: Proper iOS 11 support?

NSExceptional, 2018-03-20:
Doesn't seem to be work 100% on iOS 11. Whati

mringwal, 2018-03-20:
What doesn't work? BTstack on iOS 11 or Bluetooth Companion and BTC Mouse & Trackpad?

NSExceptional, 2018-03-20:
BTStack.

It looks like there's a [button missing in the top cell](https://i.imgur.com/YHrnJTh.png). The version on Cydia is 1.1-1, is that the latest?

mringwal, 2018-03-20:
The version in Cydia is the last on. The text in the screenshot says that BTstack is enabled automatically (and I remember removing the 'on' button). Could you start some BTstack-enabled app? You could try BTstack GPS Demo to see if a) it works, b) if there's an "OFF" button

NSExceptional, 2018-03-20:
Oh, that makes sense. The missing button threw me off, combined with the Controllers For All developer frivolously claiming that BTstack was "abandoned" on Twitter.

If you say it's updated that's good enough for me! Thanks for responding so fast.

closed 2018-03-20 by NSExceptional

mringwal, 2018-03-20:
Glad it works. FWIW, it hasn't been updated since I think iOS 9. However, no need to update it as long as it works :)

berunous, 2018-05-18:
hello mringwal
can you confirm that the BT stack GPS application is fully operational on IOS 11.1.1. if so how to get it today in view of the difficulties of CYDIA on IOS 11. thank you for an answer

mringwal, 2018-05-18:
@nikolaign BTC Mouse & Trackpad does not work  on iOS 10 or iOS 11 as something in the Bluetooth Server changed.

berunous, 2018-05-19:
> @berunous BTstack itself works fine on iOS 11. However, the BTstack GPS app is currently only compiled for 32-bit, and hence cannot run on iOS 11 (as it's 64-bit only). I'll post here once I get it to compiled (no idea what changed, just re-compiling stuff sometimes becomes tricky)

first of all thank you for the speed of the answer. So I'm waiting for the result of the compilation in 64 bits for IOS 11. my hardware architecture is as follows a bluetooth GPS brand GLOBALSAT model BT 338 An IPAD air model A1474 in WIFI Only and NAVIONIC navigation application and map NAVIONIC HD EUROPE. Can you confirm that to make all this work the iPad will be jealbroken? Can you tell me the best way currently to have CYDIA on the iPad to be able if it works to buy your application. 

----- Mail original -----

berunous, 2018-05-24:
Hello
I want to use BT stack but there are two hard points

where is the 64 bit compilation for IOS 11.1.1
Cydia does not currently allow to buy the application

can you communicate on these two subjects. Thank you

mringwal, 2018-05-25:
@berunous There is not 64-bit version at the moment (I get compile errors), hence you cannot buy it in Cydia on iOS 11. Once it compiles for 64-bit and works on iOS 11, it will be available on Cydia (if Cydia is available).

berunous, 2018-05-27:
I need your application to navigate the wait with impatience. You think to end up under contains time. good luck and good thinking for a flawless compilation.

berunous, 2018-06-02:
hello 

have you completed the compilation of BT-STACK in 64 bits to be compatible IOS 11.1.1? If NO please give us a target date. It is difficult to wait without news of whether the goal will be achieved. and when I'm sure you understand. 

beru Nous

hsiang00, 2018-06-05:
I don't think he can get that done soon. otherwise,this should be done a while ago.

xSimone, 2018-06-05:
@mringwal Hey there Matthias, I sent you an Email in hopes to have BTStack updated for iOS 11.

It's nice to see that's already happening, though may I ask, is it possible we get the old BTStack style back? Namely 0.8-3?

The fact we could select between OFF, iOS and BTStack was nice. Maybe have an Automatic AND Manual Mode? I preferred the Manual Mode since sometimes it'd crash, so I'd just switch it back on again.

iOS4all, 2018-06-06:
We really hope to could use controller for all app running on iOS 11

berunous, 2018-06-06:
how could we account for the number of users of applications relying on BT-STACK and who would like these applications to be compatible with IOS 11 ie that they are compiled in 64 bits. Only the number of users could encourage Matthias Ringwald and if necessary the OPEN SOURCE community to carry out these operations.

mringwal, 2018-06-06:
I'd like to repeat that BTstack (the actual stack with the preferences dialog) is compiled for armv7 & arm64 and runs on iOS 11 as far as I know. It provides basic profiles (SPP, L2CAP, SDP) that are sufficient for most use cases (e.g. GPS, HID devices, ..) 

My BTstack-GPS is 32-bit only, but that should not stop others from using BTstack in their apps. FWIW, it now compiles for 64-bit, but there's no ETA for an official update.

BTC Mouse & Trackpad is a different issue as mentioned earlier as well and independent from BTstack.

vienduong88, 2018-06-06:
Hello Sir, may I ask if you ever plan to update BTC Mouse & Trackpad for ios 11? This is a really useful tweak for ipad especially when ios 11.3 jailbreak might be release soon. Thank you & have a nice day.

berunous, 2018-06-06:
sorry to have irritated you but in the community there are users and develloppers. I am only a lambda user but do not want APPLE to impose the external GPS that he should use on his iPad. This is one of the objectives of OPEN SOURCE. I understand that the base brick BT STACK is operational in 64 bits but that your BT STACK GPS application is currently in 32 bits. I am unable to measure the difficulty of porting from 32 bits to 64 bits but I know that as long as your application BT STACK GPS will not be in 64 bits I could not use my external GPS on IOS 11 future versions. By passing in 64-bit APPLE has killed thousands of applications only develloppers can meet this challenge so that users do not remain under their hegemonic.

xSimone, 2018-06-06:
@mringwal 
> I'd like to repeat that BTstack (the actual stack with the preferences dialog) is compiled for armv7 & arm64 and runs on iOS 11 as far as I know. It provides basic profiles (SPP, L2CAP, SDP) that are sufficient for most use cases (e.g. GPS, HID devices, ..)

So when you say this.. are you saying...
1: PS3 Controllers can directly connect to iOS 11 via BTStack from Cydia
2: ControllersForAll (when updated) can be compatible with BTStack for iOS 11

Thank you so far for replies thus far Matthias. We all love your tweak.

mringwal, 2018-06-06:
> So when you say this.. are you saying...
1: PS3 Controllers (HID?) can directly connect to iOS 11 via BTStack from Cydia
2: ControllersForAll (when updated) can be compatible with BTStack for iOS 11

Well, close. BTstack can be used to either create outgoing HID connections or provide an HID server. BTstack itself doesn't do that. So, ControllersForAll (or even my abandoned Blutrol) should work when updated for iOS 11. If not, and there are actual issues with BTstack itself, I'll have a look.

iOS4all, 2018-06-13:
Unfortunately CFA didn’t update to iOS 11 yet. I test it but unfortunately it’s not working.

xSimone, 2018-06-13:
Yeah I know, I PM'ed Ori, and waiting for a reply..

hsiang00, 2018-06-13:
"BTC Mouse & Trackpad does not work on iOS 10 or iOS 11 as something in the Bluetooth Server changed."

Is the BTC Mouse & Trackpad going to have the update for IOS11? or just discontinue to support?

JoeE60, 2018-07-25:
@mringwal 
Hi Matthias,
is it possible to use a bluetooth mouse with BTstack tweak on a jaibreaked iOS 11 ipad yet?

leokastro, 2018-08-06:
@mringwal Please, review your Blutrol for iOS11, I want to buy it for iOS 11!!!!

ixM4S, 2018-08-21:
@mringwal Hi, and could you help me with this since i have my C4A installed v1.7.4-1 and BTstack v1.1-1 , kept saying [THIS MESSAGE](http://v.ht/C4at) and this is a screenshot of my [btstack](http://v.ht/BTSTACK), what to do?

mringwal, 2018-08-22:
I've spend some time updating BTstack GPS for iOS 11 - which requires 64-bit binaries. It is  basically working but then I've learned that iOS 11 has increased the sandboxing, so BTstack GPS cannot talk to the BTstack Daemon. Right now, I don't know how to bypass this checks (it's outgoing  unix domain sockets as well as setting up a CFMessagePort (I'm already using RocketBootstrap, but the call to setup the port already fails before RocketBootstrap would be called to expose it publicly).

ixM4S, 2018-08-22:
@mringwal so we wait for your kind update? Or what to do?

lechium, 2018-08-27:
@mringwal can you give more detail into the restrictions? Im working on making BTStack and potentially a replacement for controllers for all for tvOS (and potentially for other ios devices as well) for 10.2.2 and 11.x and would like to pre-empt any issues that you're already aware of so i don't spend time discovering the same issues.

mringwal, 2018-08-28:
@lechium in BTstack Daemon, the compiler does not want to directly call system. I assume that system is still there and doing a dlsym lookup would work. Then, all command line tools need the 'platform-application' entitlement. Now, the main issue is how a tweak/app can communicate with the BTstack server. BTstack also supports TCP, so if that works instead of Unix Domain Sockets, that can be considered.  Any ideas/links/documentation in this regard?

lechium, 2018-08-28:
afaik system does still exist on other iOS devices but we've never had it on tvOS, it was removed entirely (afaik) so we've rolled our own version as part of the ground up bootstrap re-created for arm64 support in a project called "the memo" (https://git.nito.tv/NitoTV/the_memo) i've created my own fork where i've replaced that call with RunCmd (our version of system we put together) Im not entirely certain but I had a theory about potential communication issues /tmp/BTStack might present a sandbox issue and maybe something like /Library/.BTStack or something of that nature for the pipe location from the LaunchDaemon is something i was going to try that might rectify potential sandbox issues.

That being said, i'm having issues building this project iOS sample app (WiiMoteOpenGLDemo) where various files complain about bt_register_packet_handler, bt_send_cmd et al being referenced, im pretty sure its just a header issue re the daemon section but keep finding myself in header hell when trying to rectify it. so i havent tried running that sample app yet modified to run for tvOS. Does that sample need to be updated in any way to be compatible with newer changes? Im not adept with bluetooth so some of this is new to me so i apologize if those are dumb questions.

mringwal, 2018-08-30:
Congrats. The WiMote demo is older, the command line tools are a good test though.
How did you start BTdaemon? Did it work when starting it via launchd?

Yes, BTstack requires to shutdown the Apple Bluetooth Server, which is the main drawback. Because of this, I've started the Bluetooth Companion project, that hooks into BTServer and tries to add services in there with BTC Mouse & Trackpad being the main product. My hooks that intercept the communication with the Bluetooth module stopped to work in iOS 10 and I didn't dig deep/long enough to figure out why.

On power on, BTstack unloads BTServer and uses a patched BlueTool to initialize the Bluetooth Controller before taking control.

lechium, 2018-08-30:
thanks! yeh it started successfully through the launchdaemon. i think i traced part of the problem to the springboard access tweak. we dont have springboard on tvos so i would need to adopt that to work with pineboard if i continue to try and get things working via BTStack. 

i had a feeling hooking BTServer would be the route i may be forced to go to get this working on tvOS once i realized i had to dusable BTServer first. is this bt companion project open source? even while its not working it may give me some insight on where to look. ive done very little bluetooth work in the past so this is somewhat new territory to me.

mringwal, 2018-08-31:
Interesting: so daemons started by launchdaemon are accessible - which totally makes sense.
BTstack  uses SpringBoardAccess to toggle the Bluetooth icon in the menu bar, so you might ignore that for now.

BTServer is mainly written in C++ and I wasn't good enough to reverse/hook that. Bluetooth Companion intercepts the Bluetooth communication and uses deep Bluetooth knowledge to modify packets on the fly or inject/drop them. It's closed source and mainly consists of a large Bluetooth state machine to track the state in the BTServer.

Mr-se7enX, 2019-01-28:
@mringwal Could you please update Blutrol to support at least iOS 9 and iOS 10. It's going to be wonderful if you update, because there are many good games don't support MFi controllers like Pubg and fortnite.

muzlini, 2019-02-18:
@mringwal Hey sorry to come up to you like this, im a huge fan of your work, please could you update Blutrol please so it could work on ios 10 and higher... 😞😔😔 the app does install but you cant add any games to it because it does not see any game list whenever you hit the add game (+) symbol, i figured out is the version of ios and the app requires update.. controllers for all doesn't help as you can't keymap your control pads, so Blutrol has and will always be the Best option... Thank you and StayBlessed🙏🏽

xSimone, 2019-02-26:
> @mringwal Could you please update Blutrol to support at least iOS 9 and iOS 10. It's going to be wonderful if you update, because there are many good games don't support MFi controllers like Pubg and fortnite.

Oh gawd, Matthias, even Mr-Seven is asking, he's a huge guy in the Retro Community, too! :<

Come on we need our BTStack / BluTroll! :< <3

xSimone, 2019-02-26:
Oh and, please for iOS 11, mostly iOS 11.3.1 <3

nicostubi, 2019-03-10:
> @nikolaign BTC Mouse & Trackpad does not work on iOS 10 or iOS 11 as something in the Bluetooth Server changed.

Hello, any plans to support BTC Mouse & Trackpad on iOS 12?

1Conan, 2019-05-19:
@mringwal 
Hey.
I was testing your package on cydia on iOS 11 and gave the correct entitlements needed on some binaries.
I've been facing the same issues as @lechiu 
POWER_ON always fails when I was testing `l2cap-server (Works on iOS 10 though)

I already tried stopping all bluetooth related services like `bluetooth` and the unpatched `BlueTool`

`BlueToolH4` cant access the device *I think* and is stuck at when using `device -D`. Works fine on iOS 10.
```
Opening com.apple.uart.bluetooth @ 115200 baud.
```

Has anyone actually got BTstack working on iOS 11?

1Conan, 2019-05-19:
@mringwal 
Hey.
I was testing your package on cydia on iOS 11 and gave the correct entitlements needed on some binaries.
I've been facing the same issues as @lechiu 
POWER_ON always fails when I was testing `l2cap-server (Works on iOS 10 though)

I already tried stopping all bluetooth related services like `bluetooth` and the unpatched `BlueTool`

Has anyone actually got BTstack working on iOS 11?

GuigoBassanesi, 2019-07-29:
@mringwal
Hello. I need the BTC mouse & trackpad for iOS 12.2. do you know when you will do it? I can help financially to create this Tweak.

mringwal, 2019-08-02:
With official mouse support in iOS 13, I don't expect to work on BTC Mouse & Trackpad anymore.

iripa, 2019-08-14:
@mringwal
Thanks for all your wonderful projects and everything you have given to the jailbreak community.
I would love to see support for PS3 controllers on iOS, it would be great for the planet as there are tons of this controllers still around, its a shame this controllers get to be discarded being still an awesome controller.
The new ncontrol has no support for PS3 (not sure why being the ps3 controller also Bluetooth) and your Bluetooth protocol is great if gets implemented again.
Hopefully this project is back. Is awesome to show apple how things should be done. We are buying a very expensive device but most of the time it feels like it was rented🙁.
Long live the jailbreak!!!👍
Fingers crossed on this🤞🤞🤞

1Conan, 2019-08-14:
@iripa BTStack works on ios 11/12. BTStack has nothing to do with controllers as well. 

As for why nControl doesn’t suppot DS3, it uses a weird pairing mechanism. Nito tried to implement it but its more trouble than worth.

iripa, 2019-08-17:
Thanks for your kind answer.

dpmaycry, 2019-09-17:
> With official mouse support in iOS 13, I don't expect to work on BTC Mouse & Trackpad anymore.

But there are some older devices as ipad air on iOS 12.4(jail breakable) witch cannot be upgraded to iOS13, would u kindly consdiering update BTC Mouse & Trackpad for iOS12?

dpmaycry, 2019-09-17:
> With official mouse support in iOS 13, I don't expect to work on BTC Mouse & Trackpad anymore.

Or at least open sources it. I am a iOS app developer but never developed a jail break plugin. Maybe I can try modify your code to fit for iOS12(I hope so...)