bluekitchen/btstack #121: STM32 - CC2564B - Timeout

8lank, 2018-03-21:
Hi,

I followed your blog tutorial [http://bluekitchen-gmbh.com/btstack-port-for-stm32-f4-discovery-board-with-cc256x/](url) to port bluekitchen btstack on a STM32F4 discovery board with a cc256x Bluetooth chipset. Everything was working fine until I started trying the gap inquiry example. The process is blocked after receiving the event 6E 00 and loops infinitely on it:

>[00:00:18.316] CMD => 03 0C 00
[00:00:18.319] EVT <= 6E 00
[00:00:18.517] CMD => 03 0C 00
[00:00:18.520] EVT <= 6E 00
[00:00:18.718] CMD => 03 0C 00
[00:00:18.721] EVT <= 6E 00
[00:00:18.919] CMD => 03 0C 00
[00:00:18.922] EVT <= 6E 00
[00:00:19.120] CMD => 03 0C 00
[00:00:19.123] EVT <= 6E 00
[00:00:19.321] CMD => 03 0C 00
[00:00:19.324] EVT <= 6E 00
[00:00:19.522] CMD => 03 0C 00
[00:00:19.525] EVT <= 6E 00
[00:00:19.723] CMD => 03 0C 00
[00:00:19.726] EVT <= 6E 00
[00:00:19.924] CMD => 03 0C 00`

I also tried to import directly the whole project from your repository [https://github.com/bluekitchen/btstack/tree/master/port/stm32-f4discovery-cc256x](url) but I got the same result. I also tried to update the TI Bluetooth Service Pack with the latest version.

Have you ever encountered such behavior on TI CC256x chipset?

Hw:
- STM32F407G discovery board
- TI CC256xQFN with CC2564b chipset
- CC256XE STADAPT v1.0

Thanks

mringwal, 2018-03-21:
hi. BTstack sends an HCI Reset every 200 ms but doesn't get a response. Please check the UART lines with a logic analyzer to see if a) there is no response from the CC256x or b) there is a response but it is not received by BTtstack's HCI layer.

8lank, 2018-03-21:
There is no response from the CC256x. I tried with another CC256x but doesn't work either.

mringwal, 2018-03-21:
Do you see the HCI Reset (01 03 0c 00)?

please check the wiring. you'll need to pull nShutdown to VCC, connect RTS/CTS ( check if CC256x CTS is low), and also if there's a 32.768 kHz slow clock provided. and/or check TI's E2E forum.

8lank, 2018-03-22:
Found the issue: the st adapter was non-functioning and the cc256x not well powered. 
Thanks for your help.

mringwal, 2018-03-22:
Glad you found the root cause

closed 2018-03-22 by mringwal