bluekitchen/btstack #122: hello, why can't changed LINK_SLIP_TX_CHUNK_LEN in hci_transport_h5.c  ?

907038134, 2018-03-27:
Hello, 
when I was in the test, send CSR bluetooth pskey data of more than 64 bytes, but in the h5 LINK_SLIP_TX_CHUNK_LEN limit only in 64, when I increase it to 150 (or greater than 64), there will be a host to send data to the bluetooth chip, then the bluetooth chip does not respond, and then the host has been resend. If you don't increase LINK_SLIP_TX_CHUNK_LEN, these problems will not occur, but you can't get to PSK where I need more data. Do you know what the reason is? Or what was the design in the program at that time? 
thank you！

907038134, 2018-03-27:
LOG -- btkit_hfp.c.71: Btstack init

LOG -- l2cap.c.3262: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535

LOG -- rfcomm.c.2309: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10

LOG -- l2cap.c.3262: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535

LOG -- l2cap.c.3262: L2CAP_REGISTER_SERVICE psm 0x19 mtu 65535

LOG -- l2cap.c.3262: L2CAP_REGISTER_SERVICE psm 0x19 mtu 65535

LOG -- l2cap.c.3267: l2cap_register_service: PSM 25 already registered

LOG -- l2cap.c.3262: L2CAP_REGISTER_SERVICE psm 0x17 mtu 65535

LOG -- hci.c.3745: hci_power_control: 1, current mode 0

LOG -- hci_transport_h5.c.917: enable even parity for BCSP mode

LOG -- hci_transport_h5.c.931: UART driver does not provide compatible sleep mode

LOG -- hci_transport_h5.c.821: resend timeout for 115200 baud: 405 ms

LOG -- hci_transport_h5.c.353: link send sync

LOG -- hci_transport_h5.c.347: hci_transport_link_send_control: size 4, append dic 0

LOG -- btstack_util.c.225: 0xDA, 0xDC, 0xED, 0xED, 

 0xc0 0x00 0x41 0x00 0xbe 0xda 0xdc 0xed 0xed 0xc0
2
LOG -- hci_transport_h5.c.284: slip: send 10 bytes

LOG -- hci.c.3754: hci_power_control_on

LOG -- hci.c.5344: BTSTACK_EVENT_STATE 1

EVT <= 60 01 01 
LOG -- hfp.c.554: HFP packet_handler type 4, event type 60, size 3

LOG -- ../bluetooth/btkit/btkit_common.c.519: local_info--------------------------start

LOG -- ../bluetooth/btkit/btkit_common.c.520: Power:          = 0

LOG -- ../bluetooth/btkit/btkit_common.c.521: Discoverable:   = 0

LOG -- ../bluetooth/btkit/btkit_common.c.522: Connected:      = 0

LOG -- ../bluetooth/btkit/btkit_common.c.523: Connecting:     = 0

LOG -- ../bluetooth/btkit/btkit_common.c.525: ------------------

LOG -- ../bluetooth/btkit/btkit_common.c.526: acl_handle      = 0

LOG -- ../bluetooth/btkit/btkit_common.c.527: sco_handle      = 0

LOG -- ../bluetooth/btkit/btkit_common.c.528: avrcp_cid       = 0

LOG -- ../bluetooth/btkit/btkit_common.c.529: a2dp_cid        = 0

LOG -- ../bluetooth/btkit/btkit_common.c.530: local_seid      = 1

LOG -- ../bluetooth/btkit/btkit_common.c.531: avrcp_connected = 0

LOG -- ../bluetooth/btkit/btkit_common.c.532: local_info--------------------------end

LOG -- btkit_hfp.c.823: BTstack: execute run loop

LOG -- btstack_run_loop_tirtos.c.191: RL: execute
[... 1373 lines cut ...]
LOG -- btstack_util.c.219: 0x19, 0x00, 0x00, 0x00, 0x03, 0x00, 0x75, 0xFE, 0x00, 0xEB, 0x14, 0xC1, 0x18, 0x02, 0x2B, 0xFF, 

LOG -- btstack_util.c.219: 0x0E, 0xFF, 0x00, 0x67, 0x18, 0xE7, 0x9E, 0x00, 0x00, 0xEA, 0x14, 0x67, 0x18, 0x02, 0x2B, 0xFF, 

LOG -- btstack_util.c.219: 0x0E, 0xFF, 0x00, 0x67, 0x18, 0xE7, 0x9E, 0x00, 0x18, 0x03, 0x2B, 0xFF, 0x0E, 0xFF, 0x00, 0xFE, 

LOG -- btstack_util.c.225: 0x18, 0x7A, 0xE2, 0x00, 0x6F, 0x0C, 

 0xc0 0xe3 0x65 0x04 0xb3 0x00 0xfc 0x43 0xc2 0x02 0x00 0x21 0x00 0x00 0x00 0x03 0x70 0x00 0x00 0x31 0x21 0x19 0x00 0x00 0x00 0x03 0x00 0x75 0xfe 0x00 0xeb 0x14 0xc1 0x18 0x02 0x2b 0xff 0x0e 0xff 0x00 0x67 0x18 0xe7 0x9e 0x00 0x00 0xea 0x14 0x67 0x18 0x02 0x2b 0xff 0x0e 0xff 0x00 0x67 0x18 0xe7 0x9e 0x00 0x18 0x03 0x2b 0xff 0x0e 0xff 0x00 0xfe 0x18 0x7a 0xe2 0x00 0x6f 0x0c 0x1b 0xf2 0xc0
2
LOG -- hci_transport_h5.c.284: slip: send 78 bytes

LOG -- hci_transport_h5.c.405: hci_transport_link_send_queued_packet: seq 3, ack 4, size 70. Append dic 1, dic = 0x1bf2

LOG -- btstack_util.c.219: 0x00, 0xFC, 0x43, 0xC2, 0x02, 0x00, 0x21, 0x00, 0x00, 0x00, 0x03, 0x70, 0x00, 0x00, 0x31, 0x21, 

LOG -- btstack_util.c.219: 0x19, 0x00, 0x00, 0x00, 0x03, 0x00, 0x75, 0xFE, 0x00, 0xEB, 0x14, 0xC1, 0x18, 0x02, 0x2B, 0xFF, 

LOG -- btstack_util.c.219: 0x0E, 0xFF, 0x00, 0x67, 0x18, 0xE7, 0x9E, 0x00, 0x00, 0xEA, 0x14, 0x67, 0x18, 0x02, 0x2B, 0xFF, 

LOG -- btstack_util.c.219: 0x0E, 0xFF, 0x00, 0x67, 0x18, 0xE7, 0x9E, 0x00, 0x18, 0x03, 0x2B, 0xFF, 0x0E, 0xFF, 0x00, 0xFE, 

LOG -- btstack_util.c.225: 0x18, 0x7A, 0xE2, 0x00, 0x6F, 0x0C, 

 0xc0 0xe3 0x65 0x04 0xb3 0x00 0xfc 0x43 0xc2 0x02 0x00 0x21 0x00 0x00 0x00 0x03 0x70 0x00 0x00 0x31 0x21 0x19 0x00 0x00 0x00 0x03 0x00 0x75 0xfe 0x00 0xeb 0x14 0xc1 0x18 0x02 0x2b 0xff 0x0e 0xff 0x00 0x67 0x18 0xe7 0x9e 0x00 0x00 0xea 0x14 0x67 0x18 0x02 0x2b 0xff 0x0e 0xff 0x00 0x67 0x18 0xe7 0x9e 0x00 0x18 0x03 0x2b 0xff 0x0e 0xff 0x00 0xfe 0x18 0x7a 0xe2 0x00 0x6f 0x0c 0x1b 0xf2 0xc0

907038134, 2018-03-27:
look ,host alway resent pskey 0x2131, i can't get bt response

mringwal, 2018-03-28:
hi. The SLIP chunk len is not relevant for the size of the HCI packets. there's no need to increase it.

what's the status here? What did you try, what do you expect?

907038134, 2018-03-28:
hi, Thank you for your reply.
now i know that slip chunk len it support more than 64 bytes , So I haven't changed the size of this buffer right now.

Now I put the h5 modify BCSP protocol way connection is established, and it works), but there is a problem, is I want to send data to the bluetooth chip, often in the sending of this function judgment here can't send now, because the other package hci_packet is not NULL, can you give me a hint I should how to solve this problem? Any good Suggestions?

thank you！

Static int hci_transport_h5_send_packet(uint8_t packet_type, uint8_t *packet, int size){
If (! Hci_transport_h5_can_send_packet_now (packet_type)) {
Log_error ("hci_transport_h5_send_packet called but in state %d", link_state);
The return - 1;
}
...
...
}

mringwal, 2018-03-28:
Could you explain what you want to do?
You can only send a packet (e.g. HCI Command) when the HCI Transport is ready. You can check with hci_can_send_packet_now(..) if you can send an HCI command. See http://bluekitchen-gmbh.com/btstack/protocols/#hci-host-controller-interface // Sending HCI command based on a template

907038134, 2018-03-29:
hi，I'm setting the CSR chip WBS process, at the time of chip send sco complete event to the host, I need to send commands to the bluetooth chip, but send often appear can't send out, as described above, the current transmission

mringwal, 2018-03-29:
Did you manage to route SCO over HCI?

To send HCI Commands directly, you can add a 'app_run' function which is called on every HCI Event. In the function, you first check if you can send a HCI command (hci_can_send_packet_now) and if yes, you can send your next command. This is necessary as there are no queues for commands - instead the app/the stack needs to serialize the packets it wants to send resp. generate it when they can be sent.

907038134, 2018-03-30:
Thank you for your reply,！
Now，I make the SCO (WBS/NBS) routed to the PCM/I2S, I have finished this process,In this process WBS can now stable support. 
In the NBS process will send commands to the bluetooth of probabilistic failure, as stated above.
I will continue to debug based on the method you told me yesterday. Thank you very much for the method.

mringwal, 2018-04-12:
Sounds good. If you ever manage to configure CSR for SCO over HCI, please let me know. When I tried, it didn't work, although there must be a way to do it :)

closed 2018-04-12 by mringwal