bluekitchen/btstack #123: avrcp_controller_test can't connect

jiych, 2018-04-03:
hi,
I'm using btstack/test/avrcp/avrcp_controller_test on linux to connect my nexus 6p.
Howrver, when I execute command 'c', logcat of nexus 6p give me info like below:
`01-08 23:31:17.342 15713 15791 D bt_btif : Remote device:00:1a:7d:da:71:13, no link key or ble key found`
log of avrcp_controller_test give me info:
```
 - Create AVRCP connection to addr AC:CF:85:CA:C2:A3.
AVRCP Connection failed: status 0x82
AVRCP: HCI_EVENT_DISCONNECTION_COMPLETE
```

python dump_pklg.py /tmp/hci_dump.pklg

`[2018-04-03 09:07:42.516] LOG hci_transport_h2_libusb.c.1352: registering packet handler
[2018-04-03 09:07:42.516] LOG l2cap.c.3387: L2CAP_REGISTER_SERVICE psm 0x17 mtu 65535
[2018-04-03 09:07:42.516] LOG l2cap.c.3387: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
[2018-04-03 09:07:42.516] LOG hci.c.2750: hci_power_control: 1, current mode 0
[2018-04-03 09:07:42.517] LOG hci_transport_h2_libusb.c.956: Scanning for USB Bluetooth device
[2018-04-03 09:07:42.517] LOG hci_transport_h2_libusb.c.667: 1d6b:0002 (bus 1, device 1) - class 9 subclass 0 protocol 0 
[2018-04-03 09:07:42.517] LOG hci_transport_h2_libusb.c.667: 0a12:0001 (bus 2, device 18) - class e0 subclass 1 protocol 1 
[2018-04-03 09:07:42.517] LOG hci_transport_h2_libusb.c.997: USB Bluetooth device found, index 1
[2018-04-03 09:07:42.517] LOG hci_transport_h2_libusb.c.777: libusb open 0, handle 0x55be8a7250f0
[2018-04-03 09:07:43.086] LOG hci_transport_h2_libusb.c.722: libusb_detach_kernel_driver
[2018-04-03 09:07:43.086] LOG hci_transport_h2_libusb.c.726: setting configuration 1...
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.738: claiming interface 0...
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.604: active configuration has 2 interfaces
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.610: interface 0: 3 endpoints
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.615: - endpoint 81, attributes 3
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.621: -> using 0x81 for HCI Events
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.615: - endpoint 2, attributes 2
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.631: -> using 0x02 for ACL Data Out
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.615: - endpoint 82, attributes 2
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.627: -> using 0x82 for ACL Data In
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.610: interface 1: 2 endpoints
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.615: - endpoint 3, attributes 1
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.642: -> using 0x03 for SCO Data Out
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.615: - endpoint 83, attributes 1
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.638: -> using 0x83 for SCO Data In
[2018-04-03 09:07:43.090] LOG hci_transport_h2_libusb.c.1098: Async using timers:
[2018-04-03 09:07:43.090] LOG hci.c.3797: BTSTACK_EVENT_STATE 1
[2018-04-03 09:07:43.090] EVT <= 60 01 01 
[2018-04-03 09:07:43.090] CMD => 03 0c 00 
[2018-04-03 09:07:43.094] EVT <= 0f 04 00 01 06 04 
[2018-04-03 09:07:43.094] EVT <= 6e 00 
[2018-04-03 09:07:43.095] EVT <= 05 04 00 47 00 16 
[2018-04-03 09:07:43.205] EVT <= 0e 04 01 03 0c 00 
[2018-04-03 09:07:43.205] CMD => 01 10 00 
[2018-04-03 09:07:43.216] EVT <= 0e 0c 01 01 10 00 06 bb 22 06 0a 00 bb 22 
[2018-04-03 09:07:43.216] LOG hci.c.1878: Manufacturer: 0x000a
[2018-04-03 09:07:43.217] EVT <= 6e 00 
[2018-04-03 09:07:43.217] CMD => 14 0c 00 
[2018-04-03 09:07:43.224] EVT <= 6e 00 
[2018-04-03 09:07:43.239] EVT <= 0e fc 01 14 0c 00 43 53 52 38 35 31 30 20 41 31 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[2018-04-03 09:07:43.239] LOG hci.c.1792: local name: CSR8510 A10
[2018-04-03 09:07:43.239] LOG hci.c.1584: Received local name, need baud change 0
[2018-04-03 09:07:43.239] CMD => 02 10 00 
[2018-04-03 09:07:43.242] EVT <= 6e 00 
[2018-04-03 09:07:43.245] EVT <= 0e 44 01 02 10 00 ff ff ff 03 fe ff ff ff ff ff ff ff f3 0f e8 fe 3f f7 83 ff 1c 00 00 00 61 f7 ff ff 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[2018-04-03 09:07:43.245] LOG hci.c.1888: Local supported commands summary 0x07
[2018-04-03 09:07:43.245] CMD => 09 10 00 
[2018-04-03 09:07:43.249] EVT <= 0e 0a 01 09 10 00 13 71 da 7d 1a 00 
[2018-04-03 09:07:43.249] LOG hci.c.1840: Local Address, Status: 0x00: Addr: 00:1A:7D:DA:71:13
[2018-04-03 09:07:43.249] EVT <= 6e 00 
[2018-04-03 09:07:43.249] CMD => 05 10 00 
[2018-04-03 09:07:43.254] EVT <= 0e 0b 01 05 10 00 36 01 40 0a 00 08 00 
[2018-04-03 09:07:43.254] LOG hci.c.1809: hci_read_buffer_size: ACL size module 310 -> used 310, count 10 / SCO size 64, count 8
[2018-04-03 09:07:43.254] EVT <= 6e 00 
[2018-04-03 09:07:43.254] CMD => 03 10 00 
[2018-04-03 09:07:43.258] EVT <= 0e 0c 01 03 10 00 ff ff 8f fe db ff 5b 87 
[2018-04-03 09:07:43.258] LOG hci.c.1867: Packet types cc18, eSCO 1
[... 109 lines cut ...]
[2018-04-03 09:07:47.083] LOG l2cap.c.1048: l2cap_stop_rtx for local cid 0x40
[2018-04-03 09:07:47.083] LOG l2cap.c.2378: l2cap_signaling_handle_configure_response
[2018-04-03 09:07:47.083] LOG l2cap.c.882: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr AC:CF:85:CA:C2:A3 handle 0x48 psm 0x1 local_cid 0x40 remote_cid 0x41 local_mtu 1020, remote_mtu 672, flush_timeout 0
[2018-04-03 09:07:47.083] LOG l2cap.c.884: ERTM mode 0, fcs enabled 1
[2018-04-03 09:07:47.083] EVT <= 70 16 00 a3 c2 ca 85 cf ac 48 00 01 00 40 00 41 00 fc 03 a0 02 00 00 00 
[2018-04-03 09:07:47.083] EVT <= 78 02 40 00 
[2018-04-03 09:07:47.083] ACL => 48 00 18 00 14 00 41 00 06 00 01 00 0f 35 03 19 00 17 03 fc 35 05 0a 00 01 ff ff 00 
[2018-04-03 09:07:47.083] EVT <= 13 05 01 48 00 01 00 
[2018-04-03 09:07:47.085] EVT <= 6e 00 
[2018-04-03 09:07:47.088] EVT <= 13 05 01 48 00 01 00 
[2018-04-03 09:07:47.094] ACL <= 48 20 b9 00 b5 00 40 00 07 00 01 00 b0 00 ad 35 ab 36 00 6d 09 00 01 35 03 19 11 0c 09 00 04 35 10 35 06 19 01 00 09 00 17 35 06 19 00 17 09 01 04 09 00 05 35 03 19 10 02 09 00 09 35 08 35 06 19 11 0e 09 01 04 09 00 0d 35 12 35 10 35 06 19 01 00 09 00 1b 35 06 19 00 17 09 01 04 09 01 00 25 19 41 56 20 52 65 6d 6f 74 65 20 43 6f 6e 74 72 6f 6c 20 54 61 72 67 65 74 00 09 03 11 09 00 c1 36 00 38 09 00 01 35 03 19 11 0e 09 00 04 35 10 35 06 19 01 00 09 00 17 35 06 19 00 17 09 01 04 09 00 05 35 03 19 10 02 09 00 09 35 08 35 06 19 11 0e 09 01 03 09 03 11 09 00 02 00 
[2018-04-03 09:07:47.094] LOG l2cap.c.1842: L2CAP_DISCONNECT local_cid 0x40 reason 0x0
[2018-04-03 09:07:47.094] ACL => 48 00 0c 00 08 00 01 00 06 04 04 00 41 00 40 00 
[2018-04-03 09:07:47.097] EVT <= 6e 00 
[2018-04-03 09:07:47.101] EVT <= 13 05 01 48 00 01 00 
[2018-04-03 09:07:47.106] ACL <= 48 20 0c 00 08 00 01 00 07 04 04 00 41 00 40 00 
[2018-04-03 09:07:47.106] LOG l2cap.c.2446: L2CAP signaling handler code 7, state 13
[2018-04-03 09:07:47.106] LOG l2cap.c.904: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x40
[2018-04-03 09:07:47.106] EVT <= 71 02 40 00 
[2018-04-03 09:07:47.106] LOG sdp_client.c.472: SDP Client disconnected.
[2018-04-03 09:07:47.106] LOG l2cap.c.1048: l2cap_stop_rtx for local cid 0x40
[2018-04-03 09:07:51.335] EVT <= 05 04 00 48 00 13 
[2018-04-03 09:07:51.336] LOG hci.c.868: Connection closed: handle 0x48, AC:CF:85:CA:C2:A3
[2018-04-03 09:07:51.336] LOG hci.c.3860: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[2018-04-03 09:07:51.336] EVT <= 61 01 00 `

mringwal, 2018-04-03:
hi. could you attach /tmp/hci_dump.pklg directly? (you may need to zip it first). It can be opened with Wireshark..

jiych, 2018-04-03:
[hci_dump.pklg.zip](https://github.com/bluekitchen/btstack/files/1870807/hci_dump.pklg.zip)
hi, 
sorry for that.
Have uploaded hci_dump.pklg.

mringwal, 2018-04-03:
no problem. 

ok. in the log, the connection is created and BTstack does an SDP query, where an AVRCP Target as well as an AVRCP Control is listed. It should then connect to the AVRCP Target, but doesn't. We're actively working on AVRCP to get it qualified.

Could you try with the develop branch and/or feb-a2dp branch?

jiych, 2018-04-04:
ok, I will try it and give results after i finished.

jiych, 2018-04-04:
Hi, I tried both develop branch and feb-a2dp branch.
Both of them failed. Does this means we haven't finished AVRCP?
I attached hci_dump_dev.pklg and hci_dump_feb.pklg for your reference.
[hci_dump_dev.pklg.zip](https://github.com/bluekitchen/btstack/files/1874119/hci_dump_dev.pklg.zip)
[hci_dump_feb.pklg.zip](https://github.com/bluekitchen/btstack/files/1874120/hci_dump_feb.pklg.zip)

wusi524888, 2022-03-31:
HI. the problem has been solved? and how to fix it ?

mringwal, 2022-03-31:
Hi @wusi524888 . This is an old ticket. AVRCP has been fully qualified (again) and tested. The code in test/ is for internal testing and not intended to be used directly. The a2dp_sink/source_demo examples also include support for AVRCP control (e.g. media control or volume), please try these. I'll close this ticket. If you have problems with the official examples please create a new ticket here.

closed 2022-03-31 by mringwal