bluekitchen/btstack #126: le_streamer.c missing le_streamer.h

kelsayed, 2018-04-18:
In the examples folder the le_streamer.c (in main branch)
#include "le_streamer.h"
the file does not exist. 
Including "spp_and_le_streamer.h" solves the problem partially but then 
ATT_CHARACTERISTIC_0000FF12_0000_1000_8000_00805F9B34FB_01_VALUE_HANDLE
ATT_CHARACTERISTIC_0000FF12_0000_1000_8000_00805F9B34FB_01_CLIENT_CONFIGURATION_HANDLE
are undefined there
Also, the examples/Makefile.inc has
le_streamer: le_streamer.h ${CORE_OBJ} ${COMMON_OBJ} ${ATT_OBJ} ${GATT_SERVER_OBJ} ${SM_OBJ} le_streamer.c
	${CC} $(filter-out le_streamer.h,$^) ${CFLAGS} ${LDFLAGS} -o $@
So, the file needs to be added to the source tree in order to get le_streamer to compile.

mringwal, 2018-04-18:
hi. le_streamer.h is dynamically  generated from example/le_streamer.gatt by tool/compile_gatt.py as part of the build process .

What port do you try to compile?

kelsayed, 2018-04-18:
OK, great thanks for the quick response. I am trying on libusb

mringwal, 2018-04-18:
On libusb - our main development target - the le_streamer.h will be generated when your run make in port/libusb.

closed 2018-04-18 by mringwal

mringwal, 2018-04-18:
not really. That seems to be a left over from porting - before the create_examples.py was ready. I'll make a note to remove it.

kelsayed, 2018-04-18:
OK, cool. Thnx again.