bluekitchen/btstack #129: 2 SCO channels opened, 16bit, CVSD need 99 packet size

skoperst, 2018-05-15:
https://github.com/bluekitchen/btstack/blob/62abb699839330d8f329e6c9ff2d317460d5555a/src/hci.c#L4677-L4681

I tested on CSR & BCM LM506 dongles. Not sure if its part of the spec or not.

skoperst, 2018-05-15:
Adding this causes the audio to play nicely:
`if (hci_stack->sco_voice_setting & 0x0060)
		if (hci_number_sco_connections() == 2)
			return 99;`

mringwal, 2018-05-15:
is this for USB dongles? is the LM506 an BCM20702.
You're correct that more channels need different sco packet lengths.

mringwal, 2018-05-15:
Btw. are you working on a walkie-talkie app for 2 headsets? (best idea I can come up using 2 SCO connections).

skoperst, 2018-05-15:
It will double the packet size for 8-bit CVSD too, and MSBC too.. so I think its safe to  make the function consider the number of sco connections.

The LM506 is BCM20702A:
```
[16844.485453] usb 2-1.3: Product: BCM20702A0
[16844.485456] usb 2-1.3: Manufacturer: Broadcom Corp
[16844.485459] usb 2-1.3: SerialNumber: 5CF37085D233
[16844.599681] Bluetooth: hci0: BCM: chip id 63
[16844.615693] Bluetooth: hci0: BCM20702A
[16844.616679] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[16845.486913] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1338
[16845.503861] Bluetooth: hci0: Broadcom Bluetooth Device
```

I'm working on a voice bridge, (sort of a voice repeater) mechanism that you can push voice between one or more BT devices with reasonable latency.
A walkie talkie can be easily built with that.

skoperst, 2018-05-16:
Tried 2&1 connections of 8,16bit and transparent. works nicely.

commit cf119f3b (2018-05-15): hci: calculate sco_packet_length based on number of concurrent sco connections - fixes #129
commit a3069afe (2018-05-16): hci: calculate sco_packet_length based on number of concurrent sco connections - fixes #129
closed 2018-05-16 by mringwal