bluekitchen/btstack #133: le activate discoverable with cc2564 moda chip

Lefix2, 2018-05-31:
Hi,

Today i wanted to clean up some code by undefining the CLASSIC macro.
My code using only BLE i tought it was a good idea.
But, when doing that, gap_discoverable_control() can't be called anymore (defined for classic bluetooth) and my device can't be discovered.

Here is my simple setup:
	```
// register for HCI events
	hci_event_callback_registration.callback = &packet_handler;
	hci_add_event_handler(&hci_event_callback_registration);

	l2cap_init();

	le_device_db_init();

	sm_init();

	// setup information service
	information_service_setup();

	battery_service_server_init(battery);

	// setup timer
	one_shot_timer_setup();

	att_server_init(profile_data, att_read_callback, att_write_callback);
	att_server_register_packet_handler(packet_handler);

	// setup advertisements
	uint16_t adv_int_min = 0x0030;
	uint16_t adv_int_max = 0x0030;
	uint8_t adv_type = 0;
	bd_addr_t null_addr;
	memset(null_addr, 0, 6);
	gap_advertisements_set_params(adv_int_min, adv_int_max, adv_type, 0, null_addr, 0x07, 0x00);
	gap_advertisements_set_data(adv_data_len, (uint8_t*) adv_data);
	gap_advertisements_enable(1);

	gap_discoverable_control(1);

	return 0;
```

mringwal, 2018-05-31:
gap_discoverable_control is only for Classic. Please verify if you can remove it with ENABLE_CLASSIC defined and still can connect via LE.

Lefix2, 2018-05-31:
Indeed it's working, so i made a mistake in my configuration?

mringwal, 2018-05-31:
Not sure. Now, remove ENABLE_CLASSIC and do a clean build.

Lefix2, 2018-05-31:
In fact my app is crashing in the freertos btstack mainloop, i got this trace:
```
[00:00:11.502] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:11.510] LOG -- btstack_run_loop_freertos.c.210: RL: now 11510, expires 11496
[00:00:11.518] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8007291
[00:00:11.525] LOG -- btstack_run_loop_freertos.c.210: RL: now 11525, expires 11536
[00:00:11.533] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:00:11.540] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:11.548] LOG -- btstack_run_loop_freertos.c.210: RL: now 11548, expires 11536

``` 
I tried to log a random string after this instruction, but it is never reached when the code freeze.
perhaps a segmentation fault but i wonder why it works with classic mode...

mringwal, 2018-06-01:
please disable ENABLE_LOG_DEBUG and make sure that ENABLE_LOG_INFO and ENABLE_LOG_ERROR are on. Please also enable packet logger.

how does your target look like?

In general, there are a few ports that are LE only (samv71-atwilc3000, all em9304, ..). You may try to use the btstack_config.h from one of them to test, too.

Lefix2, 2018-06-01:
```
[00:03:30.524] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x20003148
[00:03:30.532] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue item size 8
[00:03:30.540] LOG -- btstack_chipset_cc256x.c.173: cc256x: using default init script
[00:03:30.548] LOG -- btstack_tlv_flash_bank.c.364: found bank -1
[00:03:30.555] LOG -- btstack_tlv_flash_bank.c.183: test erased: bank 0, offset 0
[00:03:30.563] LOG -- btstack_tlv_flash_bank.c.192: not erased 0 - 10
[00:03:30.569] LOG -- btstack_tlv_flash_bank.c.207: bank 0 not empty, erase bank
[00:03:30.577] LOG -- btstack_tlv_flash_bank.c.408: write offset 8
[00:03:30.584] LOG -- btstack_tlv.c.48: TLV Instance 0x8018a1c
[00:03:30.590] LOG -- le_device_db_tlv.c.157: num valid le device entries 0
[00:03:30.597] LOG -- hci.c.2750: hci_power_control: 1, current mode 0
[00:03:30.604] LOG -- btstack_chipset_cc256x.c.173: cc256x: using default init script
[00:03:31.392] LOG -- hci.c.3797: BTSTACK_EVENT_STATE 1
[00:03:31.397] EVT <= 60 01 01 
                               [00:03:31.400] LOG -- hci.c.1143: hci_initializing_run: substate 0, can s1
[00:03:31.408] CMD => 03 0C 00 
                               [00:03:31.411] LOG -- btstack_run_loop_freertos.c.180: RL: execute
[00:03:31.417] EVT <= 6E 00 
                            [00:03:31.421] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4295
[00:03:31.428] LOG -- btstack_run_loop_freertos.c.210: RL: now 211428, expires 210608
[00:03:31.437] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.444] LOG -- btstack_run_loop_freertos.c.210: RL: now 211444, expires 211455
[00:03:31.453] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:03:31.460] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:03:31.468] LOG -- btstack_run_loop_freertos.c.210: RL: now 211468, expires 211455
[00:03:31.477] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.484] LOG -- btstack_run_loop_freertos.c.210: RL: now 211484, expires 211495
[00:03:31.492] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:03:31.500] EVT <= 0E 04 01 03 0C 00 
                                        [00:03:31.504] LOG -- hci.c.1464: Command complete for expected 1
[00:03:31.513] LOG -- hci.c.1143: hci_initializing_run: substate 2, can send 1
[00:03:31.520] CMD => 01 10 00 
                               [00:03:31.524] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 5
[00:03:31.532] LOG -- btstack_run_loop_freertos.c.210: RL: now 211532, expires 211495
[00:03:31.540] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.548] LOG -- btstack_run_loop_freertos.c.210: RL: now 211548, expires 211559
[00:03:31.556] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:03:31.564] EVT <= 6E 00 
                            [00:03:31.567] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4295
[00:03:31.575] LOG -- btstack_run_loop_freertos.c.210: RL: now 211575, expires 211559
[00:03:31.583] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.590] LOG -- btstack_run_loop_freertos.c.210: RL: now 211590, expires 211601
[00:03:31.599] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:03:31.606] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:03:31.614] LOG -- btstack_run_loop_freertos.c.210: RL: now 211614, expires 211601
[00:03:31.623] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.630] LOG -- btstack_run_loop_freertos.c.210: RL: now 211630, expires 211641
[00:03:31.638] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:03:31.646] LOG -- hci_transport_h4.c.239: Detected CC256x controller
[00:03:31.653] EVT <= 0E 0C 01 01 10 00 06 00 00 06 0D 00 90 1B 
                                                                [00:03:31.660] LOG -- hci.c.1878: Manufad
[00:03:31.665] LOG -- hci.c.1464: Command complete for expected opcode 1001 at substate 3
[00:03:31.674] LOG -- hci.c.1143: hci_initializing_run: substate 4, can send 1
[00:03:31.681] CMD => 14 0C 00 
                               [00:03:31.685] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 5
[00:03:31.693] LOG -- btstack_run_loop_freertos.c.210: RL: now 211693, expires 211641
[00:03:31.701] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.709] LOG -- btstack_run_loop_freertos.c.210: RL: now 211709, expires 211719
[00:03:31.717] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 10
[00:03:31.724] EVT <= 6E 00 
                            [00:03:31.727] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4295
[00:03:31.735] LOG -- btstack_run_loop_freertos.c.210: RL: now 211735, expires 211719
[00:03:31.744] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x8006f49
[00:03:31.751] LOG -- btstack_run_loop_freertos.c.210: RL: now 211751, expires 211762
[00:03:31.760] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:03:31.767] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:03:31.775] LOG -- btstack_run_loop_freertos.c.210: RL: now 211775, expires 211762
                                                                                     OneKey booting
...free heap: 1368
[... 47 lines cut ...]
[00:00:10.805] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[... cut ...]
[00:00:10.917] LOG -- btstack_run_loop_freertos.c.210: RL: now 10917, expires 10928
[00:00:10.925] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 11
[00:00:10.933] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
[00:00:10.941] LOG -- btstack_run_loop_freertos.c.210: RL: now 10941, expires 10928

```

There are logs, i didn't manage to get my code work with others configurations.
I'm trying to get backtrace of execution to know if the error comes from an other task!

mringwal, 2018-07-20:
Hi. Thanks for reporting this. In theory, the #defines in hci.h should set correct minimal sizes for the HCI buffers. E.g. HCI_EVENT_BUFFER_SIZE is set to 258. However, HCI_EVENT_BUFFER_SIZE is not used for the calculation of the HCI_PACKET_BUFFER_SIZE (which is used for the incoming buffer).

I'm a bit irritated as this issue should have come up with every port that uses the small HCI_ACL_PAYLOAD_SIZE. 

For now, please use HCI_ACL_PAYLOAD_SIZE = 256 to make the incoming buffer large enough.

The log_error is a good idea, too, I'll probably add that as well.

mringwal, 2018-07-23:
Fixed on develop.

closed 2018-07-23 by mringwal