bluekitchen/btstack #138: (ROLE_AG + ROLE_HF) callback set to hfp_handle_hci_event causing bad behavior in sco

skoperst, 2018-06-10:
If someone will try the rare case of AG+HF initialized in same implementation, both will set their 
**hci_event_callback_registration** to be: hfp_handle_hci_event

In that case when one will get HCI_EVENT_CONNECTION_REQUEST, it will be sent twice and because hf_accept_sco is one for both roles, it will cause unneeded sco messages sent. 

And for my dongle(CSR8510 USB) it will cause crash once in a while. 
I can provide the test for this case if it will be useful.
Tnx.

skoperst, 2018-06-10:
Crash still happens without the extra messages. So i guess its not correct to say the crash is caused because of this issue. But it might be cleaner to fix it.

mringwal, 2018-06-10:
hi. thanks for reporting. I think only the hf should accept the connection, although I have seen HFs that initiate the SCO connection themselves. We recently introduced the HFP_ROLE field to improve the ability to use both roles in one device. 

Did you figure out the reason for your crash?

skoperst, 2018-06-10:
its getting into this line: (transfer->status == LIBUSB_TRANSFER_STALL) on **hci_transport_h2_libusb**, but it will never ever manage to clear the stall. I think the usb is restarting. It will always happen when opening the second sco connection. It will also happen if I open change the alt usb together or one by one. (Both using MSBC if that matters)

skoperst, 2018-06-10:
It will not happen if hci_send_sco_packet_buffer never called. That means if I don't send sco at all it will not happen. If I delay some time and then do the sending it will happen again

mringwal, 2018-06-11:
Do you try HF and AG at the same time? Is your device master in the piconet?
Most controllers can support up to 3 SCO connections, but I doubt that they can do that in a scatternet configuration.

skoperst, 2018-06-12:
I use HF & AG at the same time. I route audio from AG <--> HF and it works nice. It works both in scatternet and when I set i'm the master(piconet?). However when in scatternet the voice quality is degraded.  There are some changes need to be done to enable this in the current code, for example setting the final(2 connections X 8bit) in ALT SCO usb endpoint from the beginning before acquiring the audio connections. I think its a bug in CSR8510. though it will happen the same with the BCM ones. The BCM chipsets does not have the crashing issue i'm experiencing with the CSR.

skoperst, 2018-06-12:
And the MSBC codec, I added variables into contexts to enable multiple instances. Also something I can share if someone find it useful

closed 2019-05-26 by skoperst