bluekitchen/btstack #143: A2DP Source audio stream to two Sink device

rob-bits, 2018-06-28:
Hello all, @mringwal

Is it possible to stream audio to two sink device with the current btstack?

If not how much work should needed to do that? 

I know it is not an easy task and not a small modification (_The application should handle multiple connections. Event handler/state machine should be modified to handle multiple devices. .. And so on_). I just need to know can this btstack be a good base for this application or not. And it would be a proof-of-concept I do not want to have immediatly best audio quality. It would be nice to see the limits of this technology.

There is a study from bluetooth.org about a practical implemantation of this functionality. It is called "WHITE PAPER ON USAGE OF MULTIPLE HEADPHONES":
(https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=110679)

In this study there are use cases how should a source device act if the user wants to connect two sink audio device.

One of the scenario follows the next steps:

    AVDTP & AVCTP Connection establish with Sink device 1
    A2DP SEP Disc/Selec with Sink device 1
    SET_CONFIG for Sink device 1
    OPEN connection with Sink device 1
    START meadia Sink device 1 (ESP_A2D_MEDIA_CTRL_START)
    STREAM Media to Sink device 1
    ---- new device connection----
    GAP/ACL Connect to Sink device 2
    AVDTP Establish, SEP Disc/Select with Sink device 2
    Suspend media to Sink device 1 (ESP_A2D_MEDIA_CTRL_SUSPEND)
    SET_CONFIG to Sink device 2
    OPEN connection with Sink device 2
    START Media on Sink device 2 (ESP_A2D_MEDIA_CTRL_START)
    STREAM Media to Sink device 1
    STREAM Media to Sink device 2
    HAPPY USERs

What do you think how hard is this implementation?

mringwal, 2018-06-29:
hi. that's a very good question. It's something we wanted to try but didn't yet. 

The SBC audio codec currently currently can only used with one stream, so you could not encode two different streams or one stream in two different SBC configurations. However, it makes sense to encoded the audio only once and then send the encoded data to both sinks.

I would expect that the avdtp/a2dp code can handle multiple connections already.

rob-bits, 2018-06-29:
Hi, That is a really good news! Thank you very much for your update.

At the weekend I will play with this. It would be nice if the same encoded audio could be streamed to two device.

And would you please give me a hint about the implementation?

What do you think, is it enought the modify the API layer only (the a2dp_source.demo.c file)?

I would try something like this to handle two device connection:

```
#define BT_NUM_OF_DEVICES 2
#define BD_ADDR_LEN 6
uint8_t devTargetAddr[BT_NUM_OF_DEVICES][BD_ADDR_LEN];
static a2dp_media_sending_context_t media_tracker[BT_NUM_OF_DEVICES ] = {0};
...
status[0] = a2dp_source_establish_stream(&devTargetAddr[0][0], media_tracker[0].local_seid, &media_tracker[0].a2dp_cid);
`status[1]  = a2dp_source_establish_stream(&devTargetAddr[1][0], media_tracker[1].local_seid, &media_tracker[1].a2dp_cid);
```

and with the next function call in the a2dp_demo_send_media_packet() function can we stream audio in the same time to two device already, if I call it in a sequence with two different address?
```

a2dp_source_stream_send_media_payload(media_tracker[0].a2dp_cid, media_tracker[0].local_seid, media_tracker.sbc_storage, bytes_in_storage, num_frames, 0);
a2dp_source_stream_send_media_payload(media_tracker[1].a2dp_cid, media_tracker[1].local_seid, media_tracker.sbc_storage, bytes_in_storage, num_frames, 0);
```

mringwal, 2018-06-29:
hi. it should be possible to only modify the a2dp_source_demo (without touching the stack itself).
To send a single packet, you need the 'register can send now' which will provide you an event that results in a2dp_demo_send_media_packet getting called. you are only guaranteed to be able to send a single packet, so you could send on the first connection, then request to send again, and then to send on the other connection.

rob-bits, 2018-06-29:
I do not really see it as transparent as you wrote it.
I managed the two device connection, I hope they were both connected, but if I called the a2dp_source_stream_endpoint_request_can_send_now() function with a a2dp_cid and a local_seid I got the following error code:

a2dp_source.c.522: A2DP source: a2dp cid 0x56 not known, expected 0x57

And I do the following. The btstack starts, then I turn on the BT_speaker_1. It connects automatically to the esp32 and it starts play the audio. Then I turn on the second speaker. It is also connects automatically. The first one stops to play the audio and the other one does not play. It looks like the a2dp_demo_audio_timeout_handler() does not called at all. And for your suggestion if I call the a2dp_source_stream_endpoint_request_can_send_now() function manually nothing happens.

Would you please be more specific about the implementation? What are the "only modifications" on the a2dp_source_demo.c file?

rob-bits, 2018-06-29:
Here is my log when the second device is connection while the first one is playing audio:

```
API - streaming audio to 0
API - streaming audio to 0
[00:00:26.391] LOG -- hci.c.1982: Connection_incoming: 00:58:02:CC:02:64, type 1
[00:00:26.391] LOG -- hci.c.185: create_connection_for_addr 00:58:02:CC:02:64, type ff
[00:00:26.395] LOG -- hci.c.3292: sending hci_accept_connection_request, remote eSCO 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
[00:00:26.463] LOG -- hci.c.2006: Connection_complete (status=0) 00:58:02:CC:02:64
[00:00:26.463] LOG -- hci.c.2019: New connection: handle 129, 00:58:02:CC:02:64
[00:00:26.468] LOG -- hci.c.3922: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 2
[00:00:26.475] LOG -- hci.c.2085: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding fl
ags 6, eSCO 1
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
[00:00:26.818] LOG -- hci.c.2092: HCI_EVENT_LINK_KEY_REQUEST
[00:00:26.818] LOG -- hci.c.3318: responding to link key request
[00:00:26.819] LOG -- btstack_tlv_esp32.c.54: tag 42544c00 -> 42544C00
[00:00:26.825] LOG -- btstack_tlv_esp32.c.68: read tag 42544C00
[00:00:26.831] LOG -- btstack_tlv_esp32.c.54: tag 42544c01 -> 42544C01
[00:00:26.837] LOG -- btstack_tlv_esp32.c.68: read tag 42544C01
[00:00:26.843] LOG -- btstack_tlv_esp32.c.54: tag 42544c02 -> 42544C02
[00:00:26.849] LOG -- btstack_tlv_esp32.c.68: read tag 42544C02
[00:00:26.855] LOG -- btstack_tlv_esp32.c.54: tag 42544c03 -> 42544C03
[00:00:26.861] LOG -- btstack_tlv_esp32.c.68: read tag 42544C03
[00:00:26.867] LOG -- btstack_tlv_esp32.c.54: tag 42544c04 -> 42544C04
[00:00:26.873] LOG -- btstack_tlv_esp32.c.68: read tag 42544C04
[00:00:26.878] LOG -- btstack_tlv_esp32.c.54: tag 42544c05 -> 42544C05
[00:00:26.885] LOG -- btstack_tlv_esp32.c.68: read tag 42544C05
[00:00:26.890] LOG -- btstack_tlv_esp32.c.54: tag 42544c06 -> 42544C06
[00:00:26.897] LOG -- btstack_tlv_esp32.c.68: read tag 42544C06
[00:00:26.902] LOG -- btstack_tlv_esp32.c.54: tag 42544c07 -> 42544C07
[00:00:26.908] LOG -- btstack_tlv_esp32.c.68: read tag 42544C07
[00:00:26.914] LOG -- btstack_tlv_esp32.c.54: tag 42544c08 -> 42544C08
[00:00:26.920] LOG -- btstack_tlv_esp32.c.68: read tag 42544C08
[00:00:26.926] LOG -- btstack_tlv_esp32.c.54: tag 42544c09 -> 42544C09
[00:00:26.932] LOG -- btstack_tlv_esp32.c.68: read tag 42544C09
[00:00:26.938] LOG -- btstack_tlv_esp32.c.54: tag 42544c0a -> 42544C0A
[00:00:26.944] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0A
[00:00:26.950] LOG -- btstack_tlv_esp32.c.54: tag 42544c0b -> 42544C0B
[00:00:26.956] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0B
[00:00:26.962] LOG -- btstack_tlv_esp32.c.54: tag 42544c0c -> 42544C0C
[00:00:26.968] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0C
[00:00:26.974] LOG -- btstack_tlv_esp32.c.54: tag 42544c0d -> 42544C0D
[00:00:26.980] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0D
[00:00:26.985] LOG -- btstack_tlv_esp32.c.54: tag 42544c0e -> 42544C0E
[00:00:26.992] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0E
[00:00:26.998] LOG -- btstack_link_key_db_tlv.c.95: tag 42544c0e, addr 00:58:02:CC:02:64
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
API - streaming audio to 0
[00:00:27.058] LOG -- hci.c.3953: hci_emit_security_level 2 for handle 81
[... 107 lines cut ...]
[00:00:28.517] LOG -- l2cap.c.2138: l2cap - security level update
[00:00:28.523] LOG -- l2cap.c.2148: channel state 5: actual 2 >= required 2?
[00:00:28.530] LOG -- l2cap.c.912: L2CAP_EVENT_INCOMING_CONNECTION addr 00:58:02:CC:02:64 handl
e 0x81 psm 0x17 local_cid 0x46 remote_cid 0x42
[00:00:28.542] LOG -- avrcp.c.606: L2CAP_EVENT_INCOMING_CONNECTION avrcp_cid 0x02, l2cap_signal
ing_cid 0x46
[00:00:28.552] LOG -- l2cap.c.2259: L2CAP_ACCEPT_CONNECTION local_cid 0x46
[00:00:28.558] LOG -- l2cap.c.2148: channel state 12: actual 2 >= required 2?
[00:00:28.565] LOG -- l2cap.c.2148: channel state 12: actual 2 >= required 2?
[00:00:28.572] LOG -- l2cap.c.1053: l2cap_stop_rtx for local cid 0x46
[00:00:28.578] LOG -- l2cap.c.1062: l2cap_start_rtx for local cid 0x46
[00:00:28.588] LOG -- l2cap.c.2463: L2CAP signaling handler code 4, state 11
[00:00:28.591] LOG -- l2cap.c.2315: Remote MTU 48
[00:00:28.595] LOG -- l2cap.c.2325: Flush timeout: 65535 ms
[00:00:28.601] LOG -- l2cap.c.2463: L2CAP signaling handler code 5, state 11
[00:00:28.608] LOG -- l2cap.c.1053: l2cap_stop_rtx for local cid 0x46
[00:00:28.614] LOG -- l2cap.c.2395: l2cap_signaling_handle_configure_response
[00:00:28.621] LOG -- l2cap.c.884: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr 00:58:02:CC:02:64
 handle 0x81 psm 0x17 local_cid 0x46 remote_cid 0x42 local_mtu 1691, remote_mtu 48, flush_timeo
ut 65535
[00:00:28.638] LOG -- avrcp.c.643: L2CAP_EVENT_CHANNEL_OPENED avrcp_cid 0x02, l2cap_signaling_c
id 0x46
AVRCP Target: Connected to 00:58:02:CC:02:64, avrcp_cid 0x02

```

mringwal, 2018-07-04:
hi @rob-bits. I was referring to a2dp_source_stream_endpoint_request_can_send_now() (as used in the example).

I'm not too familiar with A2DP (@milamikica did). The a2dp_x.c implementation is just a thin layer on top of avdtp_sink/source and it is meant to support single streaming between two devices. You might use avdtp_source directly (and use a2dp_source as a blueprint for what to do).

You will also need two stream endpoints - one per streaming connection.

As I said, we'd like to look into extending a2dp_x.c to support multiple streams, but have to finish another project first. This, and the recommendation from the whitepaper (thanks!) could go into a2dp_source.c.

rob-bits, 2018-07-05:
Hi @mringwal,

Thank you very much for the update. 
I will still try to implement a proof-of-concept example for this while you are working on another topic. And I hope you or your team will have time soon to dive into this problem :)

Does it make sense to post my issues/logs here during the implementation, or not?

Rob

mringwal, 2018-07-06:
hi @rob-bits feel free to post here. It wouldn't hurt anybody.

rob-bits, 2018-07-06:
Hi @mringwal 

Okay. Well I managed to connect two device to the stack and I can alter between the to device to play the audio. But when I try to altern the two device in a sequence then the audio speeds up on both device and I got the following message:
`[00:02:00.325] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1`

I think currently the data cannot be streamed as fast as the audio should be played and that is why I got fast audio playback. My two thoughs are, I should limit the bandwith by modifying the max/min bitpool value. I have not figured out what are the currently used values yet. From the log it wrotes the sink device has the max bitpool 53 value.

The other thing what I am thinking is do I need stereo audio stream? As I know it is stereo by default. If I have mono audio sink devices then I wont need stereo stream, will I? 

Anyway  @mringwal  you were rigth. The stack does not need any modification. Only just the suggested functionalities had to be reworked.

If you have any idea with my issue it would be pleasured.

Rob

rob-bits, 2018-08-24:
Hi @mringwal,

Any update on extending a2dp_x.c to support multiple streams?

I have a working example for two speakers, but it is not too stable. It is not working well with some vendors of speakers.

Regards,

Robert

rob-bits, 2018-09-01:
Hi @mringwal ,

When the **avdtp_packet_handler()**function is called, where do it get the **size** aqument?

I tracked my problem back and it is relating to this variable. When the first device is connecting during **A2DP_W2_DISCOVER_SEPS** state the value of **size** in the **avdtp_initiator_stream_config_subsm()** function during **avdtp_signaling_emit_sep()**  is **4**. When second device connecting it should be **8**. And the second device can connect. But if the value of this variable is the same as the first connection, 4, then the previously describe problem occures: [143#issuecomment-415758922](https://github.com/bluekitchen/btstack/issues/143#issuecomment-415758922). 
So only one Sep is found and it is already in the state 7 because it holds the first device connection and the second cannot connect to that sep.

Currently the variable named  **size** get the value of **8** only with one type of BT speaker. And I do not why. So I have four speakers which can connect in the first place, but only 1 BT device can make connection after the first device is connected. If I try other two device to connect in the second place then I got the previous problem.

So would you please give me a hint where do the **size** variable get its value?

Robert

d762, 2020-02-19:
hello guys. Have you implemented this already?

mringwal, 2020-02-20:
we had a few small fixes to a2dp/avdtp/avrcp in the past, but did not test a multi-speaker setup. As far as I know, the a2dp source implementation can handle multiple connections, but the demo needs to be extended - and also, some time sync for audio might be needed.

cQszasza, 2021-06-01:
Hi guys! Any news on this one?

mringwal, 2021-06-02:
Some progress was made. A2DP Source can handle multiple connections now, but there's no demo that shows that off yet (let alone tries to implement time sync).

cQszasza, 2021-06-03:
Do you guys have time to create a demo? I'm not good enough to do this myself but with a little help maybe I can. I'm thinking about creating a bluetooth speaker hub thingy and I'd like to know how to stream to multiple sink devices. 
Thank you!

mringwal, 2021-06-04:
@cQszasza not right now. We might try to extend the a2dp source demo to support multiple speakers/headphones.This will be in the context of one or more Bluetooth headsets, where audio latency doesn't matter.

One thing to consider: A2DP is not able to sync multiple devices in a reasonable way. The best it provides is "delay reporting", but I would not expect that to allow time sync down to a few ms. The upcoming LE Audio will have full support for streaming to multiple sink with audio sample accuracy (e.g. for True Wireless Headbuds)_ but there is no products out there as far as I know.