bluekitchen/btstack #144: develop merge in master and freertos bad includes

Lefix2, 2018-06-29:
Hi,
I recently merged your master in my project seeing that develop branch has been merged.
Apparently, the issue[ I've opened](https://github.com/bluekitchen/btstack/issues/132) some times ago was corrected on develop but not merged, is it normal?

I had to modify some includes too, 

https://github.com/bluekitchen/btstack/blob/7284beeecf4402e2dab3aabb82ca53a9de2b1e31/platform/freertos/btstack_run_loop_freertos.c#L57-L60

Regarding [FreeRTOS repository](https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS/Source/include/), it will be more "FreeRTOS/Source/Include" than "freertos/"
But this relative path is, on my opinion, useless. Include directly headers name will be more portable:

#include "FreeRTOS.h"
#include "task.h"
#include "queue.h"
#include "event_groups.h"

mringwal, 2018-08-03:
sorry for not responding. no, there was no intention if fixes did not show up on master. We just merged develop onto master again. compile_gatt.py is identical right now. 

I'll update the issue title.

mringwal, 2019-06-03:
Some distributions, notably ESP32's esp-idf, reorganize the FreeRTOS sources - probably to support more than one 'list.h'. I've updated the code to make the freertos/.. optional via a btstack_configh #define with the default being 'no freertos prefix'.

Fixed in 6d23ba053

closed 2019-06-03 by mringwal