bluekitchen/btstack #145: BTSTACK_TEST define warnings

Lefix2, 2018-06-29:
I think these two lines need conditionnal compilation to avoid warnings about non-implemented function exit() and printf():
https://github.com/bluekitchen/btstack/blob/cea733e4bd16389141892d269be2d57149250adc/platform/embedded/hal_flash_bank_memory.c#L94-L98

To look like something like this:
```
if (self->banks[bank][offset] != 0xff && data[i] != 0x00){
#ifdef BTSTACK_TEST
	printf("Error: offset %u written twice. Data: 0x%02x!\n", offset+i, data[i]);
	exit(10);
#endif
	return;	
}
```		
}

mringwal, 2018-06-29:
thanks. you're absolutely right here. the exit really should only be used for testing. I'll fix it.

mringwal, 2018-07-03:
Fixed on develop (commit 432f6cc7ec0690be514790f80346b8c0228c5f71)

closed 2018-07-03 by mringwal