bluekitchen/btstack #146: OBEX iterator skips headers or read past end for certain OBEX header sizes

skoperst, 2018-07-04:
I found that OBEX iterator sometimes stop before reading all headers of a OBEX packet, or continues past the last header.

I think this happens because obex_iterator.{offset,length} is uint8_t, but should be uint16_t to allows header sizes up to 2^16 as defined in the OBEX spec.

https://github.com/bluekitchen/btstack/blob/432f6cc7ec0690be514790f80346b8c0228c5f71/src/classic/obex_iterator.h#L48L52

Is there a reason its defined 8 bit?

skoperst, 2018-10-01:
Any updates?

I suggest changing to:
`typedef struct obex_iterator {
     const uint8_t * data;
     uint16_t   offset;
     uint16_t   length;
} obex_iterator_t;`

mringwal, 2018-10-01:
sorry. it's fixed on develop. please try again. I'm active on getting goep/pbap to pass the tests and just stumbled upon that bug, too.

skoperst, 2018-10-01:
Thanks. I probably missed it.

skoperst, 2018-10-19:
Fixed on https://github.com/bluekitchen/btstack/commit/820092f9d45777a9b1f0966260ebdbd92687a64d

closed 2018-10-19 by skoperst

mringwal, 2018-10-19:
hi. just (like 1h ago) found another bug: it was using an uint8_t for len and offset which won't work for > 255 byte packets.

skoperst, 2018-10-19:
Yep, that was the issue I had. It also in BT spec defined as 16bit values