bluekitchen/btstack #147: ESP32: Slow connection to mobile phone in contrast with PC connection

my-abousamra, 2018-07-10:
Hi,
I've an application based on classic Bluetooth of btstack port for ESP32 module. When connecting from PC to the module and stream data, then no data is dropped but when connecting from mobiles (Android) the connection becomes slow. After enabling logging messages for btstack I noticed that credits in case PC is 40 but in case of mobiles it becomes 10.

Here are logging for both PC connection and mobile connection.

PC connection log

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5808
ho 0 tail 12 room 4
load:0x40078000,len:7684
load:0x40080000,len:7448
entry 0x400803a0
I (30) boot: ESP-IDF untagged-456460844cbaf2d60ce1 2nd stage bootloader
I (30) boot: compile time 22:46:47
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00300000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24be0 (150496) map
I (148) esp_image: segment 1: paddr=0x00034c08 vaddr=0x3ffc0000 size=0x041d4 ( 16852) load
I (154) esp_image: segment 2: paddr=0x00038de4 vaddr=0x3ffc41d4 size=0x00000 (     0) load
I (155) esp_image: segment 3: paddr=0x00038dec vaddr=0x40080000 size=0x00400 (  1024) load
I (164) esp_image: segment 4: paddr=0x000391f4 vaddr=0x40080400 size=0x06e1c ( 28188) load
I (184) esp_image: segment 5: paddr=0x00040018 vaddr=0x400d0018 size=0x9c898 (641176) map
I (409) esp_image: segment 6: paddr=0x000dc8b8 vaddr=0x4008721c size=0x0f090 ( 61584) load
I (435) esp_image: segment 7: paddr=0x000eb950 vaddr=0x400c0000 size=0x00000 (     0) load
I (435) esp_image: segment 8: paddr=0x000eb958 vaddr=0x50000000 size=0x00000 (     0) load
I (454) boot: Loaded app from partition at offset 0x10000
I (455) boot: Disabling RNG early entropy source...
I (455) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x4008128c
I (0) cpu_start: App cpu up.
I (469) heap_init: Initializing. RAM available for dynamic allocation:
I (476) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (482) heap_init: At 3FFD3130 len 0000CED0 (51 KiB): DRAM
I (488) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (495) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (501) heap_init: At 400962AC len 00009D54 (39 KiB): IRAM
I (507) cpu_start: Pro cpu start user code
I (78) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
BTstack: setup
[00:00:00.079] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffd67b0
[00:00:00.086] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue item size 8
[00:00:00.094] LOG -- main.c.266: transport_register_packet_handler
[00:00:00.131] LOG -- btstack_tlv.c.48: TLV Instance 0x3f408654
[00:00:00.132] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.132] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.138] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.144] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.150] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.156] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.162] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.168] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.174] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.180] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.186] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.192] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.198] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[... 418 lines cut ...]
[00:00:48.465] LOG -- rfcomm.c.1511: Received MSC RSP for #2
[00:00:48.470] LOG -- rfcomm.c.1614: rfcomm_channel_ready_for_open state 8, flags needed 0020, current e03b, rf credits 7
[00:00:48.480] LOG -- rfcomm.c.1265: rfcomm_channel_opened!
[00:00:48.486] LOG -- rfcomm.c.169: RFCOMM_EVENT_CHANNEL_OPENED status 0x0 addr DC:09:4C:F8:6D:8F handle 0x80 channel #1 cid 0x01 mtu 990
I (48498) bt_classic_packet_handler: RFCOMM channel open succeeded. New RFCOMM Channel ID 1, max frame size 990

[00:00:48.572] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 4, now 10
[00:00:48.609] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 2, now 10
[00:00:48.670] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 1, now 10
[00:00:48.714] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 2, now 10
[00:00:52.752] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 2, now 7
[00:00:52.754] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 1, now 8
[00:00:52.828] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:52.915] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.020] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.124] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.229] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.317] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.415] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.529] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.615] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.729] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.815] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
[00:00:53.915] LOG -- rfcomm.c.1301: RFCOMM data UIH_PF, new credits: 6, now 10
`

mringwal, 2018-07-10:
What throughput do you get in both cases without packet logging and in what direction? An Android phone is probably much slower than a full PC, too.. :)

my-abousamra, 2018-07-10:
I send about 8 kilo bytes per second from ESP32 module to the mobile phone.

Note that, That rate was working with an older bluetooth module which is replaced by ESP32.

my-abousamra, 2018-07-10:
Another question. Why credits in mobile connection are lower than them in PC connection? and how to increase them in order to avoid slow connection?

mringwal, 2018-07-10:
The credits define how many packets we can sent before we get new ones. The amount is not really the limit, it's more how fast they are updated. 

There's nothing that can be done to get more credits per unit of time. Maybe check the remote app.

my-abousamra, 2018-07-10:
The mobile app was working fine with an old Bluetooth module without any slow connection before we replaced it with the esp32.
We tried it now with the old module and it still works.
Actually I can't figure out the difference, Can you?
Thanks in advance

mringwal, 2018-07-11:
Hi

Just to confirm: the main problem is sending from the ESP32 to Android, right?

The only thing that could cause a slower connection compared to an SPP module would be if you send short RFCOMM packets instead of sending packets with max RFCOMM frame size / MTU. BTstack sends one packet for each call to rfcomm_send.

What size does your RFCOMM messages have?

You also could try the spp_streamer example to see if you get good throughput (it send max frame size). We get > 200 kB/s between two ESP32 with that. 

Cheers
Matthias

mringwal, 2018-07-11:
> Yes, data is sent from the ESP32 to Android.
> Data packet size is 174 bytes sent per rfcomm_send call.
> Note that:
> Old Bluetooth module used before ESP32 is classic only and does not support low energy.
> We tried the mobile app from an old mobile phone (with bluetooth classic only and does not support low energy) and it works fine with BTstack/ESP32.
> Now, We want to use the mobile app with a modern mobile phone that supports Bluetooth classic and low energy to stream data from BTstack/ESP32.
> We think that the problem is about the unused Bluetooth low energy and we think disabling it from BTstack would solve the problem.

Hi again

Please try sending multiple 174 byte messages in a single rfcomm_send call. 

The credits count RFCOMM messages, so you could send roughly 5 times more data with the same amount of credits. 

As far as I know there's no overhead for BLE if you don't send advertisements or scan for them (nothing to do, nothing to schedule).

my-abousamra, 2018-07-11:
before that try, kindly note that when we use the mobile app over old mobile phone that doesn't support BLE with BTstack/ESP32, we got 40 credits (as PC case) which is sufficient for the communication and doesn't make slow connection, in contrast with using the mobile app over modern mobile phone that supports BLE we got only 10 credits that causes the problem. 
So, I want to got 40 credits as the modern phone that is sufficient for the communication without much altering the code.

mringwal, 2018-07-11:
Hi

The difference in credits together with the only partial used RFCOMM packet could explain your observation.

The optimal way to get max throughput is to fill the RFCOMM packet. You could use the BTstack Ring Buffer and put the data into it plus send a packet once the ring buffer has enough data for a full RFCOMM packet.

If you like to try, you could reduce the RFCOMM MTU, in the call to rcomm_register_service. If the mobile phone calculates the number of RFCOMM credits based on the RFCOMM frame size, it might give you more credits.

my-abousamra, 2018-07-11:
Hi

reducing RFCOMM MTU size doesn't make any difference.

I want to disable BLE from the BTstack by commenting out 

//#define ENABLE_BLE
//#define ENABLE_LE_PERIPHERAL
//#define ENABLE_LE_CENTRAL
//#define ENABLE_LE_DATA_CHANNELS
//#define ENABLE_LE_DATA_LENGTH_EXTENSION
//#define ENABLE_LE_SECURE_CONNECTIONS
//#define ENABLE_MICRO_ECC_FOR_LE_SECURE_CONNECTIONS

but I got some "gatt functions implicit declarations" errors.
So, How to disable BLE in simple and clear way?

Thanks

my-abousamra, 2018-07-12:
Hi,
I disabled BLE as following, Please confirm it is a right way or not.
- I removed lines src/ble and src/ble/gatt-service from port/esp32/components/btstack/component.mk twice one from COMPONENT_ADD_INCLUDEDIRS and the other at COMPONENT_SRCDIRS and then rerun ./integrate_btstack.py
- I changed Esp_bt_controller_enable(ESP_BT_MODE_BTDM); to only use Bluetooth Classic in .../components/btstack/main.c:transport_open().
- I avoided commenting  ENABLE_BLE + ENABLE_LE_* in btstack_config.

I compiled the project where I faced that error
esp-idf/components/btstack/main.c:368: undefined reference to `le_device_db_tlv_configure'

so, I commented out these line in the main function
    // setup LE Device DB using TLV
    le_device_db_tlv_configure(btstack_tlv_impl, NULL);

and changed the code to send 5 packets per rfcomm_send call ( 5 * 168 bytes = 840 bytes) where max frame size is 990 then the ESP32 module streams data well to the mobile phone where it sends credits (10 credits) until it stops sending credits for about 7-8 seconds so the module stops sending data until the mobile sends credits again then the module resumes sending data again. This problem occurs from now and then.

I enabled BLE again and this problem still exists as well.

Note that the old mobile phone works well without this problem with the high credits range.

Thanks in advance.

my-abousamra, 2018-07-12:
Hi,

sending 5 messages per rfcomm_send solved the problem in high specs mobile phones but with medium specs mobile phone it stops sending credits for a while in middle of streaming which makes the connection slow so I tried to increase the buffer using rfcomm_register_service to 1010 to be able send 6 messages but it refused to give me that length and only allocates 990 bytes then the module application crashes with error "rfcomm data lenght exceeds MTU!". So how can I make rfcomm_register_service give more buffer length (1010 bytes)?
In addition disabling BLE doesn't make any change but I want to confirm the steps

- I removed lines src/ble and src/ble/gatt-service from port/esp32/components/btstack/component.mk twice one from COMPONENT_ADD_INCLUDEDIRS and the other at COMPONENT_SRCDIRS and then rerun ./integrate_btstack.py
- changed Esp_bt_controller_enable(ESP_BT_MODE_BTDM); to only use Bluetooth Classic in .../components/btstack/main.c:transport_open().
    I avoided commenting ENABLE_BLE + ENABLE_LE_* in btstack_config.

I compiled the project where I faced that error
esp-idf/components/btstack/main.c:368: undefined reference to `le_device_db_tlv_configure'

so, I commented out these line in the main function
// setup LE Device DB using TLV
le_device_db_tlv_configure(btstack_tlv_impl, NULL);

I may disable it to save some memory space

thank you

mringwal, 2018-07-13:
Hi

Disabling BLE: you're steps are correct. 

Throughput: the RFCOMM framesize is negotiated between both sides. You can only make it smaller. BTstack tries to use the max size. 

If you use a ringbuffer (eg src/btstack_ringbuffer.h) you can store the messages in it and use the full RFCOMM MTU. 

Pauses: if there are no credits, BTstack is not allowed to send and must wait. Maybe the medium spec phone is still processing the incoming data? Could you try to just drop the data in the app?

my-abousamra, 2018-07-15:
Hi
Regarding the ring buffer idea, I already use a queue to store data and read 5 messages from it then send them all per rfcomm_send call but it overflows eventually.

Now, I'm trying to drop data from the mobile application and check what happens and inform you.

Thank you

my-abousamra, 2018-07-16:
After dropping messages reached to mobile phone, no messages were lost.

Now, we will try to optimize the mobile application.

Thank you for your efforts.

closed 2018-07-16 by my-abousamra