bluekitchen/btstack #148: chipset BT860 support

argh35, 2018-07-13:
I am trying to use a BT860 (Cypress20704A2). I can communicate via PC (USB->FTDI->BT860), and via STM32 using the USART (on a different hardware).
 I obtain different manufacturer code (CYPRESS vs BROADCOM). Can it be a communication issue? I started from stm32f4 port to configure usart.
In both case the stack go into working state (BTstack up and running) but with stm32 (where I read BROADCOM manufacturer) I cannot see the device on the Bluetooth network.
Any suggestion would be appreciated.

mringwal, 2018-07-13:
Hi

Do you get a different manufacturer in the HCI Local Version Information Event from the same chip? That's weird :)

Could you enable LOG_INFO and full packet HCI packet logging and post the logs for both setups?

mringwal, 2018-07-19:
Hi Arianna

I'm glad it was an hardware error. 

What hal_uart_dma.h implementation  did you use? The UART should never loose bytes. The easiest way is to use hardware handshake RTS/CTS. At least in older Cube F4 drivers, a bug did cause bytes getting lost: see http://wordpress.bluekitchen-gmbh.com/btstack-port-for-stm32-f4-discovery-board-with-cc256x/ (the section after **hal_uart_dma.h**)

FreeRTOS: BTstack isn't thread-safe, but you can run BTstack as a dedicates thread that contains all Bluetooth logic (which most of the times is a good idea anway) and then use FreeRTOS IPC mechanisms to communicate with the rest of the app. 

The ESP32 runs FreeRTOS by default, so the esp32 port integrates with FreeRTOS as well although it doesn't use a UART as the Bluetooth runs on the host MCU and has a virtual HCI interface.

In main, you can use the FreeRTOS run loop
    btstack_run_loop_init(btstack_run_loop_freertos_get_instance());

and also use the btstack_uart_block_freertos.c implementation with hci_transport_h4.

Feel free to post on the BTstack mailling list for general advice and/or contact us directly for commercial projects.

closed 2018-07-19 by mringwal

argh35, 2018-07-20:
Hi Matthias,
thanks for your help! It was the bug on F4 drivers to cause the bytes getting lost!
So I worked to change the receiving method using DMA in circular mode (https://community.st.com/thread/42689-efficiently-use-dma-with-uart-rx-on-stm32) – used Edit 1.
In this way it seems to work well also using the original drivers.

So now I am using the BTstack with freertos instance, adapted to use the Keil OS (RTX) and original F4 drivers, and everything seems to work correctly!

Thank you for your fast and helpful reply.

Have a great day,

Best regards

Arianna

ARIANNA GHEZZI
r&d engineer
PHONE:+39 049 8962844