bluekitchen/btstack #149: HSP: Timeout when receiving SCO connection request

ajraymond, 2018-07-17:
Hi,

I am trying to implement the HSP profile, and am running into a timeout which I have traced in the code to the following part of hci.c:
```
#ifdef ENABLE_CLASSIC
            case RECEIVED_CONNECTION_REQUEST:
                log_info("sending hci_accept_connection_request, remote eSCO %u", connection->remote_supported_feature_eSCO);
                connection->state = ACCEPTED_CONNECTION_REQUEST;
                connection->role  = HCI_ROLE_SLAVE;
                if (connection->address_type == BD_ADDR_TYPE_CLASSIC){
                    log_info("is classic");
                    hci_send_cmd(&hci_accept_connection_request, connection->address, hci_stack->master_slave_policy);
                } else if (connection->address_type == BD_ADDR_TYPE_SCO){
                    log_info("is sco");
                }
                return;
#endif
```

It turns out that the connection request is of `connection->address_type == BD_ADDR_TYPE_SCO`, which does nothing. I then receive a timeout event a couple of seconds later.

Here is the trace:

> Connection_incoming: XX:XX:XX:XX:XX:XX, type 2
> create_connection_for_addr XX:XX:XX:XX:XX:XX, type fe
> sending hci_accept_connection_request, remote eSCO 1
> is sco
> (a couple of seconds pass)
> Synchronous Connection Complete (status=16) XX:XX:XX:XX:XX:XX
> HFP: connection does not exist for remote with addr XX:XX:XX:XX:XX:XX.
> (e)SCO Connection failed, status 16
> Audio connection establishment failed with status 16

Any pointers?

ajraymond, 2018-07-17:
Hmm, I just noticed some changes to classic/hsp_hf.c mentioning accepting SCO connections. I'll update to the latest master and try again.

ajraymond, 2018-07-17:
No, same symptoms after update

mringwal, 2018-07-17:
Please try with develop branch and check if max num hci connections in btstack_config.h is 2 or higher

ajraymond, 2018-07-17:
I have made progress and managed to isolate another problem:
in classic/hsp_hs.c:
```
        case HCI_EVENT_CONNECTION_REQUEST:
            switch(hci_event_connection_request_get_link_type(packet)){
                case 0: //  SCO
                case 2: // eSCO
                    hci_event_connection_request_get_bd_addr(packet, event_addr);
                    printf("remote device %s\n", bd_addr_to_str(remote));
                    printf("accept sco for device %s\n", bd_addr_to_str(event_addr));

                    if (bd_addr_cmp(event_addr, remote) == 0){
                        printf("hs_accept_sco_connection \n");
                        hs_accept_sco_connection = 1;
                    }
                    break;
                default:
                    break;                    
            }            
break;
```
The only code which sets `remote` is `hsp_hs_connect()` so this condition is never satisfied if the connection is initiated by the AG, and not the HS:

`if (bd_addr_cmp(event_addr, remote) == 0)`

ajraymond, 2018-07-19:
I don't think the change has been pushed yet.

The tip of develop is:
https://github.com/bluekitchen/btstack/commit/4789e87214d93a37bf7607f1335db2a11066656c

mringwal, 2018-07-31:
The change was to set remote once the RFCOMM connection is opened.
https://github.com/bluekitchen/btstack/blob/develop/src/classic/hsp_hs.c#L618

Does this make sense/does it work this way for you?

ajraymond, 2018-08-03:
Your fix solves my issue.
Thanks!

closed 2018-08-03 by ajraymond