bluekitchen/btstack #151: Example programs on Raspberry Pi 3 Model B

peet1993, 2018-07-25:
Hi,

I tried to run the example programs on a Raspberry Pi 3 Model B with the built-in bluetooth chip. I used [this](https://groups.google.com/forum/#!topic/btstack-dev/KVmKohlVqUI) thread as a reference, even though I don't quite understand how the powercycle part is supposed to work. Should I add something to the main.c of the posix-h4 port? After I uninstalled bluez from the Pi, the bluetooth entry in rfkill wasn't there anymore.

Currently, it does some startup stuff at first try (like getting the local name etc) and then ends in an unknown HCI packet 0xf6. Every successive run skips the startup stuff and directly fails with the unknown packet (see attached packet log). I used the hog_keyboard_demo, but all examples behave the same.

Maybe you got it to run on a Pi 3 since the referenced thread was written and i'm just doing something wrong. It works fine with a bluetooth dongle and libusb-port.

Regards,
Peter Oettig

[hci_dump.zip](https://github.com/bluekitchen/btstack/files/2228185/hci_dump.zip)

mringwal, 2018-07-25:
hi @peet1993 you've got good timing. A few days ago, it was only possible to have bluez upload the patch ram, disable it and then start BTstack.

If you now try the 'raspi3' branch - port/raspi, it works without bluez. You still need to power cycle the Bluetooth module. For this, you can use this tool https://github.com/6by9/rpi3-gpiovirtbuf - gpiovirtbuf w 128 0 .... gpiovirtbuf w 128 1 - we will integrate the functionality into the the port/main.c soon, too.

The code might not compile (after some minor refactorings) but it is working.

peet1993, 2018-07-25:
Hi @mringwal ,
thanks for the fast reply! That sounds awesome, I will try it out and add a response here how it went.

mringwal, 2018-07-25:
Well. I can compile again, and there are many compile errors. I'll fix them tomorrow.

mringwal, 2018-07-26:
Please pull again. It compiles now.

Do you have the file BCM43430A1.hcd in /lib/firmware or in /lib/firmware/brcm/
The code assumes to find it in /lib/firmware/

Example run:
pi@raspberrypi:~ $ ./rpi3-gpiovirtbuf w 128 0
Set state of 128 to 0
pi@raspberrypi:~ $ ./rpi3-gpiovirtbuf w 128 1
Set state of 128 to 1
pi@raspberrypi:~ $ ./le_counter
Packet Log: /tmp/hci_dump.pklg
Hardware UART without flowcontrol
Phase 1: Download firmware
Phase 2: Main app
BTstack counter 0001
BTstack up and running at B8:27:EB:EE:FF:CC

mringwal, 2018-07-26:
and.. now the power cycle is included, no need for rpi3-gpiovirtbuf anymore

peet1993, 2018-07-26:
I finally got to trying it out. I have to say that it works flawlessly after I downloaded the broadcom firmware. Thanks for your terrific and fast help!

mringwal, 2018-07-26:
You're welcome - no worries. As I said before, you're timing was lucky. It wasn't working for a long time. I'll just need to add a README before merging it to develop and eventually master as well.

mringwal, 2018-07-29:
Added README, tested on Raspberry Pi Zero W -> doesn't work, merged onto develop.

closed 2018-07-29 by mringwal