bluekitchen/btstack #153: Some bugs on the raspi3/develop branch

peet1993, 2018-07-26:
Hi,

i know that the raspi3/develop branch is experimental, but maybe this report helps you.

* The libusb port can't be compiled, there is a compiler error with the PATH_MAX define in the tinydir 3rd-party tool.
* Using the hog_keyboard_example in the raspi port, the program doesn't find a tlv instance to store persistency data in. That means that reconnecting to a previously paired device doesn't work as the key_report gatt attributes aren't automatically resubscribed. It works fine on master with the libusb port. As the libusb port doesn't compile on develop, i can't check if it works with that.

Regards,
Peter Oettig

mringwal, 2018-07-27:
thanks for reporting. libusb works for raspi on master but not on develop/raspi?

good catch about the persistency stuff. all is there, it just need to be configured :)
you may copy the snippet in HCI_STATE_WORKING from libusb over.
I'll make a note to deal with these as well when cleaning up and merging back.

peet1993, 2018-07-27:
> libusb works for raspi on master but not on develop/raspi?
Exactly, [here](https://pastebin.com/etUiNFVH) is a pastebin of the makefile output. I tried to look at why it happens, I think the define changed from MAX_PATH to PATH_MAX, not sure though. It doesn't compile on my desktop either for the same reason.

Thanks for the hint with the code snippet, I copied it and it is working now :)

schodet, 2018-08-14:
About `PATH_MAX`, it is not defined when building with `-std=c99`. Without the flag, it build.

The code is on master now, so master cannot build on Linux.

mringwal, 2018-08-14:
thanks for the reminder. I've update to the latest from tinydir repo and added a #ifdef for PATH_MAX. If it's not defined, tinydir will use a default of 4096. Please let me know if this works for you.

mringwal, 2018-08-14:
TLV is also setup in the Raspi port. Closing this now. Feel free to post a new issue if theres's something else wrong.

mringwal, 2018-08-14:
the update for tinydir is on develop for now

closed 2018-08-14 by mringwal

schodet, 2018-08-14:
It works for me.

I have open an issue on tinydir side: https://github.com/cxong/tinydir/issues/61