bluekitchen/btstack #158: hid_host_demo stops receiving data packets

ASPrabhu96, 2018-08-09:
I was trying to pair a playstation 4 dualshock controller with an esp32 running the hid_host_demo in the btstack examples. It pairs with the device, and starts reading apckets. however, after 26 packets it randomly stops (the connection is still maintained, the devices are paired). I am not sure as to why this is happening. Any idea ? or, if anyone has tried something similar and are willing to share source code it would be great! thanks!

mringwal, 2018-08-09:
do you get HID Control messages? The demo prints "HID Control:..". An HID Host is supposed to answer these requests. Could you try with e.g. a Bluetooth keyboard as well?

mringwal, 2020-07-15:
No response, closing.

closed 2020-07-15 by mringwal