bluekitchen/btstack #160: Privacy support

Tsai1005, 2018-08-10:
Is there a schedule to add privacy feature (Core Ver.4.2) and hci cmd set?

mringwal, 2018-08-12:
BTstack allows to use private addresses (gap_random_address_set_mode) and can also resolve private addresses (sm_address_resolution_lookup) if asked to. Connecting to a paired device with a private address requires to do a scan and have BTstack do a address lookup for each device in the application.

What feature are you looking for/what do you want to do? What Bluetooth Controller do you use?

Tsai1005, 2018-08-13:
The Core ver4.2 mentioned that the privacy 1.2 was support not only by the host but also the controller, it would reduce the host works to handle the RPA, but need some new HCI commands to support, I bought a new dongle from Bluetooth SIG which supports LL feature Privacy in the controller,So I just want to test this feature.
However, I quite confuse about the Privacy feature, many central devices like mobile phone used the RPA1, after boned to a periphery device and disconnect by any reason, the peer device using direct adv packet, if the central device change to the RPA2, how would the periphery device do when reconnect because the direct adv packet includes the InitA = RPA1, according to the BQB spec, the periphery(v4.0) would not be connected, but periphery (v4.2) would be connected. So is there any conflict about direct adv & privacy?

mringwal, 2018-08-13:
Hi @Tsai1005 - this discussion probably would be better on the BTstack mailing list, maybe others have ideas, too. 

Well, I don't see how directed advertisement could work together with Random Private Addresses. There's not enough time (just 150 us) to decide if a connection request is sent after an advertisement is received. This isn't enough to do some IRK hash calculations. (Although it could calculate them and then respond to the next advertisement with that particular private address).

When privacy was introduced, I was wondering how Whitelists could be used for outgoing connections as the RPA of the remote isn't known at that time. With the new Resolving List, this seems to be possible now.

If you want to try out the new commands, you can define them using a template http://bluekitchen-gmbh.com/btstack/protocols/#hci-host-controller-interface (Sending Commands using a template). Just make sure that BTstack is fully working and hci_can_send_command_packet_now() returns true. 

To answer your original question, we don't have a timeline to support/use the new HCI commands yet.

Tsai1005, 2018-08-16:
Thx for your advice, I post the question to the mailing list.
I agree with you, ref to the spec (<LL.TS.4.2.1.pdf>  4.3.3.6 TP/CON/INI/BV-07-C [Initiation Device Filtering: Directed]),
[LL.TS.4.2.1.pdf](https://github.com/bluekitchen/btstack/files/2292894/LL.TS.4.2.1.pdf)
Finally, I would try the new commands add by myself.

closed 2018-08-16 by Tsai1005