bluekitchen/btstack #161: hsp_ag_ audio connection establishment issue

sapthas, 2018-08-21:
hi,
I was unable to communicate with a headset using the hsp_ag_demo example code.
I am getting an error message 
[00:00:23.465] LOG -- hsp_ag.c.602: (e)SCO Connection failed, status 26         
Audio connection establishment failed with status 26 
 full statement is given below.
BTstack: setup                                                                  
[00:00:00.032] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffcf3e8 
[00:00:00.034] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue ite8
[00:00:00.042] LOG -- main.c.272: transport_register_packet_handler             
[00:00:00.053] LOG -- btstack_tlv.c.48: TLV Instance 0x3f407290                 
[00:00:00.053] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400          
[00:00:00.060] LOG -- btstack_tlv_esp32.c.68: read tag 42544400                 
[00:00:00.065] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401          
[00:00:00.071] LOG -- btstack_tlv_esp32.c.68: read tag 42544401                 
[00:00:00.077] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402          
[00:00:00.083] LOG -- btstack_tlv_esp32.c.68: read tag 42544402                 
[00:00:00.089] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403          
[00:00:00.095] LOG -- btstack_tlv_esp32.c.68: read tag 42544403                 
[00:00:00.101] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404          
[00:00:00.107] LOG -- btstack_tlv_esp32.c.68: read tag 42544404                 
[00:00:00.113] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405          
[00:00:00.119] LOG -- btstack_tlv_esp32.c.68: read tag 42544405                 
[00:00:00.125] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406          
[00:00:00.131] LOG -- btstack_tlv_esp32.c.68: read tag 42544406                 
[00:00:00.137] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407          
[00:00:00.143] LOG -- btstack_tlv_esp32.c.68: read tag 42544407                 
[00:00:00.149] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408          
[00:00:00.155] LOG -- btstack_tlv_esp32.c.68: read tag 42544408                 
[00:00:00.160] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409          
[00:00:00.167] LOG -- btstack_tlv_esp32.c.68: read tag 42544409                 
[00:00:00.172] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A          
[00:00:00.179] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A                 
[00:00:00.184] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B          
[00:00:00.190] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B                 
[00:00:00.196] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C          
[00:00:00.202] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C                 
[00:00:00.208] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D          
[00:00:00.214] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D                 
[00:00:00.220] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E          
[00:00:00.226] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E                 
[00:00:00.232] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F          
[00:00:00.238] LOG -- btstack_tlv_esp32.c.68: read tag 4254440F                 
[00:00:00.244] LOG -- le_device_db_tlv.c.160: num valid le device entries 0     
SCO Demo: Sending sine wave, hexdump received data.                             
[00:00:00.255] LOG -- l2cap.c.3400: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535    
SDP service record size: 90                                                     
[00:00:00.264] LOG -- rfcomm.c.2301: RFCOMM_REGISTER_SERVICE channel #1 mtu 6550
[00:00:00.274] LOG -- l2cap.c.3400: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535    
[00:00:00.281] LOG -- hci.c.2800: hci_power_control: 1, current mode 0          
[00:00:00.287] LOG -- main.c.204: transport_init                                
[00:00:00.291] LOG -- main.c.220: transport_open                                
I (295) BTDM_INIT: BT controller compile version [9525061]                      
                                                                                
I (302) system_api: Base MAC address is not set, read default base MAC address E
I (528) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0            
[00:00:00.529] LOG -- main.c.251: transport: configure SCO over HCI, result 0x00
[00:00:00.531] LOG -- hci.c.3859: BTSTACK_EVENT_STATE 1                         
BTstack: execute run loop                                                       
[00:00:00.539] LOG -- hci.c.1902: Manufacturer: 0x0060                          
[00:00:00.544] LOG -- hci.c.1816: local name:                                   
[00:00:00.548] LOG -- hci.c.1605: Received local name, need baud change 0       
[00:00:00.555] LOG -- hci.c.1912: Local supported commands summary 0x3f         
[00:00:00.561] LOG -- hci.c.1864: Local Address, Status: 0x00: Addr: 30:AE:A4:02
[00:00:00.568] LOG -- hci.c.1833: hci_read_buffer_size: ACL size module 1021 ->4
[00:00:00.579] LOG -- hci.c.1891: Packet types cc18, eSCO 1                     
[00:00:00.584] LOG -- hci.c.1894: BR/EDR support 1, LE support 1                
[00:00:00.593] LOG -- hci.c.3971: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1          
[00:00:00.597] LOG -- hci.c.1844: hci_le_read_buffer_size: size 251, count 10   
[00:00:00.603] LOG -- hci.c.1713: Supported commands 30                         
[... 103 lines cut ...]
[00:00:10.903] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, fla1
[00:00:10.914] LOG -- rfcomm.c.1762: Sending MSC RSP for #2                     
[00:00:10.920] LOG -- rfcomm.c.1925: Sending MSC CMD for #2                     
[00:00:10.925] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, fla1
[00:00:10.935] LOG -- rfcomm.c.1512: Received MSC RSP for #2                    
[00:00:10.941] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, fla1
[00:00:10.951] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 2, now 3  
[00:00:10.958] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, fla3
[00:00:10.969] LOG -- rfcomm.c.1932: Providing credits for #2                   
[00:00:10.975] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, fla3
[00:00:10.985] LOG -- rfcomm.c.1266: rfcomm_channel_opened!                     
[00:00:10.990] LOG -- rfcomm.c.169: RFCOMM_EVENT_CHANNEL_OPENED status 0x0 addr4
[00:00:11.002] LOG -- hsp_ag.c.657: RFCOMM_EVENT_CHANNEL_OPENED packet_handler 0
[00:00:11.011] LOG -- hsp_ag.c.668: RFCOMM channel open succeeded. New RFCOMM C3
RFCOMM connection established.                                                  
Received speaker gain change 8                                                  
[00:00:11.040] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3  
Received microphone gain change 15                                              
[00:00:11.051] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3  
Connect to 48:C1:AC:C3:4F:70                                                    
[00:00:23.436] LOG -- hsp_ag.c.551: Received button press AT+CKPD=200           
[00:00:23.465] LOG -- hci.c.2048: Synchronous Connection Complete (status=26) 40
[00:00:23.465] LOG -- hsp_ag.c.602: (e)SCO Connection failed, status 26         
Audio connection establishment failed with status 26                            
[00:00:23.474] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3

mringwal, 2018-08-22:
Hi. Please enable HCI packet log in app_main of components/btstack/main.c and post a new log.

SCO doesn't work with BTstack on ESP32, see https://github.com/espressif/esp-idf/issues/1118

sapthas, 2018-08-23:
Hi, 
I am new to btstack. is there any method to enable hsp.
I commented the  sco_demo_init(); function from Bt stack main .

here is the new log

BTstack: setup
[00:00:00.032] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffcf3e8
[00:00:00.034] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue item size 8
[00:00:00.042] LOG -- main.c.272: transport_register_packet_handler
[00:00:00.051] LOG -- btstack_tlv.c.48: TLV Instance 0x3f407290
[00:00:00.053] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.060] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.065] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.071] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.077] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.083] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.089] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.095] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.101] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.107] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.113] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.119] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.125] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[00:00:00.131] LOG -- btstack_tlv_esp32.c.68: read tag 42544406
[00:00:00.137] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407
[00:00:00.143] LOG -- btstack_tlv_esp32.c.68: read tag 42544407
[00:00:00.149] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408
[00:00:00.155] LOG -- btstack_tlv_esp32.c.68: read tag 42544408
[00:00:00.160] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409
[00:00:00.167] LOG -- btstack_tlv_esp32.c.68: read tag 42544409
[00:00:00.172] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A
[00:00:00.179] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A
[00:00:00.184] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B
[00:00:00.190] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B
[00:00:00.196] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C
[00:00:00.202] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C
[00:00:00.208] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D
[00:00:00.214] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D
[00:00:00.220] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E
[00:00:00.226] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E
[00:00:00.232] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F
[00:00:00.238] LOG -- btstack_tlv_esp32.c.68: read tag 4254440F
[00:00:00.244] LOG -- le_device_db_tlv.c.160: num valid le device entries 0
[00:00:00.250] LOG -- l2cap.c.3400: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
SDP service record size: 82
[00:00:00.260] LOG -- rfcomm.c.2301: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10
[00:00:00.269] LOG -- l2cap.c.3400: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
hsp ag init completed------|[00:00:00.277] LOG -- hci.c.2800: hci_power_control: 1, current mode 0
[00:00:00.285] LOG -- main.c.204: transport_init
[00:00:00.289] LOG -- main.c.220: transport_open
I (293) BTDM_INIT: BT controller compile version [9525061]

I (300) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (522) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
[00:00:00.523] LOG -- main.c.251: transport: configure SCO over HCI, result 0x0000
[00:00:00.525] LOG -- hci.c.3859: BTSTACK_EVENT_STATE 1
hci power controlee is enabledBTstack: execute run loop
[00:00:00.536] LOG -- hci.c.1902: Manufacturer: 0x0060
[00:00:00.540] LOG -- hci.c.1816: local name: 
[00:00:00.544] LOG -- hci.c.1605: Received local name, need baud change 0
[00:00:00.551] LOG -- hci.c.1912: Local supported commands summary 0x3f
[00:00:00.558] LOG -- hci.c.1864: Local Address, Status: 0x00: Addr: 30:AE:A4:0A:63:42
[00:00:00.565] LOG -- hci.c.1833: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:00.576] LOG -- hci.c.1891: Packet types cc18, eSCO 1
[00:00:00.581] LOG -- hci.c.1894: BR/EDR support 1, LE support 1
[00:00:00.590] LOG -- hci.c.3971: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[00:00:00.594] LOG -- hci.c.1844: hci_le_read_buffer_size: size 251, count 10
[00:00:00.600] LOG -- hci.c.1713: Supported commands 30
[00:00:00.605] LOG -- hci.c.1851: hci_le_read_maximum_data_length: tx octets 251, tx time 2120 us
[... 101 lines cut ...]
[00:00:19.855] LOG -- rfcomm.c.1505: Received MSC CMD for #2, 
[00:00:19.861] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, flags needed 0020, current 1c1b, rf credits 1
[00:00:19.872] LOG -- rfcomm.c.1762: Sending MSC RSP for #2
[00:00:19.877] LOG -- rfcomm.c.1925: Sending MSC CMD for #2
[00:00:19.882] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, flags needed 0020, current 701b, rf credits 1
[00:00:19.893] LOG -- rfcomm.c.1512: Received MSC RSP for #2
[00:00:19.898] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, flags needed 0020, current 703b, rf credits 1
[00:00:19.909] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 2, now 3
[00:00:19.916] LOG -- rfcomm.c.1615: rfcomm_channel_ready_for_open state 8, flags needed 0020, current 703b, rf credits 3
[00:00:19.927] LOG -- rfcomm.c.1932: Providing credits for #2
[... cut ...]
[00:00:19.998] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3
Received microphone gain change 15
[00:00:20.008] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3
[00:00:23.570] LOG -- hsp_ag.c.551: Received button press AT+CKPD=200
[00:00:23.596] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3
[00:00:23.597] LOG -- hci.c.2048: Synchronous Connection Complete (status=26) 48:C1:AC:C3:4F:70
[00:00:23.600] LOG -- hsp_ag.c.602: (e)SCO Connection failed, status 26
Audio connection establishment failed with status 26

mringwal, 2018-08-23:
Hi. Please enable HCI packet log in app_main of components/btstack/main.c by uncommenting the call to hci_dump_open(..) and post a new log.

SCO doesn't work with BTstack on ESP32, see espressif/esp-idf#1118

hsp_ag / hfp_ag works on most other platforms, e.g. stm32f4-discovery-cc2564.

sapthas, 2018-08-30:
Hi, sorry for the delay. I uncommented the call to hci_dump_open(..). Please follow the log

BTstack: setup                                                                  
[00:00:00.032] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffcf3e8 
[00:00:00.034] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue ite8
[00:00:00.042] LOG -- main.c.272: transport_register_packet_handler             
[00:00:00.051] LOG -- btstack_tlv.c.48: TLV Instance 0x3f407290                 
[00:00:00.053] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400          
[00:00:00.060] LOG -- btstack_tlv_esp32.c.68: read tag 42544400                 
[00:00:00.065] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401          
[00:00:00.071] LOG -- btstack_tlv_esp32.c.68: read tag 42544401                 
[00:00:00.077] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402          
[00:00:00.083] LOG -- btstack_tlv_esp32.c.68: read tag 42544402                 
[00:00:00.089] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403          
[00:00:00.095] LOG -- btstack_tlv_esp32.c.68: read tag 42544403                 
[00:00:00.101] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404          
[00:00:00.107] LOG -- btstack_tlv_esp32.c.68: read tag 42544404                 
[00:00:00.113] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405          
[00:00:00.119] LOG -- btstack_tlv_esp32.c.68: read tag 42544405                 
[00:00:00.125] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406          
[00:00:00.131] LOG -- btstack_tlv_esp32.c.68: read tag 42544406                 
[00:00:00.137] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407          
[00:00:00.143] LOG -- btstack_tlv_esp32.c.68: read tag 42544407                 
[00:00:00.149] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408          
[00:00:00.155] LOG -- btstack_tlv_esp32.c.68: read tag 42544408                 
[00:00:00.160] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409          
[00:00:00.167] LOG -- btstack_tlv_esp32.c.68: read tag 42544409                 
[00:00:00.172] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A          
[00:00:00.179] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A                 
[00:00:00.184] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B          
[00:00:00.190] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B                 
[00:00:00.196] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C          
[00:00:00.202] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C                 
[00:00:00.208] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D          
[00:00:00.214] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D                 
[00:00:00.220] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E          
[00:00:00.226] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E                 
[00:00:00.232] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F          
[00:00:00.238] LOG -- btstack_tlv_esp32.c.68: read tag 4254440F                 
[00:00:00.244] LOG -- le_device_db_tlv.c.160: num valid le device entries 0     
[00:00:00.250] LOG -- l2cap.c.3400: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535    
SDP service record size: 82                                                     
[00:00:00.260] LOG -- rfcomm.c.2301: RFCOMM_REGISTER_SERVICE channel #1 mtu 6550
[00:00:00.269] LOG -- l2cap.c.3400: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535    
hsp ag init completed------|[00:00:00.277] LOG -- hci.c.2800: hci_power_control0
[00:00:00.285] LOG -- main.c.204: transport_init                                
[00:00:00.289] LOG -- main.c.220: transport_open                                
I (293) BTDM_INIT: BT controller compile version [9525061]                      
                                                                                
I (300) system_api: Base MAC address is not set, read default base MAC address E
I (526) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0            
[00:00:00.527] LOG -- main.c.251: transport: configure SCO over HCI, result 0x00
[00:00:00.529] LOG -- hci.c.3859: BTSTACK_EVENT_STATE 1                         
hci power controlee is enabledBTstack: execute run loop                         
[00:00:00.540] LOG -- hci.c.1902: Manufacturer: 0x0060                          
[00:00:00.544] LOG -- hci.c.1816: local name:                                   
[00:00:00.548] LOG -- hci.c.1605: Received local name, need baud change 0       
[00:00:00.555] LOG -- hci.c.1912: Local supported commands summary 0x3f         
[00:00:00.561] LOG -- hci.c.1864: Local Address, Status: 0x00: Addr: 30:AE:A4:02
[00:00:00.569] LOG -- hci.c.1833: hci_read_buffer_size: ACL size module 1021 ->4
[00:00:00.580] LOG -- hci.c.1891: Packet types cc18, eSCO 1                     
[00:00:00.584] LOG -- hci.c.1894: BR/EDR support 1, LE support 1                
[00:00:00.594] LOG -- hci.c.3971: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1          
[00:00:00.598] LOG -- hci.c.1844: hci_le_read_buffer_size: size 251, count 10   
[00:00:00.604] LOG -- hci.c.1713: Supported commands 30                         
[00:00:00.609] LOG -- hci.c.1851: hci_le_read_maximum_data_length: tx octets 25s
[00:00:00.618] LOG -- hci.c.1857: hci_le_read_white_list_size: size 12          
[00:00:00.624] LOG -- hci.c.1469: hci_init_done -> HCI_STATE_WORKING            
[... cut ...]
Received microphone gain change 15                                              
[00:00:11.080] LOG -- rfcomm.c.1302: RFCOMM data UIH_PF, new credits: 1, now 3  
Connect to 48:C1:AC:C3:4F:70                                                    
Establish audio connection                                                      
[00:00:25.563] LOG -- hci.c.2048: Synchronous Connection Complete (status=26) 40
[00:00:25.563] LOG -- hsp_ag.c.602: (e)SCO Connection failed, status 26         
Audio connection establishment failed with status 26

mringwal, 2018-08-30:
hi. there are no hci packets in the log. Keep in mind that you need to modify the file in the esp-idf folder (not in the btstack checkout). Alternatively, you can modify it in btstack and then re-run ./integrate_btstack.py again.

sapthas, 2018-08-30:
Hi,
I modified and re-run the script that you mentioned. Please find the latest log below,
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4748
load:0x40078000,len:7304                                                        
load:0x40080400,len:7076                                                        
entry 0x40080748                                                                
I (202) cpu_start: Pro cpu up.                                                  
I (202) cpu_start: Single core mode                                             
I (202) heap_init: Initializing. RAM available for dynamic allocation:          
I (206) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM                       
I (212) heap_init: At 3FFCC3F0 len 00013C10 (79 KiB): DRAM                      
I (218) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM                    
I (224) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM                   
I (231) heap_init: At 4009236C len 0000DC94 (55 KiB): IRAM                      
I (237) cpu_start: Pro cpu start user code                                      
I (31) cpu_start: Starting scheduler on PRO CPU.                                
BTstack: setup                                                                  
[00:00:00.032] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffcf3e8 
[00:00:00.034] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue ite8
[00:00:00.042] LOG -- main.c.272: transport_register_packet_handler             
[00:00:00.051] LOG -- btstack_tlv.c.48: TLV Instance 0x3f407290                 
[00:00:00.053] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400          
[00:00:00.060] LOG -- btstack_tlv_esp32.c.68: read tag 42544400                 
[00:00:00.065] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401          
[00:00:00.071] LOG -- btstack_tlv_esp32.c.68: read tag 42544401                 
[00:00:00.077] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402          
[00:00:00.083] LOG -- btstack_tlv_esp32.c.68: read tag 42544402                 
[00:00:00.089] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403          
[00:00:00.095] LOG -- btstack_tlv_esp32.c.68: read tag 42544403                 
[00:00:00.101] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404          
[00:00:00.107] LOG -- btstack_tlv_esp32.c.68: read tag 42544404                 
[00:00:00.113] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405          
[00:00:00.119] LOG -- btstack_tlv_esp32.c.68: read tag 42544405                 
[00:00:00.125] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406          
[00:00:00.131] LOG -- btstack_tlv_esp32.c.68: read tag 42544406                 
[00:00:00.137] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407          
[00:00:00.143] LOG -- btstack_tlv_esp32.c.68: read tag 42544407                 
[00:00:00.149] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408          
[00:00:00.155] LOG -- btstack_tlv_esp32.c.68: read tag 42544408                 
[00:00:00.160] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409          
[00:00:00.167] LOG -- btstack_tlv_esp32.c.68: read tag 42544409                 
[00:00:00.172] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A          
[00:00:00.179] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A                 
[00:00:00.184] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B          
[00:00:00.190] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B                 
[00:00:00.196] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C          
[00:00:00.202] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C                 
[00:00:00.208] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D          
[00:00:00.214] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D                 
[00:00:00.220] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E          
[00:00:00.226] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E                 
[00:00:00.232] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F          
[00:00:00.238] LOG -- btstack_tlv_esp32.c.68: read tag 4254440F                 
[00:00:00.244] LOG -- le_device_db_tlv.c.160: num valid le device entries 0     
SCO Demo: Sending sine wave, hexdump received data.                             
[00:00:00.255] LOG -- l2cap.c.3401: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535    
SDP service record size: 90                                                     
[00:00:00.264] LOG -- rfcomm.c.2301: RFCOMM_REGISTER_SERVICE channel #1 mtu 6550
[00:00:00.274] LOG -- l2cap.c.3401: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535    
[00:00:00.281] LOG -- hci.c.2800: hci_power_control: 1, current mode 0          
[00:00:00.287] LOG -- main.c.204: transport_init                                
[00:00:00.291] LOG -- main.c.220: transport_open                                
I (295) BTDM_INIT: BT controller compile version [9525061]                      
                                                                                
I (302) system_api: Base MAC address is not set, read default base MAC address E
[... 95 lines cut ...]
[00:00:01.089] EVT <= 0E 04 05 0B 20 00                                         
[... cut ...]
[00:00:35.300] LOG -- l2cap.c.884: L2CAP_EVENT_CHANNEL_OPENED status 0x4 addr 40
[00:00:35.317] EVT <= 70 16 04 70 4F C3 AC C1 48 FF FF 01 00 40 00 00 00 9B 06  
[00:00:35.325] LOG -- sdp_client.c.448: SDP Client Connection failed, status 0x.
[00:00:35.332] LOG -- hsp_ag.c.728: Service not found, status 4.                
                                                                                
Audio connection establishment failed with status 4                             
[00:00:35.343] LOG -- l2cap.c.1053: l2cap_stop_rtx for local cid 0x40

mringwal, 2018-08-31:
Hi. In the log, you get a connection complete (status = 4) which is a Page Timeout. It usually means that the device you wanted to connect to is not here or off. Does your HF unit have the Bluetooth  address 48:C1:AC:C3:4F:70?

sapthas, 2018-09-01:
Hi, I got this message only 2 times. now I am getting an error message that is given below. I don't know what the status message means. how can I solve the issue? Right now I enabled hci.
I just past here the last part of the log message.

[00:00:10.505] LOG -- hci.c.2048: Synchronous Connection Complete (status=26) 40
[00:00:10.513] LOG -- hsp_ag.c.603: (e)SCO Connection failed, status 26         
Audio connection releasing failed with status 26

sapthas, 2018-09-04:
Hi,

I hope I provided enough information in my previous comment/query. Status26 means->dial string too long. 
Please let me know what could be causing this issue according to your analysis and how to tackle it.
Currently, my progress is stalled on account of this. 

Kindly let me know if you require more information regarding this.

sapthas, 2018-09-04:
Hi,
The remote device is Voyager Legend-Mobile Bluetooth Headset. ( https://www.plantronics.com/in/en/product/voyager-legend )

>> **Could you paste (at least) the part where the SCO connection is created incl. the HCI dump?**
Not sure if I understand this correctly but to add more information to my previous query, the only changes I did to the source code is that I changed device name and mac ID. I also uncommented the call to hci_dump_open(..)

Does it require any more changes other than the ones mentioned above to make this work?

mringwal, 2018-09-04:
Not really, but you did only paste the  error message and not the HCI data

sapthas, 2018-09-04:
Hi,
here is the full log :
I (201) cpu_start: Pro cpu up.
I (201) cpu_start: Single core mode
I (202) heap_init: Initializing. RAM available for dynamic allocation:
I (205) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM                       
I (211) heap_init: At 3FFCC3F0 len 00013C10 (79 KiB): DRAM                      
I (217) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM                    
I (224) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM                   
I (230) heap_init: At 4009236C len 0000DC94 (55 KiB): IRAM                      
I (236) cpu_start: Pro cpu start user code                                      
I (31) cpu_start: Starting scheduler on PRO CPU.                                
BTstack: setup                                                                  
[00:00:00.032] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffcf3e8 
[00:00:00.034] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue ite8
[00:00:00.042] LOG -- main.c.272: transport_register_packet_handler             
[00:00:00.051] LOG -- btstack_tlv.c.48: TLV Instance 0x3f407290                 
[00:00:00.053] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400          
[00:00:00.060] LOG -- btstack_tlv_esp32.c.68: read tag 42544400                 
[00:00:00.065] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401          
[00:00:00.071] LOG -- btstack_tlv_esp32.c.68: read tag 42544401                 
[00:00:00.077] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402          
[00:00:00.083] LOG -- btstack_tlv_esp32.c.68: read tag 42544402                 
[00:00:00.089] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403          
[00:00:00.095] LOG -- btstack_tlv_esp32.c.68: read tag 42544403                 
[00:00:00.101] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404          
[00:00:00.107] LOG -- btstack_tlv_esp32.c.68: read tag 42544404                 
[00:00:00.113] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405          
[00:00:00.119] LOG -- btstack_tlv_esp32.c.68: read tag 42544405                 
[00:00:00.125] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406          
[00:00:00.131] LOG -- btstack_tlv_esp32.c.68: read tag 42544406                 
[00:00:00.137] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407          
[00:00:00.143] LOG -- btstack_tlv_esp32.c.68: read tag 42544407                 
[00:00:00.149] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408          
[00:00:00.155] LOG -- btstack_tlv_esp32.c.68: read tag 42544408                 
[00:00:00.160] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409          
[00:00:00.167] LOG -- btstack_tlv_esp32.c.68: read tag 42544409                 
[00:00:00.172] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A          
[00:00:00.179] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A                 
[00:00:00.184] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B          
[00:00:00.190] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B                 
[00:00:00.196] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C          
[00:00:00.202] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C                 
[00:00:00.208] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D          
[00:00:00.214] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D                 
[00:00:00.220] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E          
[00:00:00.226] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E                 
[00:00:00.232] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F          
[00:00:00.238] LOG -- btstack_tlv_esp32.c.68: read tag 4254440F                 
[00:00:00.244] LOG -- le_device_db_tlv.c.160: num valid le device entries 0     
SCO Demo: Sending sine wave, hexdump received data.                             
[00:00:00.255] LOG -- l2cap.c.3401: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535    
SDP service record size: 90                                                     
[00:00:00.264] LOG -- rfcomm.c.2301: RFCOMM_REGISTER_SERVICE channel #1 mtu 6550
[00:00:00.274] LOG -- l2cap.c.3401: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535    
[00:00:00.281] LOG -- hci.c.2800: hci_power_control: 1, current mode 0          
[00:00:00.287] LOG -- main.c.204: transport_init                                
[00:00:00.291] LOG -- main.c.220: transport_open                                
I (295) BTDM_INIT: BT controller compile version [9525061]                      
                                                                                
I (302) system_api: Base MAC address is not set, read default base MAC address E
I (527) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0            
[00:00:00.528] LOG -- main.c.251: transport: configure SCO over HCI, result 0x00
[00:00:00.530] LOG -- hci.c.3859: BTSTACK_EVENT_STATE 1                         
[00:00:00.535] EVT <= 60 01 01                                                  
[00:00:00.539] CMD => 03 0C 00                                                  
BTstack: execute run loop                                                       
[00:00:00.543] EVT <= 6E 00                                                     
[00:00:00.546] EVT <= 0E 04 05 03 0C 00                                         
[00:00:00.550] CMD => 01 10 00                                                  
[... 348 lines cut ...]
[... cut ...]
[00:00:16.061] LOG -- hsp_ag.c.314: HSP_RFCOMM_CONNECTION_ESTABLISHED
[00:00:16.061] CMD => 28 04 11 81 00 40 1F 00 00 40 1F 00 00 FF FF 60 00 FF 3F 00 
[00:00:16.066] EVT <= 6E 00 
[00:00:16.068] EVT <= 0F 04 00 05 28 04 
[00:00:16.072] EVT <= 2C 11 1A 81 01 70 4F C3 AC C1 48 02 24 06 B4 00 B4 00 02 
[00:00:16.079] LOG -- hci.c.2048: Synchronous Connection Complete (status=26) 48:C1:AC:C3:4F:70
[00:00:16.087] LOG -- hsp_ag.c.603: (e)SCO Connection failed, status 26
Audio connection establishment failed with status 26

mringwal, 2018-09-04:
that's interesting. the last time I've tried the ESP32 just crashes.  However, there are only outgoing SCO packets. Do you see something like 'transport: configure SCO over HCI, result..' in the log, too? 

On the headset, you should hear a crystal clear sine signal -- which you didn't.
However, the HCI dump slows the SCO connection down. You cant try without HCI logging.

How to get the audio into the ESP32 is up to you. I guess you could use an I2S encoder and connect that to the I2S interface of the ESP32. Or, you might be able to use its ADC.

sapthas, 2018-09-04:
Hi,
  Now I am also facing the same crash issue. How did you solve this issue?

sapthas, 2018-09-04:
Here is the log message

TBcAAAcAAAB0AQAA                                                                
bEj9PwBG/T9kSP0/                                                                
gEb9PwBI/T8AAAAA4D/8P+A//D9sSP0/2D/8PwIAAAA0OP0/NDj9P2xI/T8AAAAA                
FwAAAGg4/T9idENvbnRyb2xsZXIAAAAAAAAAAGRI/T8AAAAAIAQGABcAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFjV/D/A1fw/KNb8PwAAAAAAAAAA                
AQAAAAAAAAAQxUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                                
4UX9P5U+CEA0DAYAGsIBgMBG/T8BAAAAAQAAAAAAAAAYCPw/AQAAALAF/D+QPgiA                
oEb9PykAAAApAAAAEAAAAP////8AAAAA/f///wQAAAAFAAAAAAAAADk+CEBAPghA                
AAAAABAAAAD/////WBMIQJD5+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAIAF/D8EBvw/AQAAALgF/D/8hgKA4Eb9P9Q9CEAQR/0/                
YAX8PwMAAAAjAAYAHQAAADqmAYAQR/0/AAAAAADX+z//////wSAIQGAF/D8E2QhA                
AAAAAAzn+j8DAAAAIwAGABSdAYBAR/0//Lf7P2CD+z9DAAAAAAAAAAAAAAAAAAAA                
qMEBQEIAAADwlPs/AQAAAFBbBYBwR/0/lKQBQKSB+z+1hgiAcEf9PwAAAACkgfs/                
LQUAAGj9+j/Agvk/AQAAAGb7DoCQR/0/AQAAAAAAAAAfAAAAkAX8PwEAAACwBfw/                
d/wOgLBH/T84WwVAtCr8P1Q4/T8AAAAAXDj9PwAAAAAAAAAA0Ef9P6wq/D/cKvw/                
/////2xI/T8BAAAAAAAAAAAAAAAASP0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAADYKvw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAxI/T8AAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                                
cAr9PwAJ/T9oCv0/                                                                
AAn9PwAK/T+PC9R9FD78PxQ+/D9wCv0/DD78PxkAAABhTIurKX5YT3AK/T8AAAAA                
AAAAAGwG/T9JRExFAPG1hVWJ7t5e2skAAAAAAGgK/T8AAAAAIwAGAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFjV/D/A1fw/KNb8PwAAAAAAAAAA                
AQAAAAAAAAAQxUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADpv8                                
yB4IQNqWEUAwCgYARscIgMAJ/T8IAAAAAAAAACMABgBwCv0/AQAAAAAAAABEPfw/                
KD38PwAAAAAjAAYAeKAIgJAF/D8hAAYAsAX8PwAAAAAgCAYAgAX8PwAAAAAAAAAA                
AAAAAMEgCECQBfw/BNkIQJC7+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAHigCICQBfw/IQAGALAF/D8AAAAA4An9PwAAAAAAAAAA                
IwAGAHAK/T8BAAAAAAAAAAAAAAAACv0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADAr9PwAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                                                
cAT9P2AC/T9oBP0/                                                                
YAL9PwAE/T8NSgAAAD78PwA+/D9wBP0/+D38PxgAAADQfP0/0Hz9P3AE/T8AAAAA                
AQAAAGzy/D9tYWluADcRbmilv+I+98QAAAAAAGgE/T8AAAAAIQAGAAEAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFjV/D/A1fw/KNb8PwAAAAAAAAAA                
AQAAAAAAAAAQxUA/WH39P0gdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEAAAAAFlez                                
yB4IQPC/AEAwBgYASMMIgCAD/T8AAAAAIwYGACAGBgAQ+/w//////wAEAADgEwiA                
8AL9P8AI/D8NSgAAIAYGAAAAAAAAAAAAAAAAAAQAAADIbg6A4AL9P+DCAED2wgBA                
AAAAAMEgCEAAAAAABNkIQJC1+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAIAAAAQ+/w//////wAEAACm0QiAMAP9P+gI/D8NSgAA                
QHQOgFAD/T/oCPw//////zRs/T8AAAAAAAAAALAF/D+QMQ6AcAP9P9NNYhDA1fw/                
AAAAAAYAAAAAAAAAAAAAAK8pDoCgA/0/CgAAAAEAAAAAAAAAAAAAAAAAAADtDwAA                
BQAAAAAAAAAgAAAA7Q8AAEEgDYDAA/0/bCYOQAAAAADtDwAAjKL8P2gW/T8jAAYA                
AAAAAOAD/T8AAAAAAAAAAAAAAABwBP0/AQAAAAAAAAAAAAAAAAT9PwAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAwE/T8AAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==                        
5BT9P1AT/T/cFP0/                                                                
UBP9P3AU/T8AAAAA7D38P+w9/D/kFP0/5D38PxgAAAAUDP0/FAz9P+QU/T8MDP0/                
AQAAAOAM/T9UbXIgU3ZjAAozT6mkoh8AAAAAANwU/T8AAAAAIQAGAAEAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAFjV/D/A1fw/KNb8PwAAAAAAAAAA                
AQAAAAAAAAAQxUA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
[... cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAYPD8PwAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAIDw/D8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACGDwiAsDv+P4g9/D9EQPw/                
AAAAAAAAAACM8Pw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA                
AAAAAAAAAAAAAAAA                                                                
================= CORE DUMP END =================                               
CPU halted.

mringwal, 2018-09-04:
According to  Espressif, this should work, but it just crashes for me.

I doesn't work for me, feel free to ask on the existing issue for updates.

beckmx, 2019-10-21:
hello, I was reading the full log, is the HSP profile the one that setup a microphone for remote streaming right? and according to the log the esp32 is still missing some documentation?

mringwal, 2019-10-22:
HSP + HFP is about mono audio at 8 Khz. HFP may support Wide-band speech at 16 khz.
Either ESP32 cannot correctly send SCO data via the VHCI interface, or, they are not able to tell me or document on how to do it correctly.

Changes729, 2021-04-12:
Has the problem been solved? I also reported 0x1A error when setting up SCO using HFP

mringwal, 2021-04-12:
@Changes729 Please try with develop branch of BTstack and master of esp-idf. We didn't test in a while.

Changes729, 2021-04-14:
I saw that my cell phone uses enhanced_setup_synchronous_connection and the connection succeeds.

What's the difference between HCI_OPCODE_HCI_SETUP_SYNCHRONOUS_CONNECTION and HCI_OPCODE_HCI_ENHANCED_SETUP_SYNCHRONOUS_CONNECTION?

Changes729, 2021-04-14:
I know what's up. the device does not support SCO_PACKET_TYPES_2EV3. use SCO_PACKET_TYPES_EV3 is ok.

mringwal-logitech, 2021-04-14:
@Changes729 The HSP HS should use 'all packet types' in the accept synchronous connectioncall.

The enhanced version allows to specify the local routing, SCO over HCI vs. PCM/I2S, and for PCM/I2S, even to configure PCM/I2S. Only rather new controllers support the enhanced command and it is not relevant for the connection setup (if it works, it works with both commands).

mringwal, 2021-04-14:
@Changes729 The HSP HS should use 'all packet types' in the accept synchronous connectioncall.

The enhanced version allows to specify the local routing, SCO over HCI vs. PCM/I2S, and for PCM/I2S, even to configure PCM/I2S. Only rather new controllers support the enhanced command and it is not relevant for the connection setup (if it works, it works with both commands).

mringwal, 2021-07-08:
Closing this due to inactivity. :)

closed 2021-07-08 by mringwal