bluekitchen/btstack #162: Resolving empty BT address

darkyline, 2018-08-22:
Hello Matthias,

I am trying to run the hog_keyboard_demo on an embedded config with  a CSR chip, I can discover the keyboard on my phone and connect but the logs show that a device with a null addr and fails to resolve it.

Here are the logs concerned 
`[00:00:20.940] LOG -- sm.c.2903: sm: connected
[00:00:20.940] LOG -- sm.c.2916: New sm_conn, role slave
[00:00:20.940] LOG -- att_server.c.280: SM_EVENT_IDENTITY_RESOLVING_STARTED
[00:00:20.940] LOG -- sm.c.1860: LE Device Lookup: device 0/1
[00:00:20.940] LOG -- sm.c.1866: device type 255, addr: 00:00:00:00:00:00
[00:00:20.940] LOG -- sm.c.1881: LE Device Lookup: calculate AH
[00:00:20.940] LOG -- btstack_util.c.259: IRK    00000000000000000000000000000000
[00:00:21.360] LOG -- sm.c.1860: LE Device Lookup: device 0/1
[00:00:21.360] LOG -- sm.c.1866: device type 255, addr: 00:00:00:00:00:00
[00:00:21.360] LOG -- sm.c.1860: LE Device Lookup: device 1/1
[00:00:21.360] LOG -- sm.c.1893: LE Device Lookup: not found
[00:00:21.360] LOG -- att_server.c.296: SM_EVENT_IDENTITY_RESOLVING_FAILED
[00:00:21.600] LOG -- att_db.c.836: ATT_READ_BY_GROUP_TYPE_REQUEST: from 0001 to FFFF, buffer size 185, type:
[00:00:21.600] LOG -- btstack_util.c.225: 0x00, 0x28,
[00:00:22.020] LOG -- att_db.c.836: ATT_READ_BY_GROUP_TYPE_REQUEST: from 0033 to FFFF, buffer size 185, type:
[00:00:22.020] LOG -- btstack_util.c.225: 0x00, 0x28,
[00:00:22.440] LOG -- att_db.c.516: ATT_READ_BY_TYPE_REQUEST: from 0030 to 0032, type:
[00:00:22.440] LOG -- btstack_util.c.225: 0x03, 0x28,
[00:00:22.440] LOG -- att_db.c.315: att_validate_security. flags 0x0002 (=> security level 0, key size 0) authorized 0, authenticated 0, encryption_key_size 0
[00:00:22.860] LOG -- att_db.c.516: ATT_READ_BY_TYPE_REQUEST: from 0001 to 0003, type:
[00:00:22.860] LOG -- btstack_util.c.225: 0x00, 0x2A,`

Thanks

mringwal, 2018-08-22:
Hi Rabeb. SM_EVENT_IDENTITY_RESOLVING_FAILED isn't an error per se. It just means that BTstack could not find an existing pairing for that device. It's a problem if you've paired before and the pairing was stored (= bonded), and you're connecting again.

Did you bond before? What do you use for le_device_db?

darkyline, 2018-08-22:
I did bond before but I got the same results after trying "forget this device" (I am trying to pair it with my phone).

I tried comparing the logs with the example turning on my computer, I don't have logs coming from le_device_db

Here are all the logs

[log_keyboard.txt](https://github.com/bluekitchen/btstack/files/2310243/log_keyboard.txt)

darkyline, 2018-08-22:
[log_keyboard.pklg.zip](https://github.com/bluekitchen/btstack/files/2310294/log_keyboard.pklg.zip)

I am not working on an existing port. But I turned libusb on my computer (the logs I compared to)

mringwal, 2018-08-22:
Please turn on HCI logging as described in the manual. In summary, you need to call hci_dump_open(NULL, HCI_DUMP_STDOUT)

For bonding to work, you need to pick one of the provided le_device_db_*.c implementations.

Does it work in the libusb port?
How does your port look like? What happens?

darkyline, 2018-08-22:
I am using le_device_db_memory btw, if that is what you are asking

darkyline, 2018-08-22:
My bad, here are the logs.

[log_keyboard.pklg.zip](https://github.com/bluekitchen/btstack/files/2310386/log_keyboard.pklg.zip)

mringwal, 2018-08-22:
Thanks, now it's getting interesting. In the latest log, there is this:
[Jan 01 11:00:49.620]  [NOTE]  btstack_crypto.c.979: public key invalid ffffb380

what ECC library did you use? micro-ecc from the 3rd-party folder of BTstack or something else?

darkyline, 2018-08-22:
I used mbedtls :) 

`#define HAVE_MBEDTLS_ECC_P256`

mringwal, 2018-08-22:
Ok. BTstack uses micro-ecc by default - as it is malloc free.

Blind guess - I've mixed up the API for the public key validation. Will report back.

darkyline, 2018-08-22:
Okay, thank you for you time !

mringwal, 2018-08-22:
for now, you may disable support for LE Secure Connections in the le_keyboard_setup by removing SM_AUTHREQ_SECURE_CONNECTION from this call

sm_set_authentication_requirements(SM_AUTHREQ_SECURE_CONNECTION | SM_AUTHREQ_BONDING)

mringwal, 2018-08-22:
there's more. before this happens, we have this:
[Jan 01 11:00:10.380]  [NOTE]  btstack_crypto.c.430: gen keypair fffffffc

That's in src/btstack_crypto.c line 424-430. mbed TLS returns MBEDTLS_ERR_MPI_BAD_INPUT_DATA 

Any idea why this happens? BTstack provides it some random looking random data from the Bluetooth Controller. Do you need to configure mbed TLS for EC stuff?

darkyline, 2018-08-22:
I don't believe I do. On the other hand I had to change the btstack_crypto.c where it called mbedtls_ec_group 
i.g:     `int res = mbedtls_ecp_gen_keypair(&mbedtls_ec_group, &d, &P, &sm_generate_f_rng_mbedtls, NULL); `

which my compiler did not recognise, I checked in the included headers and in their library it is **mbedtls_ecp_gen_keypair**

darkyline, 2018-08-22:
The example works just fine now, it just keeps running even after I disconnect, the loop doesn't break. It might be a problem in my main.

mringwal, 2018-08-22:
sorry, I don't see what you had to change. Can you show me the diff or a before & after?

darkyline, 2018-08-22:
Before: 

```
#ifdef USE_MBEDTLS_ECC_P256
    mbedtls_mpi d;
    mbedtls_ecp_point P;
    mbedtls_mpi_init(&d);
    mbedtls_ecp_point_init(&P);
    int res = mbedtls_ecp_gen_keypair(&mbedtls_ec_group, &d, &P, &sm_generate_f_rng_mbedtls, NULL);
    log_info("gen keypair %x", res);
    mbedtls_mpi_write_binary(&P.X, &btstack_crypto_ecc_p256_public_key[0],  32);
    mbedtls_mpi_write_binary(&P.Y, &btstack_crypto_ecc_p256_public_key[32], 32);
    mbedtls_mpi_write_binary(&d, btstack_crypto_ecc_p256_d, 32);
    mbedtls_ecp_point_free(&P);
    mbedtls_mpi_free(&d);
#endif  /* USE_MBEDTLS_ECC_P256 */
}
```

After:
```
#ifdef USE_MBEDTLS_ECC_P256
    mbedtls_mpi d;
    mbedtls_ecp_point P;
    mbedtls_ecp_group G;
    mbedtls_mpi_init(&d);
    mbedtls_ecp_point_init(&P);
    int res = mbedtls_ecp_gen_keypair(&G, &d, &P, &sm_generate_f_rng_mbedtls, NULL);
    log_info("gen keypair %x", res);
    mbedtls_mpi_write_binary(&P.X, &btstack_crypto_ecc_p256_public_key[0],  32);
    mbedtls_mpi_write_binary(&P.Y, &btstack_crypto_ecc_p256_public_key[32], 32);
    mbedtls_mpi_write_binary(&d, btstack_crypto_ecc_p256_d, 32);
    mbedtls_ecp_point_free(&P);
    mbedtls_mpi_free(&d);
#endif  /* USE_MBEDTLS_ECC_P256 */
}

```

same for line 464 and 971

mringwal, 2018-08-22:
Before, it didn't compile because mbedtls_ec_group was not defined. After, it compiled, but it cannot work as G wasn't initialized. Please switch to the develop branch and use the latest version. I've fixed the code to provide and initialize mbedtls_ec_group.

(If you run into a compile error with our code, please file an issue)

closed 2018-08-22 by mringwal