bluekitchen/btstack #163: CC256xc: convert_bts_init_scripts.py doesn't generate proper HCI_VS_SET_POWER_VECTOR(EDR3)

ajraymond, 2018-08-22:
```
if not have_power_vector_gfsk:
    part_size += append_power_vector_gfsk(additions, str_list, data_indent)
    have_power_vector_gfsk = True; 
if not have_power_vector_edr2:
    part_size += append_power_vector_edr2(additions, str_list, data_indent)
    have_power_vector_edr2 = True;                            
if not have_power_vector_edr3:
/* BUG HERE, should be append_power_vector_edr3 */
    part_size += append_power_vector_edr2(additions, str_list, data_indent)
    have_power_vector_edr3 = True;                            
if not have_class2_single_power:
    part_size += append_class2_single_power(additions, str_list, data_indent)
    have_class2_single_power = True;
```

ajraymond, 2018-08-22:
I believe this means that the power limit set by `btstack_chipset_cc256x_set_power()` is not applied to the EDR3 modulation, as it should be overriden by `update_init_script_command()` when encountered in the init script, but the since this bug causes the absence of this modulation in the init script, it remains at its default value described in http://processors.wiki.ti.com/index.php/CC256x_VS_HCI_Commands#HCI_VS_DRPb_Set_Power_Vector_.280xFD82.29

This would mean that any previously-generated init script will keep the bug until regenerated (if ever).

mringwal, 2018-08-23:
Thanks for pointing this out. Interestingly, nobody (including me) spotted the duplication in the console output. I've committed the fix and left a note in the CHANGELOG.md.

```
make
Extracting cc256xb_bt_sp_v1.6/initscripts-TIInit_6.7.16_bt_spec_4.1.bts
Extracting cc256xb_bt_sp_v1.6/initscripts-TIInit_6.7.16_ble_add-on.bts
Creating bluetooth_init_cc2564B_1.6_BT_Spec_4.1.c
Main file: cc256xb_bt_sp_v1.6/initscripts-TIInit_6.7.16_bt_spec_4.1.bts
Add-on file: cc256xb_bt_sp_v1.6/initscripts-TIInit_6.7.16_ble_add-on.bts
- AKA TIInit_6.7.16.bts
- LMP Subversion: 0x1b90
Creating bluetooth_init_cc2564B_1.6_BT_Spec_4.1.c
- parsing cc256xb_bt_sp_v1.6/initscripts-TIInit_6.7.16_bt_spec_4.1.bts
- parsing cc256xb_bt_sp_v1.6/initscripts-TIInit_6.7.16_ble_add-on.bts
- part 1, size 9277
- total size 9277
- added HCI_VS_SET_POWER_VECTOR(GFSK) template
- added HCI_VS_SET_POWER_VECTOR(EDR2) template
- added HCI_VS_SET_POWER_VECTOR(EDR2) template
- added HCI_VS_SET_CLASS2_SINGLE_POWER template
```

commit 3342e387 (2018-08-23): cc256x: append power vector for edr3 correctly. fixes #163
closed 2018-08-23 by mringwal