bluekitchen/btstack #165: Non-compatible BT audio devices (BT v3.0/v4.2)

rob-bits, 2018-08-26:
Hi @mringwal ,

I was unable to use two types of bluetooth audio devices with the btstack on esp32 (esp-idf v3.1). I have already used this stack with three different vendors well but with LG PH1, and SONY WH500 I have some troubles. The BT devices what works uses BT v2.1 protocoll while the new devices use v3.0 or v4.2 and they do not seem work.

I am using this code:
[A2DP source](https://github.com/bluekitchen/btstack/blob/master/example/a2dp_source_demo.c)

With LG PH1 the devices connects and starts the stream and shows no error/ warning in the debug log. But no pleasurable audio comes out from the speaker. Instead of the music only some spike, loud, random noise can be heard. Like in every 1-2 sec this spike noise appear. It sounds like a codec problem. The speaker supports the SBC so it should work.

The other device, the SONY WH500 plays the music but with interruptions. It plays a bit of music for 1 sec then stops for another 1 sec and this is repeating. During audio play I got this message in debug:

`LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1`

Any hints what can be the problem or how can I track it back?

Thank you in advance!

Regards,

Rob

mringwal, 2018-08-27:
Hi @rob-bits  Interesting. Could you run a2dp_source on a desktop system (mac,linux,win) and post the full packet log?

For the LG, maybe they cannot agree on a codec config. If music pauses, you could check if the audio data is send at regular intervals. If the ESP32 runs a bit too slow, this would cause repeated pauses. -- just blind guesses.

mringwal, 2018-08-27:
FWIW, the Bluetooth spec version is not relevant for this. If anything, the A2DP or AVDTP spec might be relevant.

rob-bits, 2018-08-27:
Hi @mringwal,

Thank you very much for the input. Unfortunately I do not have time and energy to play on different platform with the BTStack for now. I can share the ESP32 log file and some measurement result. But if you really need the linux testing then I can do some test later.

I did a small modification for the a2dp_source_demo application to debug the problem. When the software enters into the a2dp_demo_send_media_packet() function then it toggles a GPIO pin. So when it first steps into this function then it sets the GPIO. When next time it enters it resets the GPIO ... and so on.

And you can see a scope shot from this GPIO for a normal operation (please ignore the orange spots on the screen :)):

[image]

And this is when the Sony WH500 is connected:

[image]

Log:

```
I (208) cpu_start: Pro cpu up.
I (208) cpu_start: Single core mode
I (209) heap_init: Initializing. RAM available for dynamic allocation:
I (212) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (218) heap_init: At 3FFCFA30 len 000105D0 (65 KiB): DRAM
I (224) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (230) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (237) heap_init: At 40091C20 len 0000E3E0 (56 KiB): IRAM
I (243) cpu_start: Pro cpu start user code
I (37) cpu_start: Starting scheduler on PRO CPU.
BTstack: setup
[00:00:00.038] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffd2a14
[00:00:00.040] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue item
 size 8
[00:00:00.048] LOG -- main.c.266: transport_register_packet_handler
[00:00:00.059] LOG -- btstack_tlv.c.48: TLV Instance 0x3f404820
[00:00:00.060] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.066] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.071] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.077] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.083] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.089] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.095] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.101] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.107] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.113] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.119] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.125] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.131] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[00:00:00.137] LOG -- btstack_tlv_esp32.c.68: read tag 42544406
[00:00:00.143] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407
[00:00:00.149] LOG -- btstack_tlv_esp32.c.68: read tag 42544407
[00:00:00.155] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408
[00:00:00.161] LOG -- btstack_tlv_esp32.c.68: read tag 42544408
[00:00:00.166] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409
[00:00:00.173] LOG -- btstack_tlv_esp32.c.68: read tag 42544409
[00:00:00.178] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A
[00:00:00.185] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A
[00:00:00.190] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B
[00:00:00.196] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B
[00:00:00.202] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C
[00:00:00.208] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C
[00:00:00.214] LOG -- btstack_tlv_esp32.c.54: tag 4254440d -> 4254440D
[00:00:00.220] LOG -- btstack_tlv_esp32.c.68: read tag 4254440D
[00:00:00.226] LOG -- btstack_tlv_esp32.c.54: tag 4254440e -> 4254440E
[00:00:00.232] LOG -- btstack_tlv_esp32.c.68: read tag 4254440E
[00:00:00.238] LOG -- btstack_tlv_esp32.c.54: tag 4254440f -> 4254440F
[00:00:00.244] LOG -- btstack_tlv_esp32.c.68: read tag 4254440F
[00:00:00.250] LOG -- le_device_db_tlv.c.157: num valid le device entries 0
[00:00:00.256] LOG -- l2cap.c.3404: L2CAP_REGISTER_SERVICE psm 0x19 mtu 65535
[00:00:00.263] LOG -- l2cap.c.3404: L2CAP_REGISTER_SERVICE psm 0x17 mtu 65535
[00:00:00.270] LOG -- l2cap.c.3404: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
loaded mod 'nao-deceased by disease', size 7600
[... 338 lines cut ...]
[00:00:31.706] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:32.806] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:36.205] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:36.755] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:37.536] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:38.075] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:38.801] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:39.406] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:40.000] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:41.045] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:43.641] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:44.411] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:44.950] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:45.621] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:46.490] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:47.656] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:48.635] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:48.976] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:49.075] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:49.790] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:51.891] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:52.606] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:52.705] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
[00:00:53.376] LOG -- hci.c.565: hci_stack->hci_packet_buffer_reserved  1
```

rob-bits, 2018-08-28:
HI @mringwal,

I have another issue with btstack+esp32. I got a new BT device which unfortunately does not work well. And it is the PEAQ PPA34BT-B.
It looks like it cannot set the proper security level for the connection. Sometimes it can be paired and connected to the esp32 and it plays the music. But sometimes it drops connection failed messages.

Any hint for this?

Thank you in advance!

Guessed problemetic part from log:
```
[00:00:09.305] LOG -- l2cap.c.1762: l2cap received remote supported features, se
c_level_0_allowed for psm 25 = 0
[00:00:09.315] LOG -- hci.c.4064: gap_request_security_level requested level 2,
planned level 0, current level 0
...
[00:00:09.513] LOG -- hci.c.3961: hci_emit_security_level 0 for handle 81
[00:00:09.513] LOG -- l2cap.c.2143: l2cap - security level update
[00:00:09.514] LOG -- l2cap.c.2153: channel state 6: actual 0 >= required 2?
[00:00:09.593] LOG -- l2cap.c.1058: l2cap_stop_rtx for local cid 0x41
[00:00:09.593] LOG -- l2cap.c.885: L2CAP_EVENT_CHANNEL_OPENED status 0x6a addr 0
...
```

Full log:

```
BTstack: execute run loop
[00:00:00.524] LOG -- hci.c.1907: Manufacturer: 0x0060
[00:00:00.529] LOG -- hci.c.1821: local name:
[00:00:00.533] LOG -- hci.c.1610: Received local name, need baud change 0
[00:00:00.540] LOG -- hci.c.1917: Local supported commands summary 0x3f
[00:00:00.546] LOG -- hci.c.1869: Local Address, Status: 0x00: Addr: 30:AE:A4:46
:D4:4A
[00:00:00.554] LOG -- hci.c.1838: hci_read_buffer_size: ACL size module 1021 ->
used 1021, count 9 / SCO size 255, count 4
[00:00:00.564] LOG -- hci.c.1896: Packet types cc18, eSCO 1
[00:00:00.569] LOG -- hci.c.1899: BR/EDR support 1, LE support 1
[00:00:00.578] LOG -- hci.c.3979: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[00:00:00.582] LOG -- hci.c.1849: hci_le_read_buffer_size: size 251, count 10
[00:00:00.589] LOG -- hci.c.1718: Supported commands 30
[00:00:00.594] LOG -- hci.c.1856: hci_le_read_maximum_data_length: tx octets 251
, tx time 2120 us
[00:00:00.602] LOG -- hci.c.1862: hci_le_read_white_list_size: size 12
[00:00:00.608] LOG -- hci.c.1474: hci_init_done -> HCI_STATE_WORKING
[00:00:00.614] LOG -- hci.c.3867: BTSTACK_EVENT_STATE 2
BTstack: up and running.
[00:00:07.082] LOG -- l2cap.c.1826: L2CAP_CREATE_CHANNEL addr 00:58:56:48:75:20
psm 0x1 mtu 1691 -> local mtu 1691
[00:00:07.082] LOG -- hci.c.3571: Create_connection to 00:58:56:48:75:20
[00:00:07.087] LOG -- hci.c.185: create_connection_for_addr 00:58:56:48:75:20, t
ype ff
[00:00:07.095] LOG -- hci.c.3583: conn state 0
b - Create AVDTP Source connection to addr 00:58:56:48:75:20, cid 0x56.
[00:00:09.001] LOG -- hci.c.2011: Connection_complete (status=0) 00:58:56:48:75:
20
[00:00:09.002] LOG -- hci.c.2024: New connection: handle 129, 00:58:56:48:75:20
[00:00:09.005] LOG -- hci.c.3930: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:00:09.011] LOG -- l2cap.c.1734: l2cap_handle_connection_complete expected st
ate
[00:00:09.019] LOG -- hci.c.2090: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPL
ETE, bonding flags 6, eSCO 1
[00:00:09.028] LOG -- l2cap.c.1762: l2cap received remote supported features, se
c_level_0_allowed for psm 1 = 1
[00:00:09.038] LOG -- l2cap.c.1058: l2cap_stop_rtx for local cid 0x40
[00:00:09.044] LOG -- l2cap.c.1067: l2cap_start_rtx for local cid 0x40
[00:00:09.135] LOG -- l2cap.c.2468: L2CAP signaling handler code 3, state 10
[00:00:09.135] LOG -- l2cap.c.1058: l2cap_stop_rtx for local cid 0x40
[00:00:09.137] LOG -- l2cap.c.1058: l2cap_stop_rtx for local cid 0x40
[00:00:09.143] LOG -- l2cap.c.1067: l2cap_start_rtx for local cid 0x40
[... 39 lines cut ...]
[00:00:09.419] LOG -- btstack_tlv_esp32.c.68: read tag 42544C06
[00:00:09.425] LOG -- btstack_tlv_esp32.c.54: tag 42544c07 -> 42544C07
[00:00:09.431] LOG -- btstack_tlv_esp32.c.68: read tag 42544C07
[00:00:09.437] LOG -- btstack_tlv_esp32.c.54: tag 42544c08 -> 42544C08
[00:00:09.443] LOG -- btstack_tlv_esp32.c.68: read tag 42544C08
[00:00:09.449] LOG -- btstack_link_key_db_tlv.c.95: tag 42544c08, addr 00:58:56:
48:75:20
[00:00:09.513] LOG -- hci.c.3961: hci_emit_security_level 0 for handle 81
[00:00:09.513] LOG -- l2cap.c.2143: l2cap - security level update
[00:00:09.514] LOG -- l2cap.c.2153: channel state 6: actual 0 >= required 2?
[00:00:09.593] LOG -- l2cap.c.1058: l2cap_stop_rtx for local cid 0x41
[00:00:09.593] LOG -- l2cap.c.885: L2CAP_EVENT_CHANNEL_OPENED status 0x6a addr 0
0:58:56:48:75:20 handle 0x81 psm 0x19 local_cid 0x41 remote_cid 0x0 local_mtu 16
91, remote_mtu 672, flush_timeout 0
[00:00:09.605] LOG -- avdtp.c.596: L2CAP_EVENT_CHANNEL_OPENED: status 106, cid 0
x41 , signaling connection 0x3ffd7170
[00:00:09.616] LOG -- avdtp.c.605: L2CAP connection to 00:58:56:48:75:20 failed.
 status code 0x6a
[00:00:09.624] LOG -- a2dp_source.c.240: AVDTP_SUBEVENT_SIGNALING_CONNECTION fai
led status 106 ---
A2DP Source: Connection failed, status 0x6a, cid 0x56, a2dp_cid 0x56
[00:00:09.639] LOG -- hci.c.894: Connection closed: handle 0x81, 00:58:56:48:75:
20
[00:00:09.646] LOG -- hci.c.3930: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
```

mringwal, 2018-08-28:
well. here the HCI log would help. Please enable the call to hci_dump_open in app_main and convert the log with tool/convert_packetlog.py to .pklg and post it here.

rob-bits, 2018-08-29:
Okay. At the eve I will do that.

BTW, after this security issue the L2CAP drops a 0x6A status code which defined as L2CAP_SERVICE_ALREADY_REGISTERED. 

Does this status code make any sense for you in this context?

mringwal, 2018-08-29:
L2CAP_SERVICE_ALREADY_REGISTERED doesn't make sense in this context. The log might shine some light on the details.

rob-bits, 2018-08-31:
Hi @mringwal ,

Have you find something?

Robert

mringwal, 2018-08-31:
Sorry, I did get to look at it yet. Thanks for the remainder.

mringwal, 2018-08-31:
hi. it looks like in the logs, there have been newlines inside some packets - the tool is not able to process that properly. Could you try to use a tool that does not add additional newlines?

With full packet log enabled, the console output at 115200, but it might help with headset/speakers that don't play music at all. 

Could you pick one device that doesn't play music and try to create a new log file without the newlines? You can open the .pklg in Wireshark and check if it looks fine. (undecoded events are ok, these are BTstack internal)