bluekitchen/btstack #166: HOG Keyboard arrows support

darkyline, 2018-08-27:
Hello Matthias,

I am trying to add arrows support to the HOG Keyboard example but nothing works, do I need to look somewhere in particular to add the mapping of the arrows ?

Thank you.

mringwal, 2018-08-27:
What did you try? The keycodes are defined in the "Universal Serial Bus (USB)
HID Usage Tables" http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

The HID Descriptor in the example allows for key codes 0x00-0xff. The arrow keys are 0x4f-0x53.

darkyline, 2018-08-27:
I tried adding these descriptors to the keytable but that didn't work.

mringwal, 2018-08-27:
Hm. Hard to tell. Please check if you you can see the arrow key codes in the outgoing HID reports.

darkyline, 2018-08-27:
Stupid question but where can I find that ?

mringwal, 2018-08-27:
That's the HCI Packet Logs we've used before. You can open them with macOS PacketLogger (part of Hardware tools accessible after free registration as Apple Developer) or Wireshark.

darkyline, 2018-08-27:
Oh yeah I checked that, here are the logs I get

[hci_dump (2).pklg.zip](https://github.com/bluekitchen/btstack/files/2323968/hci_dump.2.pklg.zip)

mringwal, 2018-08-27:
Ok, so, I don't see the values 0x4f-0x52 in the data of the notifications. Do you ? :)

closed 2018-08-29 by darkyline

Webunity, 2019-03-12:
@mringwal i compiled on my Raspberry Pi (libusb port), ran the example (hog_keyboard_demo) and when pressing the arrow keys or home, i see this in my terminal:
```
sending: t
sending: e
sending: s
sending: t
sending:
ending: [
sending: 1
sending: ~
```
The missing 's' on the 3rd to last line is strange. I am assuming keys like HOME, and arrow keys are not mapped correctly.

Second question if i may;
If i extend the reportmap of `hid_descriptor_keyboard_boot_mode` in with consumer control:
```
     /*
       Consumer Control
     */
    0x05, 0x0C,                     // Usage Page (Consumer Devices)
    0x09, 0x01,                     // Usage (Consumer Control)
    0xA1, 0x01,                     // Collection (Application)
    0x85, 0x02,                     //      Report ID
    0x75, 0x09, 0x01,               //      Report Size
    0x95, 0x09, 0x01,               //     Report Count
    0x15, 0x00,                     //      Logical Minimum (0)
    0x26, 0xFF, 0x07,               //      Logical Maximum (2047)
    0x19, 0x00,                     //      Usage Minimum (0)
    0x2A, 0xFF, 0x07,               //      Usage Maximum (2047)
    0x81, 0x00,                     //      Input (Data, Ary, Abs)
    0xC0,
```

How can i choose which reportId to send? [This btstack sample](https://github.com/MinatsuT/esp32_remote_shutter/blob/master/main/remote_shutter.c#L238) supports this, but this is BT classic i think?

mringwal, 2019-03-13:
quick follow up here. general discussion would be better on the mailing list.

the demo code took the path of least resistance and shows that it is working. for a product or so, the input would come from somewhere else anyway.

to handle cursor, you'd need to properly parse the input (e.g. the \033 = ESC sequence) using a state machine and then send the correct usage.

--

For HOG: the higs.gatt in src/ble/gatt-service only lists a single Input report. To use two reports, you could copy and modify the .gatt file for your project, but as the GATT DB is statically compiled, I don't see how we can make that more configurable. Alternatively, you could remove the Report ID from the HID descriptor and append the additional data at the end.

Webunity, 2019-03-13:
@mringwal thank you for your reply; i have asked the question in the google groups (is that the right starting point?)

Webunity, 2019-03-13:
> quick follow up here. general discussion would be better on the mailing list.
> 
> For HOG: the higs.gatt in src/ble/gatt-service only lists a single Input report. To use two reports, you could copy and modify the .gatt file for your project, but as the GATT DB is statically compiled, I don't see how we can make that more configurable. Alternatively, you could remove the Report ID from the HID descriptor and append the additional data at the end.

I know this specific question is not related to this issue, but a quick guidance is needed here. I hope you can answer me to unblock me. You mention to remove the report ID, and combine both into 1 reportmap, but keyboard and consumer control both have a different `report` (and corresponding `report_len`) to send; i wonder if that would work (although this would be the most easy solution).

If i go the long route would this be about the right path to take? (i have no C experience, only OO-oriented experience)

- Add an additional input report (with id 2) to the gatt file (`higs.gatt`) (or my project)
- Increase the reportid's for the output report (to 3) and the feature report (to 4) in `higs.gatt`
- Extend `hids_device_init` in `hids_device.c` to make sure the second report is loaded
- Extend `hids_device_send_input_report` with an parameter which input report to send
- modify my code accordingly, right?

Would that do the trick?

mringwal, 2019-03-14:
after removing the second report ID, you'll end up with both a report that consists of both reports concatenated. Just add the data from the consumer device after the data from the keyboard.

the other changes sound reasonable as well, but they are more work... :)

Webunity, 2019-03-14:
@mringwal I think i get what you are saying; in the meantime about the other route, [i've asked the question also on stackoverflow](https://stackoverflow.com/questions/55166120/combine-keyboard-consumer-control-in-descriptor); at least i now know why it is called "hid_descriptor_keyboard_boot_mode" :)

So what you are saying is that you are sending the report:
{ modifier, 0, 0, keycode, 0, 0, 0, 0, 0 }

Is actually composed of (looking at hid_descriptor_keyboard_boot_mode):
// Modifier byte (1 = input)
// Reserved byte (1 =input )
// LED report (size 1, but output so not in above list)
// Padding (size 3, but output so not in above list)
// Report Size (8 = input, but only 7 bytes remain?)

If i am correct, do i have to just add the consumer report inside the first collection? p.s. about the hids_device.c error, i will raise a pull request.

mringwal, 2019-03-18:
Well, yes. I'm not expert in HID but I think it works if you add it to the collection. In my mental model, it might also work as is, just removing the Report ID tag.