bluekitchen/btstack #170: Sniff mode support

Yanjiong, 2018-09-20:
Hi,

It seems sniff mode not support yet. Any plan or any branch to implement that?

Thanks.

mringwal, 2018-09-20:
hi. sniff isn't supported in gap/anywhere. It's possible to send the corresponding HCI Commands though.

In which scenario would you like to use it and what do you expect from the stack to do for this case?

Yanjiong, 2018-09-20:
hi. Like in hands free case or hid case, acl connection should stay in sniff mode most of time. It's part of pm strategy to minimum the power consumption.

From the stack's view, it should provide such mechanism and interface to upper profile to enable enter/exist sniff mode and maintain such state machine to align between multiple requests from upper layer profiles.

mringwal, 2018-09-20:
Thanks for your examples and sharing your expectations. 

The HFP profile did not mandate sniff mode (didn't see any tests related to it), but we just started HID Device testing, which has some sniff mode requirements - let's see what's needed to fulfil these.

Going over the HID Spec 1.1.1, it looks like it's enough to send a sniff mode command after both HID channels are opened.

If you need sniff mode for a commercial product, please get in touch with us directly at contact@bluekitchen-gmbh.com

mringwal, 2018-10-22:
We've added support for configuring sniff mode manually,.

You can use gap_set_default_link_policy_settings(4) to enable sniff mode and gap_sniff_mode_enter(..) to enter or gap_sniff_mode_exit(..) to exit sniff mode.

There's no automatic yet (e.g. enter if there's no data for x time), but we may consider adding some mechanism after getting more feedback on actual use cases.

mringwal, 2018-10-22:
btw.. the changes are only on the develop branch yet..

closed 2018-10-22 by mringwal