bluekitchen/btstack #172: Accessing rfcomm channel after free

skoperst, 2018-09-24:
I tried running valgrind on pbap_client_demo and got the following;

`=26135== Invalid read of size 4
==26135==    at 0x13A420: rfcomm_channel_ready_to_send (rfcomm.c:1647)
==26135==    by 0x13C83B: rfcomm_channel_state_machine_with_dlci (rfcomm.c:1384)
==26135==    by 0x13C9DE: rfcomm_channel_packet_handler (rfcomm.c:1466)
==26135==    by 0x13CCC7: rfcomm_packet_handler (rfcomm.c:1612)
==26135==    by 0x13D379: l2cap_dispatch_to_channel (l2cap.c:869)
==26135==    by 0x13FBBF: l2cap_acl_classic_handler (l2cap.c:3261)
==26135==    by 0x13FCBE: l2cap_acl_handler (l2cap.c:3372)
==26135==    by 0x11BB66: hci_emit_acl_packet (hci.c:3786)
==26135==    by 0x120FD6: acl_handler (hci.c:862)
==26135==    by 0x12105D: packet_handler (hci.c:2474)
==26135==    by 0x1173F5: handle_completed_transfer (hci_transport_h2_libusb.c:614)
==26135==    by 0x117806: usb_process_ds (hci_transport_h2_libusb.c:710)
==26135==  Address 0x98a6318 is 40 bytes inside a block of size 64 free'd
==26135==    at 0x4C30F78: free (vg_replace_malloc.c:530)
==26135==    by 0x11B3B3: btstack_memory_rfcomm_channel_free (btstack_memory.c:281)
==26135==    by 0x13B41B: rfcomm_channel_finalize (rfcomm.c:1366)
==26135==    by 0x13BE90: rfcomm_channel_state_machine_with_channel (rfcomm.c:2013)
==26135==    by 0x13C833: rfcomm_channel_state_machine_with_dlci (rfcomm.c:1383)
==26135==    by 0x13C9DE: rfcomm_channel_packet_handler (rfcomm.c:1466)
==26135==    by 0x13CCC7: rfcomm_packet_handler (rfcomm.c:1612)
==26135==    by 0x13D379: l2cap_dispatch_to_channel (l2cap.c:869)
==26135==    by 0x13FBBF: l2cap_acl_classic_handler (l2cap.c:3261)
==26135==    by 0x13FCBE: l2cap_acl_handler (l2cap.c:3372)
==26135==    by 0x11BB66: hci_emit_acl_packet (hci.c:3786)
==26135==    by 0x120FD6: acl_handler (hci.c:862)
`

After some debugging I tracked the problem with the rfcomm_channel_state_machine_with_dlci function.

After getting a valid channel:
`rfcomm_channel_t * channel = rfcomm_channel_for_multiplexer_and_dlci(multiplexer, dlci);`

It will run rfcomm_channel_state_machine_with_channel, which in this case will finilize and free the channel. 
Then calling rfcomm_channel_ready_to_send without checking that channel is already invalid while keeping previous pointer copy.

Any how to deal with that?

mringwal, 2018-09-24:
What packet was received at this point? Would you have a hci_dump.pklg for this?
How do I read the valgrind output? looks like the last line calls the one above until 'Address 0x98a6318 is 40 bytes inside a block of size 64 free'd", then it unwinds and calls handle_completed_transfer again?

skoperst, 2018-09-24:
I will try to get what packet/hci_dump tomorrow.

About reading the valgrind output, in case you compile with -g -Og (debug symbols + no optimization for debug) you can just run: sudo valgrind ./pbap_client_demo and see what happens right after disconnection. 

Why 40 bytes inside 64 freed, I think its because of:
```
static int rfcomm_channel_ready_to_send(rfcomm_channel_t * channel){
    switch (channel->state){
```
`state` seems about 40 bytes into `rfcomm_channel_t`. I didn't count exactly.

skoperst, 2018-09-24:
It will happen after all contacts were transferred (PBAP) and then I close the connection `pbap_disconnect(pbap_cid);`

There is some race problem in which the channel is closed but then an incoming message to the RFCOMM will try to go through and trigger the issue above

mringwal, 2018-09-24:
thanks for the feedback. I hope to have a look at it tomorrow. Just checked, current valgrind (3.13) starts on mac, but I get an "invalid read" right on start - I think it is still not usable and I need to fire up a Linux VM.

mringwal, 2018-09-25:
Valgrind worked great on LInux. I'd love to have it run as part of automated tests, but only have a server in a data center where there's no Bluetooth dongles... 

Anyway, thanks again for reporting. Whenever the channel struct was freed, it should not be accessed afterwards. Fixed in 8f904593d on develop.

closed 2018-09-25 by mringwal