bluekitchen/btstack #173: hfp_hf_establish_audio_connection() incomplete?

ajraymond, 2018-09-24:
Hi,

I am trying to establish an audio connection in HFP HF starting from a service level connection, using `hfp_hf_establish_audio_connection()`, but it doesn't appear to be doing anything.

I traced the code, and it looks like 
`hfp_hf_establish_audio_connection()` sets `hfp_connection->establish_audio_connection`, but `hfp_run_for_context()` doesn't do anything with it.

Whereas `hfp_ag_run_for_audio_connection()` calls `hfp_setup_synchronous_connection()` if `hfp_connection->establish_audio_connection` is set, `hfp_hf_run_for_audio_connection()` does not.

Am I missing something?

Alexandre

skoperst, 2018-09-24:
If you setup your HF to support CODEC_NEGOTIATION then the implementation of HF & AG audio connection establishment is different and reflecting the difference in the BT specifications(I think from HFP V1.6 but I'm not an expert).

When you call hfp_hf_establish_audio_connection your goal is to send AG a request to start audio connection and this why the only effect of it is set of the command to: HFP_CMD_TRIGGER_CODEC_CONNECTION_SETUP

Then codec exchange happen. 

If everything is cake, the audio connection will be setup next.

If I were you, I would remove MSBC (wide band) and remove codec-exchange from feature list to see if you can get a connection. Its little easier to debug that way.

ajraymond, 2018-09-24:
Thanks for your response.

So the backstory is that I am connecting my HF to a new AG device and, unlike other AGs I am using, this one doesn't set-up the audio connection automatically. I am therefore trying to establish the audio link from the HS to see if that works.

I have already disabled mSBC and the remote AG doesn't support CODEC_NEGOTIATION, therefore I already go through this code in `hfp_hf_establish_audio_connection()`:
```
    if (!has_codec_negotiation_feature(hfp_connection)){
        CONSOLE_OUT("hfp_ag_establish_audio_connection - no codec negotiation feature, using defaults\n");
        hfp_connection->codecs_state = HFP_CODECS_EXCHANGED;
        hfp_connection->establish_audio_connection = 1;
    }
```

My question was thus regarding `hfp_connection->establish_audio_connection = 1`; it isn't used by any HF code (only AG code), which I found suspicious.

skoperst, 2018-09-24:
Thanks for clearing this,

You might have a legacy device then, because in modern devices AG is responsible to start the synchronous connection although as an HF you can trigger it (Ex. by dialing (hfp_hf_dial_number)).
I didn't try but it might be interesting to run: `hfp_setup_synchronous_connection` directly from your code after the establishment, to see if it can actually open such connection.

About the `hfp_connection->establish_audio_connection = 1;`, I think you're right, this looks obsolete.

ajraymond, 2018-09-25:
If I attempt to start the connection manually from the HF using `hfp_setup_synchronous_connection()`, I get

```
Create connection HFP_LINK_SETTINGS_D0
max_latency=65535 voice_setting=96 retransmission_effort=255 packet_types=961

(e)SCO Connection failed status 0x0d
```

What is strange is that the service link is properly established, but the AG doesn't establish the synchronous link...
```
HFP_TX AT+BRSF=0
HFP_RX 
+BRSF:0
HFP_RX 
OK

HFP_TX AT+CIND=?
HFP_RX 
OK

HFP_TX AT+CIND?
HFP_RX 
OK

HFP_TX AT+CMER=3,0,0,1
HFP_RX 
OK

Service level connection established aa:bb:cc:dd:ee:ff.

Speaker volume: 9
Microphone volume: 9
HFP_TX AT+VGM=9
HFP_RX 
OK

HFP_TX AT+VGS=9
HFP_RX 
OK
```

mringwal, 2018-09-26:
hi. hfp_hf_establish_audio_connection() should trigger the codec connection exchange via AT+BCC. Does it do that?

The SCO setup is tricky as there are different link settings and the stack need to try all them (best to worst) one at a time. (If the remote doens't support eSCO, then eSCO is skipped).

Could you post a packet log for the case where you call hfp_hf_establish_audio_connection() to see if the AG fails or the stack?

ajraymond, 2018-09-27:
My understanding is that since the AG answers `+BRSF:0`, `has_codec_negotiation_feature(hfp_connection)` is false, and the codec connection exchange doesn't run.

```
    if (!has_codec_negotiation_feature(hfp_connection)){
        log_info("hfp_ag_establish_audio_connection - no codec negotiation feature, using defaults");
        hfp_connection->codecs_state = HFP_CODECS_EXCHANGED;
        hfp_connection->establish_audio_connection = 1;
    } else {
        switch (hfp_connection->codecs_state){
            case HFP_CODECS_W4_AG_COMMON_CODEC:
                break;
            default:
                hfp_connection->command = HFP_CMD_TRIGGER_CODEC_CONNECTION_SETUP;
                break;
        } 
    }
```

[hci_dump_20180927_0954.pklg.gz](https://github.com/bluekitchen/btstack/files/2424381/hci_dump_20180927_0954.pklg.gz)

ajraymond, 2018-10-01:
Actually, this appears to be two loosely-related issues.

I'll leave this bug open for the hfp_hf_establish_audio_connection being incomplete if the AG doesn't support codec negotiations.

And I'll open another one for the 0x0d error when attempting to open a synchronous link.

mringwal, 2018-10-02:
Thanks for digging deeper and even managing the sub-issues here.

Going over the HFP 1.5, 1.6, and 1.7 specs again, we see the following:
- 1.7 does not specify what should happen when AG doesn't support Codec Negotiation
- 1.5 - before Codec Negotation was added - shows the HF opening the SCO connection as suggested by you, as well as no AT commands

Combining both facts, we'll update the code to trigger SCO setup  on HFP and notify here when done.

mringwal, 2018-10-02:
Please update to the current develop branch. hfp_hf_establish_audio_connection does what's expected and even reports an audio connection establishment with status 0x0d if it fails after trying all relevant settings. I'm closing this one (BTstack behaves as expected, although it doesn't help)

closed 2018-10-02 by mringwal

ajraymond, 2018-10-02:
I am checking the code in the develop branch, and I don't see how things can work.

Just to be sure, we are talking about the case where codec negotiation isn't supported?

hfp_hf_establish_audio_connection():
```
...
if (!has_codec_negotiation_feature(hfp_connection)){
        log_info("hfp_ag_establish_audio_connection - no codec negotiation feature, using defaults");
        hfp_connection->codecs_state = HFP_CODECS_EXCHANGED;
        hfp_connection->establish_audio_connection = 1;
    }
...
```

establish_audio_connection isn't hooked up to anything in hfp_hf.c

mringwal, 2018-10-02:
Sorry, my bad. I thought my colleague did already push her changes, but this didn't happen. I'll ping you once that has happened. In any case, the code to setup the sco connection is moved to hfp.c to be available for both roles.

mringwal, 2018-10-02:
Ok. Pushed.