bluekitchen/btstack #174: Fix compilation error and warnings in libusb port

ajraymond, 2018-09-26:
```
In file included from ../../platform/posix/btstack_link_key_db_fs.c:45:0:
../../3rd-party/tinydir/tinydir.h:92:28: error: ‘PATH_MAX’ undeclared here (not in a function); did you mean ‘RAND_MAX’?
 # define _TINYDIR_PATH_MAX PATH_MAX
                            ^
../../3rd-party/tinydir/tinydir.h:189:23: note: in expansion of macro ‘_TINYDIR_PATH_MAX’
  _tinydir_char_t path[_TINYDIR_PATH_MAX];
                       ^~~~~~~~~~~~~~~~~
../../3rd-party/tinydir/tinydir.h: In function ‘tinydir_open_subdir_n’:
../../3rd-party/tinydir/tinydir.h:609:18: warning: unused variable ‘path’ [-Wunused-variable]
  _tinydir_char_t path[_TINYDIR_PATH_MAX];
                  ^~~~
../../3rd-party/tinydir/tinydir.h: In function ‘tinydir_file_open’:
../../3rd-party/tinydir/tinydir.h:638:18: warning: unused variable ‘dir_name_buf’ [-Wunused-variable]
  _tinydir_char_t dir_name_buf[_TINYDIR_PATH_MAX];
                  ^~~~~~~~~~~~
<builtin>: recipe for target 'btstack_link_key_db_fs.o' failed
make: *** [btstack_link_key_db_fs.o] Error 1
```

_Fix_
3rd-party/tinydir/tinydir.h needs:
```
# include <linux/limits.h>
```

------------------

```
cc -I. -I../../src/ble -I../../src/classic -I../../src -I../../3rd-party/hxcmod-player -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -g -std=c99 -Wall -Wmissing-prototypes -Wstrict-prototypes -Wshadow -Wunused-parameter -Wredundant-decls -Wsign-compare -I../../platform/posix -I../../platform/embedded -I../../3rd-party/tinydir -I/usr/include/libusb-1.0   -c -o hci_dump.o ../../src/hci_dump.c
../../src/hci_dump.c: In function ‘hci_dump_open’:
../../src/hci_dump.c:109:21: warning: implicit declaration of function ‘fileno’; did you mean ‘fopen’? [-Wimplicit-function-declaration]
         dump_file = fileno(stdout);
                     ^~~~~~
                     fopen
../../src/hci_dump.c: In function ‘hci_dump_packet’:
../../src/hci_dump.c:206:13: warning: implicit declaration of function ‘ftruncate’; did you mean ‘strncat’? [-Wimplicit-function-declaration]
             ftruncate(dump_file, 0);
             ^~~~~~~~~
             strncat
```

_Fix_
src/hci_dump.c needs, before all includes:
```
#define _POSIX_C_SOURCE  200112L
```

mringwal, 2018-09-26:
Thanks. could you check the current version of develop branch?

I've added the _POSIX_C_SOURCE in 359cfe47123c7dd8481c6a0ca8c7552805b72871 and patched tinydir.h in ec8ca29f14e3b738f9bc53e3f9b016e8a472ef04

ajraymond, 2018-09-26:
Indeed, the latest develop compiles cleanly.

closed 2018-09-26 by ajraymond