bluekitchen/btstack #175: HFP: Error 0x0d (connection rejected due to limited resources) when HF initiates a synchronous connection

ajraymond, 2018-10-01:
This was initially discussed in #173 .

I am trying to connect to a commercial AG device using the hfp_hf_demo code.

I have tested this AG with a commercial hardware HF (MPOW headset) device successfully, so it works.

The issue I am facing is that the AG doesn't establish a synchronous connection after the service link is established. After implementing code to establish a synchronous connection from the HF when the AG doesn't set it up itself, I am faced with error 0x0d.

My code modifications are enclosed.
[issue.patch.txt](https://github.com/bluekitchen/btstack/files/2435132/issue.patch.txt)

I have tested that this modified code does establish a SCO connection to my Android phone acting as an AG.

**Analysis**

The service-level connection is established correctly, as far as i can tell (Section 4.2.1 of HFP Spec v1.6):
```
316	29.918690			PKTLOG	32	hfp.c.130: HFP_TX AT+BRSF=948  
323	29.957084			PKTLOG	41	hfp.c.1556: HFP_HF_RX: '\r\n+BRSF:0\r\n'
327	29.960366			PKTLOG	36	hfp.c.1556: HFP_HF_RX: '\r\nOK\r\n'
329	29.960384			PKTLOG	30	hfp.c.130: HFP_TX AT+CIND=?  
338	29.998395			PKTLOG	36	hfp.c.1556: HFP_HF_RX: '\r\nOK\r\n'
340	29.998423			PKTLOG	29	hfp.c.130: HFP_TX AT+CIND?  
349	30.025281			PKTLOG	36	hfp.c.1556: HFP_HF_RX: '\r\nOK\r\n'
351	30.025319			PKTLOG	36	hfp.c.130: HFP_TX AT+CMER=3,0,0,1  
360	30.053169			PKTLOG	36	hfp.c.1556: HFP_HF_RX: '\r\nOK\r\n'
```

The AG doesn't support codec negotiation:
```
323	29.957084			PKTLOG	41	hfp.c.1556: HFP_HF_RX: '\r\n+BRSF:0\r\n'
```

The AG supports eSCO:
```
hci.c.2126: HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 6, eSCO 1
```

I have disabled ENABLE_HFP_WIDE_BAND_SPEECH to simplify things.

I have hard-coded the HFP link setting to S1 in `hfp.c:hfp_setup_synchronous_connection()`, which is described as "safe" in the HFP 1.6 spec for eSCO.
```
hfp.c.1503: HFP: sending hfp_setup_synchronous_connection, sco_voice_setting 0x60 latency=7 retrans=1 packets=968
```

sco_voice_setting is set to 0x60, which is:
* input coding: linear
* input data format: 2's complement
* input sample size: 16-bit
* linear_pcb_bit_pos: 0
* air coding format: CVSD

Request
```
Frame 394: 21 bytes on wire (168 bits), 21 bytes captured (168 bits)
    Encapsulation type: PacketLogger (118)
    Arrival Time: Oct  1, 2018 14:21:50.370335000 EDT
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1538418110.370335000 seconds
    [Time delta from previous captured frame: 0.000025000 seconds]
    [Time delta from previous displayed frame: 0.000025000 seconds]
    [Time since reference or first frame: 39.104985000 seconds]
    Frame Number: 394
    Frame Length: 21 bytes (168 bits)
    Capture Length: 21 bytes (168 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: bluetooth:packetlogger:bthci_cmd]
Bluetooth
    [Source: host]
    [Destination: controller]
PacketLogger HCI Command
    Type: HCI Command (0x00)
Bluetooth HCI Command - Setup Synchronous Connection
    Command Opcode: Setup Synchronous Connection (0x0428)
    Parameter Total Length: 17
    Connection Handle: 0x0100
    Tx Bandwidth (bytes/s): 8000
    Rx Bandwidth (bytes/s): 8000
    Max. Latency (ms): 7
    0000 00.. .... .... = Unused bits: 0x00
    .... ..00 .... .... = Input Coding: Linear (0)
    .... .... 01.. .... = Input Data Format: 2's complement (1)
    .... .... ..1. .... = Input Sample Size: 16 bit (only for Linear PCM) (1)
    .... .... ...0 00.. = Linear PCM Bit Position: 0
    .... .... .... ..00 = Air Coding Format: CVSD (0)
    Retransmission Effort: At least 1 retransmission, optimize for power consumption (1)
    Packet Type: 0x03c8, 3-EV5, 2-EV5, 3-EV3, 2-EV3, EV3
    [Pending in frame: 396]
    [Command-Pending Delta: 2.364ms]
    [Response in frame: 398]
    [Command-Response Delta: 39.152ms]

```

Response:
```
Frame 398: 20 bytes on wire (160 bits), 20 bytes captured (160 bits)
    Encapsulation type: PacketLogger (118)
    Arrival Time: Oct  1, 2018 14:21:50.409487000 EDT
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1538418110.409487000 seconds
    [Time delta from previous captured frame: 0.024182000 seconds]
    [Time delta from previous displayed frame: 0.024182000 seconds]
    [Time since reference or first frame: 39.144137000 seconds]
    Frame Number: 398
    Frame Length: 20 bytes (160 bits)
    Capture Length: 20 bytes (160 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: bluetooth:packetlogger:bthci_evt]
Bluetooth
    [Source: controller]
    [Destination: host]
PacketLogger HCI Event
    Type: HCI Event (0x01)
Bluetooth HCI Event - Synchronous Connection Complete
    Event Code: Synchronous Connection Complete (0x2c)
    Parameter Total Length: 17
    Status: Connection Rejected due to Limited Resources (0x0d)
    Connection Handle: 0x0000
    BD_ADDR: Fidustro_01:44:c2 (00:1e:9c:01:44:c2)
    Link Type: eSCO connection (0x02)
    Transmit Interval: 0 slots (0 msec)
    Retransmit Window: 1 slots (0.625 msec)
    Rx Packet Length: 0
    Tx Packet Length: 0
    Air Mode: CVSD (2)
    [Command in frame: 394]
    [Pending in frame: 396]
    [Pending-Response Delta: 36.788ms]
    [Command-Response Delta: 39.152ms]
```

Full packet dump is here:
[hci_dump.pklg.txt](https://github.com/bluekitchen/btstack/files/2435139/hci_dump.pklg.txt)

PS: there is a typo in sco_demo_util.c (0x60 is actually 2's complement):
```
hci_set_sco_voice_setting(0x60); // linear, unsigned, 16-bit, CVSD
```

mringwal, 2018-10-02:
How do you trigger audio setup from HF (MPOW headset)?

Could you try D0 and D1 link settings as well? eSCO might be supported by the Bluetooth Controller, but the HFP Profile might not know/care about it. (If BTstack would handle the SCO setup fully, it would try all of them in decreasing order).

ajraymond, 2018-10-02:
Same problem with D0:

Request:
```
Frame 396: 21 bytes on wire (168 bits), 21 bytes captured (168 bits)
Bluetooth
PacketLogger HCI Command
Bluetooth HCI Command - Setup Synchronous Connection
    Command Opcode: Setup Synchronous Connection (0x0428)
    Parameter Total Length: 17
    Connection Handle: 0x0100
    Tx Bandwidth (bytes/s): 8000
    Rx Bandwidth (bytes/s): 8000
    Max. Latency (ms): 65535
    0000 00.. .... .... = Unused bits: 0x00
    .... ..00 .... .... = Input Coding: Linear (0)
    .... .... 01.. .... = Input Data Format: 2's complement (1)
    .... .... ..1. .... = Input Sample Size: 16 bit (only for Linear PCM) (1)
    .... .... ...0 00.. = Linear PCM Bit Position: 0
    .... .... .... ..00 = Air Coding Format: CVSD (0)
    Retransmission Effort: Don't Care (255)
    Packet Type: 0x03c1, 3-EV5, 2-EV5, 3-EV3, 2-EV3, HV1
    [Pending in frame: 398]
    [Command-Pending Delta: 1.152ms]
    [Response in frame: 400]
    [Command-Response Delta: 40.504ms]
```

Response:
```
Frame 400: 20 bytes on wire (160 bits), 20 bytes captured (160 bits)
Bluetooth
PacketLogger HCI Event
Bluetooth HCI Event - Synchronous Connection Complete
    Event Code: Synchronous Connection Complete (0x2c)
    Parameter Total Length: 17
    Status: Connection Rejected due to Limited Resources (0x0d)
    Connection Handle: 0x0000
    BD_ADDR: Fidustro_01:44:c2 (00:1e:9c:01:44:c2)
    Link Type: SCO connection (0x00)
    Transmit Interval: 0 slots (0 msec)
    Retransmit Window: 0 slots (0 msec)
    Rx Packet Length: 0
    Tx Packet Length: 0
    Air Mode: CVSD (2)
    [Command in frame: 396]
    [Pending in frame: 398]
    [Pending-Response Delta: 39.352ms]
    [Command-Response Delta: 40.504ms]
```

ajraymond, 2018-10-02:
Same problem with D1.

Request:
```
Frame 397: 21 bytes on wire (168 bits), 21 bytes captured (168 bits)
Bluetooth
PacketLogger HCI Command
Bluetooth HCI Command - Setup Synchronous Connection
    Command Opcode: Setup Synchronous Connection (0x0428)
    Parameter Total Length: 17
    Connection Handle: 0x0100
    Tx Bandwidth (bytes/s): 8000
    Rx Bandwidth (bytes/s): 8000
    Max. Latency (ms): 65535
    0000 00.. .... .... = Unused bits: 0x00
    .... ..00 .... .... = Input Coding: Linear (0)
    .... .... 01.. .... = Input Data Format: 2's complement (1)
    .... .... ..1. .... = Input Sample Size: 16 bit (only for Linear PCM) (1)
    .... .... ...0 00.. = Linear PCM Bit Position: 0
    .... .... .... ..00 = Air Coding Format: CVSD (0)
    Retransmission Effort: Don't Care (255)
    Packet Type: 0x03c4, 3-EV5, 2-EV5, 3-EV3, 2-EV3, HV3
    [Pending in frame: 399]
    [Command-Pending Delta: 2.464ms]
    [Response in frame: 401]
    [Command-Response Delta: 34.297ms]
```

Response:
```
Frame 401: 20 bytes on wire (160 bits), 20 bytes captured (160 bits)
Bluetooth
PacketLogger HCI Event
Bluetooth HCI Event - Synchronous Connection Complete
    Event Code: Synchronous Connection Complete (0x2c)
    Parameter Total Length: 17
    Status: Connection Rejected due to Limited Resources (0x0d)
    Connection Handle: 0x0000
    BD_ADDR: Fidustro_01:44:c2 (00:1e:9c:01:44:c2)
    Link Type: SCO connection (0x00)
    Transmit Interval: 0 slots (0 msec)
    Retransmit Window: 0 slots (0 msec)
    Rx Packet Length: 0
    Tx Packet Length: 0
    Air Mode: CVSD (2)
    [Command in frame: 397]
    [Pending in frame: 399]
    [Pending-Response Delta: 31.833ms]
    [Command-Response Delta: 34.297ms]
```

ajraymond, 2018-10-02:
> How do you trigger audio setup from HF (MPOW headset)?
I don't have to do anything on the MPOW headset apart from putting it in pairing mode. The AG connects to it and audio starts automatically.

mringwal, 2018-10-02:
While being aware that this sounds obvious: could you try to make the hfp_hf_demo look like your MPOW headset and wait until AG connects to it? :)

As for how it finds it: in Classic, there's mainly the CoD, so please use the same one as your MPOW headset.  Then, there could be an Extended Inquiry Response and/or the name.

Try to match it as close as possible.

mringwal, 2021-07-08:
Closing this due to inactivity. :)

closed 2021-07-08 by mringwal