bluekitchen/btstack #179: the a2dp_sink_demo can't save a wav file ?

WM-CH, 2018-10-09:
I see that “HAVE_POSIX_FILE_IO”  is defined in the btstack_config.h， which means that:
[a2dp_sink_demo.c]
#ifdef HAVE_POSIX_FILE_IO
#include "wav_util.h"
#define STORE_SBC_TO_SBC_FILE 
#define STORE_SBC_TO_WAV_FILE 
#endif

so, I cross-compile and run this a2dp_sink_demo on an arm-board
and connect to my cellphone through BT, then I paly some musice and hope that it will be a wav file created there.
but ,there is nothing...,only a wav file ocupying 44Bytes there. I think it's the head of the wav file.
but no PCM data behind.

WM-CH, 2018-10-09:
there are many messy code in my /tmp/hci_dump.pklg file.

WM-CH, 2018-10-09:
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/2459924/hci_dump.zip)

mringwal, 2018-10-09:
thanks for the logs. Nothing obviously wrong there. I did a quick test with the libusb port and my .wav file also was empty. We'll have a look and get back to you.

mringwal, 2018-10-09:
That's a side-effect of introducing btstack_audio. The btstack_audio implementation, when started, will request samples from the app, which are then decoded from the buffered SBC frames. In the example, the decoded samples are also stored in the wav. 

Hm.. we'll discuss if we a) add additional code to drive it without a btstack_audio implementation, b) implement a dummy btstack_audio that just asks for samples, or consider implementing a btstack_audio that writes to .wav.

Do you have a portaudio implementation for your target? :)

WM-CH, 2018-10-10:
Sorry, I didn't understand what you mean. Do you mean porting portAudio on ubuntu (PC) or porting it on arm-board?

In addition, I just touched btstack, I don't understand how to modify the code. Could you plz tell me clearly, what should I do?

Thank you

mringwal, 2018-10-10:
On Ubuntu, you can just install portaudio and use either port/libusb or port/posix-h4. At least in port/libusb/Makefile, the additional flags to link against portaudio are commented out. You could try that first to see if it works as expected.

What OS/target embedded hardware do you Assuming you want to actually play the received audio, you'd need to implement the btstack_audio_t interface for your target. There are examples for stm32-f4discovery, esp32, or posix systems with portaudio.

WM-CH, 2018-10-10:
I still want to save the generated wav file and then play it on my pc. How to do this   ; )

WM-CH, 2018-10-10:
I feel the handle_pcm_data function (in a2dp_sink_demo.c ) havn't be called.

mringwal, 2018-10-10:
correct. it will be called when decoding SBC frames, which currently is expected to be triggered by btstack_port_audio_t.

I'll try to update the example soon-ish. The idea could be to directly decode the received SBC data instead of putting it into a ring buffer.

mringwal, 2018-10-11:
Fixed. Please switch to develop branch and git pull. received SBC data is processed directly if no btstack_audio implementation is active.

closed 2018-10-11 by mringwal