bluekitchen/btstack #180: [GOEP] Wireshark malformed packet in some phones that stream PBAP contacts

skoperst, 2018-10-11:
In some mobile phones(Tested Galaxy S6 & J7) when pulling PBAP contacts using OBEX I'm getting corrupted information(ex. name missing a letter). 
Finally I did a test with Wireshark and noticed that the OBEX message is fragmented into 2 parts one being with size=1.

After some testing I got to this point and I think the mtu need to be subtracted 1. Then the wireshark will display the message as correct and no fragmenting observed.

https://github.com/bluekitchen/btstack/blob/2f98d3c60ab84a4e7ec4a33c528cd323900cd4a3/src/classic/goep_client.c#L468

mringwal, 2018-10-11:
Hi. Thanks for reporting and digging into. Could you post a .pklg here for analysis please?
Was this with 'L2CAP Support' or just RFCOMM?
I'd like to figure out if the mtu is calculated wrongly, or just bad luck and other phones might like to have 2 bytes less..

skoperst, 2018-10-11:
Its RFCOMM.
Will catch log later

closed 2019-07-15 by skoperst