bluekitchen/btstack #182: [FEATURE] Passing context variable via packet_handler

skoperst, 2018-11-25:
It might be useful to have some kind of void* context passing in packet_handler, similar to what context is for btstack_timer_source.

It will be useful when using the same packet handler with different profiles in an OO design (not C++ god forbid), but in some design patterns in C.

mringwal, 2018-11-26:
Thanks for the suggestion. For newer data structures, e.g. btstack_crypto_t and send requests, we have started to provide a context variable. In these setups, there is a triggering action that can request something and pass the context variable.

For the packet handlers, asides from the initial setup, it is triggered asynchronously, with no local source for the event. You could simulate a context variable by providing a specialized packet handler that you only register once and that adds the fixed context variable in the call to your centralized packet handler that needs such a context.

As for different profiles: they usually have distinct events, too, so having one packet handler per profile seems like a good idea anyway. 

Does this make sense? Do you have a killer use case? :)

skoperst, 2018-11-27:
Thanks for the answer. 
Your suggestion to centralized packet handler is useful, I did something similar but with one of the profiles being the centralized, I will change it too to use a specialized handler.

I will try to explain my point by an example:
The application is some sort of walkie-talkie that has 2 HFP connections (HF/AG can be both or two types of the same), both have PBAP connection. **Logically** each PBAPs depends on HFP connection(for example it should stop syncing when voice call is ongoing etc..) so the HFP need an API to manage the PBAP.

So to make life easier PBAP is a state machine and implemented in a different file, however I don't want this PBAP file to know anything about its higher layer(which uses it) to be able to write tests etc.., so when creating the PBAP dynamically I need to register it with a packet_handler that doesn't have the PBAP state machine object, so I make it global which is good unless you want to be able to create more than one PBAP task. The only option left is to pass the packet handler to the PBAP creator that will use it to register which I think could be nicer with a context, however right now I'm not so sure, so maybe the centralized is an even better solution.
Hope its understood.

Thanks for the support.

closed 2019-05-26 by skoperst