bluekitchen/btstack #183: ESP32 example with error

HotIce0, 2018-12-04:
I followed [http://bluekitchen-gmbh.com/btstack/ports/existing_ports/#btstack-port-for-the-espressif-esp32-platform](url) but some error occur.

run  `make`

```
CC build/bootloader_support/src/flash_qio_mode.o
CC build/bootloader_support/src/secure_boot.o
CC build/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader_support/libbootloader_support.a
CC build/bt/bt.o
AR build/bt/libbt.a
CC build/micro-ecc/micro-ecc/uECC.o
AR build/micro-ecc/libmicro-ecc.a
CC build/btstack/3rd-party/bluedroid/decoder/srce/alloc.o
make[1]: *** No rule to make target `3rd-party/bluedroid/decoder/srce/bitalloc-sbc.o', needed by `libbtstack.a'.  Stop.
make: *** [component-btstack-build] Error 2
```

HotIce0, 2018-12-04:
I solved this problem by copy merged 3rd-party/bluedroid/ into esp-idf/components/btstack

mringwal, 2018-12-04:
Hi. As far as I can tell, that's a bug in older versions of make. if you run make again after the error, the compile will succeed. This is mentioned in the ESP32 port readme you've referenced.

closed 2018-12-04 by mringwal