bluekitchen/btstack #184: classic HID keyboard example

joseffallman, 2018-12-31:
hello
I'm trying to use esp32 and btstack to create a simple enter button. This have to work for a old windows mobile 6.5 device with bluetooth below 2.0

When i'm trying the hid_keyboard_demo i can make a small change by adding "HCI_EVENT_PIN_CODE_REQUEST" in [packet_handler](https://bluekitchen-gmbh.com/btstack/profiles/#lst:PinCodeRequest) and then they can connect fine but i don't receive any keystrokes on the windows mobile 6.5 device. On my win 10 computer its working, and if i'm connecting a real hid keyboard to the windows mobile device its also working.

Any ideas of what could be wrong and where to start looking?

mringwal, 2019-01-03:
no idea what's wrong yet. HID is annoying as there's no error etc if it's not working.

What you can do is to enable packet log in main.c of the esp32 port file by uncommenting the call to hci_dump(..). You can then compare the log from your Windows 10 to the one from Windows Mobile.

For HID, there's the HID Descriptor that is read from the HID Host and then there should be both HID Control and Interrupt L2CAP channels getting created before HID events can be sent.

joseffallman, 2019-01-04:
Thanks!

After a change on line 227 from
`uint8_t report[] = { 0xa1, modifier, 0, 0, keycode, 0, 0, 0, 0, 0};`
to
`uint8_t report[] = { 0xa1, modifier, 0, keycode, 0, 0, 0, 0, 0};`
is it now working as expected.

I also noticed a real performance boost when i sending keystrokes to serial monitor and having the device connected to the same pc.

Could you try it out and update the example?

mringwal, 2020-07-15:
we've made some changes to hid_keyboard a while ago. Please open a new issue if you have problems with that.

closed 2020-07-15 by mringwal