bluekitchen/btstack #185: windows-winusb make with error

wslkh, 2019-01-12:
Hi, I tried to build the port 'windows-winusb' but failed:

> avrcp_browsing_controller.o: In function `avrcp_browsing_connect':
C:\msys64\home\lkh\btstack\port\windows-winusb/../../src/classic/avrcp_browsing_controller.c:167: undefined reference to `l2cap_create_ertm_channel'
avrcp_browsing_controller.o: In function `avrcp_browsing_controller_configure_incoming_connection':
C:\msys64\home\lkh\btstack\port\windows-winusb/../../src/classic/avrcp_browsing_controller.c:776: undefined reference to `l2cap_accept_ertm_connection'

I see the same function is used by other files without error. Any clue? Thanks.

mringwal, 2019-01-14:
AVRCP requires L2CAP Enhanced Retransmission Mode and it's not enabled in the windows-winusb port.

Please add #define ENABLE_L2CAP_ENHANCED_RETRANSMISSION_MODE to btstack_config.h and make clean && make

wslkh, 2019-01-16:
It works, thanks, mringwal.

closed 2019-01-16 by wslkh