bluekitchen/btstack #187: hid parser globals belong to the next usage

ricardoquesada, 2019-01-27:
using:
- master (hash: e034024d16933df0720cab3582d625294e26c667) Dec 7, 2018.
- libusb port

Problem:
I'm parsing a hid report like this:

```c
btstack_hid_parser_init(&parser, device->hid_descriptor, device->hid_descriptor_len, HID_REPORT_TYPE_INPUT, report, report_len);
    while (btstack_hid_parser_has_more(&parser)){
        btstack_hid_parser_get_field(&parser, &usage_page, &usage, &value);
        
       // <-- problem here:
       // parser.global_xxx variables belong to the next usage, and not the current one.
       // eg:
       // paser.global_logical_minimum / global_logical_maximum have the values
       // of the next usage, so process_usage() can't use them. 
        process_usage( parser, usage_page, usage, value);
    }
}
```

Test case:

HID Descriptor (which belongs to the Xbox Wireless Controller.  I also tried with other gamepads, and all trigger this bug)
```
05 01 09 05 A1 01 85 01 09 01 A1 00 09 30 09 31 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 09 01 A1 00 09 33 09 34 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 05 01 09 32 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0A 15 00 25 01 75 01 95 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 A1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 C0 05 0F 09 21 85 03 A1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0E 15 00 26 FF 00 75 08 95 01 91 02 09 A7 15 00 26 FF 00 75 08 95 01 91 02 65 00 55 00 09 7C 15 00 26 FF 00 75 08 95 01 91 02 C0 85 04 05 06 09 20 15 00 26 FF 00 75 08 95 01 81 02 C0
```

Report:
```
A1 01 EE 88 E8 7A AE 84 7E 85 00 00 00 00 00 00 00
```

mringwal, 2019-01-31:
thanks for reporting. I don't remember the details, but I think the idea was that the user of the API only gets usage_page, usage, and value for each field. The fields in the btstack_hid_parser_t should be opaque. 

Is { usage_page, usage, value } sufficient for your application? If not, what would you need more and how would you like to access it?

ricardoquesada, 2019-02-01:
I'm implementing a HID server, and currently I need the `logical_minimum` and `logical_maximum` values to know what the client is sending me, specially with gamepad axis.

Regarding how to access them, any way is fine.

And since you ask, it would be great to know even more than just the globals. For example, I'd like to know the "parent" usage... for example, I'd like to know if it belongs to a Gamepad or a Pointer. But I couldn't find a way to get that info with the current API.

Thanks!

ricardoquesada, 2022-05-08:
bringing back this old topic:

Ideally `btstack_hid_parser_get_field()` should do some kind of calculations with `logical min / max`, `physical min/max`, `unit exponent` and `unit.

But that will break backward compatibility, so it is not an option.
Another possibility is to add a new function like:

`btstack_hid_parser_get_filed_with_multiplier()` that does what `get_filed()` does in addition to apply the "resolution multiplier", as defined here in "HID Specification section 6.2.2.7 Global Items":

  `resolution = (logical max - logical min) / ((physical max - physical min) * (10 ^ unit exponent)`

Some extra calculation will be needed depending on the units being used: centimeters, inches, etc...
But normalizing the unit to milimeters could help.

ricardoquesada, 2022-05-09:
mmm... I thought I was going to need physical_min/max, unit and unit_exponent, but apparently not. I'll update the bug in case I need it.