bluekitchen/btstack #189: hid_host_demo

Flint174, 2019-02-06:
Hello!
I need to connect Bluetooth remote controller (which is bt classic hid device) to esp32. I started with hid_host_demo and got an issue: a minute after the connection is established  esp stops receiving packets and one more minute later "ASSERT_ERR(0), in lc_lmppdu.c at line 66" acquires:

HCI_EVENT_PACKET: 6e->default
ASSERT_ERR(0), in lc_lmppdu.c at line 66
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400837b7  PS      : 0x00060134  A0      : 0x8001c21a  A1      : 0x3ffcdf30
0x400837b7: r_assert_err at ??:?

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffbdbb4
A6      : 0x00000003  A7      : 0x00060223  A8      : 0x800837b4  A9      : 0x3ffcdf10
A10     : 0x00000029  A11     : 0x00000029  A12     : 0x00000010  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffd  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4008375d  LEND    : 0x40083764  LCOUNT  : 0x00000000
0x4008375d: r_assert_err at ??:?

0x40083764: r_assert_err at ??:?

Backtrace: 0x400837b7:0x3ffcdf30 0x4001c217:0x3ffcdf50 0x400286f9:0x3ffcdf80 0x4001a637:0x3ffcdfb0 0x40019d11:0x3ffcdfe0 0x40055b4d:0x
3ffce000 0x400ece13:0x3ffce020 0x400ed305:0x3ffce040 0x4008ef39:0x3ffce070
0x400837b7: r_assert_err at ??:?

0x400ece13: r_rw_schedule at ??:?

0x400ed305: btdm_controller_task at ??:?

0x4008ef39: vPortTaskWrapper at D:/msys32/home/***/esp/esp-idf/components/freertos/port.c:403

Can you help me?

mringwal, 2019-02-06:
Hi. Thanks for reporting. If you get crash deep inside the close-source Bluetooth Controller implementation, please file an issue with Espressif/ESP-IDF framework and also add provide the version of the esp-idf framework. https://github.com/espressif/esp-idf/issues

Do you do a lot of console output while receiving? If yes, reduce the output a bit or disable it for testing.

Flint174, 2019-02-06:
> Do you do a lot of console output while receiving?

I do, added a few printf-s because of error, but it didn't help. Looks like memory leak. Actually, magic happens only after establishing connection

mringwal, 2019-02-06:
I'm suggesting that it might run better with less debug output, although it seems more like some other issue in the Controller implementation.

closed 2019-02-11 by Flint174