bluekitchen/btstack #194: spp_and_le_streamer crash

uborfly, 2019-03-05:
I added spp_and_le_streamer to my project, it crashed.
Here is the logs:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40122c1f  PS      : 0x00060c30  A0      : 0x80100afc  A1      : 0x3ffc7f20  
0x40122c1f: btstack_linked_list_add_tail at /home/ubor/esp/esp-adf/esp-idf/components/btstack/src/btstack_linked_list.c:123

A2      : 0x00000020  A3      : 0x3ffc48b0  A4      : 0x00000001  A5      : 0x3ffc7c10  
A6      : 0x3ffc7c10  A7      : 0x00000002  A8      : 0x800e7b35  A9      : 0x3ffc7be0  
A10     : 0x3ffcd0c4  A11     : 0x3ffcd06c  A12     : 0x3ffc7e34  A13     : 0x00000000  
A14     : 0x00000007  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000020  LBEG    : 0x400917ed  LEND    : 0x400917fd  LCOUNT  : 0xfffffffd  
0x400917ed: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84

0x400917fd: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96

Backtrace: 0x40122c1f:0x3ffc7f20 0x40100af9:0x3ffc7f40 0x400fd1ab:0x3ffc7f60 0x400d758b:0x3ffc7f80 0x400d715a:0x3ffc7fc0 0x400d2872:0x3ffc7ff0
0x40122c1f: btstack_linked_list_add_tail at /home/ubor/esp/esp-adf/esp-idf/components/btstack/src/btstack_linked_list.c:123

0x40100af9: hci_add_event_handler at /home/ubor/esp/esp-adf/esp-idf/components/btstack/src/hci.c:4223

0x400fd1ab: l2cap_init at /home/ubor/esp/esp-adf/esp-idf/components/btstack/src/l2cap.c:1229

0x400d758b: btstack_init at /home/ubor/esp/esp32_lock/main/spp_and_le_streamer.c:417

0x400d715a: app_main at /home/ubor/esp/esp32_lock/main/mc_lock_main.c:197

0x400d2872: main_task at /home/ubor/esp/esp-adf/esp-idf/components/esp32/cpu_start.c:470

what's wrong with it?

mringwal, 2019-03-05:
hard to say. according to the backtrace, strlen is called from btstack_linked_list_add_tail - which doesn't make sense.

did ESP32 BTstack examples work for you before you integrated it with your code?
you could start with one of them and then slowly/step-by-step add your code in to learn what's causing this.

uborfly, 2019-03-06:
> hard to say. according to the backtrace, strlen is called from btstack_linked_list_add_tail - which doesn't make sense.
>
> did ESP32 BTstack examples work for you before you integrated it with your [...]

Thanks! I'll try it.

closed 2019-03-06 by uborfly